pyocd/pyOCD #806: LPC-Link 2 Failing to ACK

buzmeg, 2020-01-14:
The CMSIS-DAP firmware from both LPCScrypt 1.8.2_723 and 2.1.0_842 are failing on an LPC-Link 2:

```
$ pyocd list
  #   Probe                                           Unique ID  
-----------------------------------------------------------------
  0   NXP Semiconductors LPC-LINK2 CMSIS-DAP V5.182   I3FSFVKS   
(venv) MacBook-Pro:pyocd foouser$ pyocd gdbserver
0000298:INFO:board:Target type is cortex_m
0000298:WARNING:board:Generic 'cortex_m' target type is selected; is this intentional? You will be able to debug but not program flash. To set the target type use the '--target' argument or 'target_override' option. Use 'pyocd list --targets' to see available targets types.
0000364:CRITICAL:__main__:No ACK received
Traceback (most recent call last):
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/probe/cmsis_dap_probe.py", line 255, in read_dp_result_callback
    return result()
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 810, in read_reg_cb
    res = transfer.get_result()
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 141, in get_result
    self.daplink.flush()
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 665, in flush
    self._read_packet()
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 894, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 447, in decode_data
    data = self._decode_transfer_block_data(data)
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 416, in _decode_transfer_block_data
    self._check_response(data[3])
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 339, in _check_response
    raise DAPAccessIntf.TransferError("No ACK received")
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferError: No ACK received

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/__main__.py", line 344, in run
    self._COMMANDS[self._args.cmd](self)
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/__main__.py", line 572, in do_gdbserver
    with session:
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/core/session.py", line 291, in __enter__
    self.open()
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/core/session.py", line 371, in open
    self._board.init()
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/board/board.py", line 83, in init
    self.target.init()
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/core/coresight_target.py", line 162, in init
    seq.invoke()
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/coresight/dap.py", line 133, in init
    self.read_id_code()
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/coresight/dap.py", line 234, in read_id_code
    self.dpidr = self.read_reg(DP_IDCODE)
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/coresight/dap.py", line 248, in read_reg
    return self.read_dp(addr, now)
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/coresight/dap.py", line 368, in read_dp
    return read_dp_cb()
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/coresight/dap.py", line 360, in read_dp_cb
    result = result_cb()
  File "/Users/foouser/python/pyocd/venv/lib/python3.7/site-packages/pyocd/probe/cmsis_dap_probe.py", line 258, in read_dp_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.TransferError: No ACK received

```

flit, 2020-03-01:
What target MCU are you using?

What state is the target in when you attempt to connect? A failure to ack upon connect can be caused by the device being in a deep sleep state.

buzmeg, 2020-06-26:
Closing because it doesn't seem to be replicating with my current LPC Link II.  I'll open another bug if I see it again.

closed 2020-06-26 by buzmeg