pyocd/pyOCD #808: MIMXRT1050-EVKB flash fails

juhhov, 2020-01-21:
Flash MIMXRT1050-EVKB board constantly fails. There seems to be two type of errors occurring. Daplink details.txt and full traces are attached.

Error number one:
```pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferError: No ACK received```
Error number two:
```pyocd.core.exceptions.DebugError: No cores were discovered!```
```
$ pyocd --version
0.24.1
```

[daplink.txt](https://github.com/mbedmicro/pyOCD/files/4091386/daplink.txt)
[trace.txt](https://github.com/mbedmicro/pyOCD/files/4091387/trace.txt)
[trace2.txt](https://github.com/mbedmicro/pyOCD/files/4091388/trace2.txt)

JanneKiiskila, 2020-01-21:
@mmahadevan108  - can you look into this, please? This is a blocking issue for RaaS / CI.

mmahadevan108, 2020-01-21:
Release 254 does not have fixes that were uploaded to DAPLink repo in Nov 2019. Please see if you can get the latest from DAPLink master

JanneKiiskila, 2020-01-22:
@mmahadevan108  - can you build the DAPLink? At least I'm not able to build it...

mmahadevan108, 2020-01-22:
Below is the binary that I compiled
[k20dx_mimxrt1050_evk_hyper_if_crc_legacy_0x8000.zip](https://github.com/mbedmicro/pyOCD/files/4097995/k20dx_mimxrt1050_evk_hyper_if_crc_legacy_0x8000.zip)

maclobdell, 2020-02-04:
@JanneKiiskila - please confirm, is this resolved now?

JanneKiiskila, 2020-02-05:
Tried with the attached image, unfortunately it still fails.

```
(mbed) jankii01@ubuntu:~/mbed/mbed-os-example-sockets$ pyocd list
  #   Probe                                     Unique ID                                         
--------------------------------------------------------------------------------------------------
  0   MIMXRT1050-EVKB [mimxrt1050_hyperflash]   02270b0341114e45004b3009c204003fdf91000097969900  
(mbed) jankii01@ubuntu:~/mbed/mbed-os-example-sockets$ pyocd flash ./BUILD/MIMXRT1050_EVK/GCC_ARM/mbed-os-example-sockets.bin
0011426:ERROR:board:link exception during target disconnect:
Traceback (most recent call last):
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/board/board.py", line 96, in uninit
    self.target.disconnect(resume)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/core/coresight_target.py", line 302, in disconnect
    core.disconnect(resume)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 506, in disconnect
    self.resume()
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 1013, in resume
    if self.get_state() != Target.State.HALTED:
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 974, in get_state
    dhcsr = self.read_memory(CortexM.DHCSR)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 638, in read_memory
    result = self.ap.read_memory(addr, transfer_size, now)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/ap.py", line 142, in _locking
    return func(self, *args, **kwargs)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/ap.py", line 583, in _read_memory
    result = read_mem_cb()
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/ap.py", line 565, in read_mem_cb
    res = result_cb()
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/dap.py", line 412, in read_ap_cb
    result = result_cb()
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/probe/cmsis_dap_probe.py", line 297, in read_ap_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/six.py", line 737, in raise_from
    raise value
TransferFaultError: SWD/JTAG Transfer Fault @ 0xe000edf0-0xe000edf3
0011568:CRITICAL:__main__:No ACK received
Traceback (most recent call last):
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/__main__.py", line 344, in run
    self._COMMANDS[self._args.cmd](self)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/__main__.py", line 481, in do_flash
    file_format=self._args.format)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/flash/file_programmer.py", line 157, in program
    self._loader.commit()
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/flash/loader.py", line 168, in commit
    keep_unwritten=self._keep_unwritten)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/flash/builder.py", line 449, in program
    sector_erase_count, page_program_time = self._compute_sector_erase_pages_and_weight(fast_verify)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/flash/builder.py", line 623, in _compute_sector_erase_pages_and_weight
    self._analyze_pages_with_crc32(fast_verify)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/flash/builder.py", line 603, in _analyze_pages_with_crc32
    crc_list = self.flash.compute_crcs(sector_list)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/flash/flash.py", line 318, in compute_crcs
    result = self._call_function_and_wait(self.flash_algo['analyzer_address'], self.begin_data, len(data))
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/flash/flash.py", line 585, in _call_function_and_wait
    return self.wait_for_completion()
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/flash/flash.py", line 533, in wait_for_completion
    while self.target.get_state() == Target.State.RUNNING:
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/core/coresight_target.py", line 390, in get_state
    return self.selected_core.get_state()
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 974, in get_state
    dhcsr = self.read_memory(CortexM.DHCSR)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 638, in read_memory
    result = self.ap.read_memory(addr, transfer_size, now)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/ap.py", line 142, in _locking
    return func(self, *args, **kwargs)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/ap.py", line 583, in _read_memory
    result = read_mem_cb()
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/ap.py", line 565, in read_mem_cb
    res = result_cb()
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/dap.py", line 412, in read_ap_cb
    result = result_cb()
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/probe/cmsis_dap_probe.py", line 297, in read_ap_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/six.py", line 737, in raise_from
    raise value
TransferError: No ACK received
(mbed) jankii01@ubuntu:~/mbed/mbed-os-example-sockets$ pyocd --version
0.24.1
```

@mmahadevan108 - anything we can analyse further?

JanneKiiskila, 2020-02-05:
USB mass storage mode works, though. 

`cp ./BUILD/MIMXRT1050_EVK/GCC_ARM/mbed-os-example-sockets.bin /media/jankii01/RT1050-EVK/ && sync`

JanneKiiskila, 2020-02-05:
Well, that was a bit premature. Actually, even the USB mode flashing fails now. `FAIL.TXT` says now:

```
error: The interface firmware FAILED to reset/halt the target MCU
type: target
```

So, the board is now stuck in a non-programmable mode.

mmahadevan108, 2020-02-05:
Can you share your DAPLink Details.TXT file.

JanneKiiskila, 2020-02-06:
```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 02270b0341114e45004b3009c204003fdf91000097969900
HIC ID: 97969900
Auto Reset: 0
Automation allowed: 0
Overflow detection: 0
Daplink Mode: Interface
Interface Version: 0254
Bootloader Version: 0244
Git SHA: e51b50693ea112e00552ad9fce625b85587f78d7
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Bootloader CRC: 0xfdb85682
Interface CRC: 0x8d300c77
Remount count: 1
URL: http://www.nxp.com/imxrt1050evk
```

JanneKiiskila, 2020-02-06:
Git hash seems to be pointing to: 

```
HEAD is now at e51b5069 Merge pull request #688 from flit/cleanup/misc2
(mbed) jankii01@ubuntu:~/mbed/DAPLink$ git log --oneline
e51b5069 (HEAD, origin/master, origin/HEAD, master) Merge pull request #688 from flit/cleanup/misc2
3bfd3028 Trailing whitespace cleanup.
72f97fed Merged macro.h into util.h.
```

mmahadevan108, 2020-02-06:
Are you able to drag-n-drop a binary to the target? Below is a hello world binary.
[hello_world.zip](https://github.com/mbedmicro/pyOCD/files/4165308/hello_world.zip)

mmahadevan108, 2020-02-06:
Prints hello world on the serial terminal, make sure baudrate is set to 115200.

mmahadevan108, 2020-02-06:
I am not sure whats currently programmed to flash. I would recommend erasing flash via a debugger: Keil, IAR, MCUXpresso before trying to program.

JanneKiiskila, 2020-02-06:
I had the [mbed-os-example-sockets](https://github.com/ARMmbed/mbed-os-example-sockets) as the trial application. It worked once.

JanneKiiskila, 2020-02-06:
Well, no change really here. Nothing on the serial, I think the actual flashing somehow fails?
(as there is the `FAIL.TXT` in the USB drive). I don't think the FAIL.TXT should even appear on the drive, if the flashing works.

mmahadevan108, 2020-02-06:
Can you please erase the flash contents and then try. I would suggest starting from an empty Flash.

JanneKiiskila, 2020-02-06:
Not having much luck that, either:
```
jankii01@ubuntu:~/mbed/mbed_cloud_docs/Docs/products/client-lite$ pyocd erase --mass-erase
0000489:ERROR:dap:Exception while probing AP#0: 
0000493:CRITICAL:__main__:No cores were discovered!
Traceback (most recent call last):
  File "/home/jankii01/venv/cmsis/local/lib/python2.7/site-packages/pyocd/__main__.py", line 344, in run
    self._COMMANDS[self._args.cmd](self)
  File "/home/jankii01/venv/cmsis/local/lib/python2.7/site-packages/pyocd/__main__.py", line 500, in do_erase
    with session:
  File "/home/jankii01/venv/cmsis/local/lib/python2.7/site-packages/pyocd/core/session.py", line 291, in __enter__
    self.open()
  File "/home/jankii01/venv/cmsis/local/lib/python2.7/site-packages/pyocd/core/session.py", line 371, in open
    self._board.init()
  File "/home/jankii01/venv/cmsis/local/lib/python2.7/site-packages/pyocd/board/board.py", line 83, in init
    self.target.init()
  File "/home/jankii01/venv/cmsis/local/lib/python2.7/site-packages/pyocd/core/coresight_target.py", line 162, in init
    seq.invoke()
  File "/home/jankii01/venv/cmsis/local/lib/python2.7/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
  File "/home/jankii01/venv/cmsis/local/lib/python2.7/site-packages/pyocd/core/coresight_target.py", line 296, in check_for_cores
    raise exceptions.DebugError("No cores were discovered!")
DebugError: No cores were discovered!
```

mmahadevan108, 2020-02-06:
Suggest using a tool like Keil, IAR etc,

JanneKiiskila, 2020-02-06:
It really should not matter WHAT is currently in the flash, IMHO. Whatever is there, should get overwritten by the DAPLINK application anyway (if you're flashing a `.bin` -file). So, we clearly have some underlying issue here at hand.

mmahadevan108, 2020-02-06:
Sharing the below issue: 
https://github.com/ARMmbed/mbed-os/issues/12376

Hence I would suggest, erasing flash. I am not sure if there is something programmed to flash that keeps the platform continuously in deep-sleep mode.

JanneKiiskila, 2020-02-06:
Well, I've now played with the KEIL MDK v5.29 and got the Flash menu activated. Tried erase there and well, I think it's still using the same underlying DAPLink, so no luck there either.
[image]

JanneKiiskila, 2020-02-06:
So,  I don't think we can progress anywhere until we get a better working DAPLink. Is there an older, better working image available somewhere?

mmahadevan108, 2020-02-06:
Please try setting switch to SW7 to OFF, ON, ON, ON. This will prevent the ROM from running the programmed binary. Then try erasing via Keil. 
Once erased, set SW7 back to OFF, ON, ON, OFF

JanneKiiskila, 2020-02-06:
Should not the programmed binary at least try to print the starting statements? It's not printing out anything.

mmahadevan108, 2020-02-06:
Hello World binary  should print "Hello World" on the UART terminal and echo whatever you type.

JanneKiiskila, 2020-02-06:
Error changes to "INVALID ROM TABLE" in Keil. It will not erase nor download the example.

JanneKiiskila, 2020-02-06:
KEIL Flash settings.

[image]

[image]

JanneKiiskila, 2020-02-06:
I downgraded the DAPLink firmware to the one available on the NXP website, I get a slightly different error to `FAIL.TXT` with that one;

`The interface firmware FAILED to reset/halt the target MCU`

But, still flashing fails.

mmahadevan108, 2020-02-06:
244 had the wrong flashing algorithm and would not work for drag-n-drop. 
I am able to erase on my board via Keil. I don't see anything different in terms the settings you sent.

jarlamsa, 2020-02-07:
The hello world application that @mmahadevan108 provided is working for me with the daplink he compiled. Flashing a larger binary to the board is very very slow now though.

JanneKiiskila, 2020-02-07:
The official OpenSDA pages listed for this board are offering that, might be useful to update the official NXP offering for that? 

Ref: 
https://www.nxp.com/design/microcontrollers-developer-resources/ides-for-kinetis-mcus/opensda-serial-and-debug-adapter:OPENSDA?&tid=vanOpenSDA#MIMXRT1050-EVK

Other ref doc;

https://community.nxp.com/docs/DOC-340156

mentions that I need to:

```
1. Configure the board for serial downloader mode by setting SW7 to OFF-ON-OFF-ON. 
2. Press SW3 to reset the processor.
```

Is that correct?

mmahadevan108, 2020-02-07:
OFF-ON-OFF-ON would be better than the one I sent. Yes, reset is SW4.

JanneKiiskila, 2020-02-10:
With the OFF-ON-OFF-ON config the flashing seems to work, or at least there is no `FAIL.TXT` generated. However, the application does not seem to start or at least nothing comes out of the serial anymore.
- I've tried with the `mbed-os-example-sockets` (9600 baud) and the `hello-world.bin` (115200 baud). 
- Both with the public DAPLink version and the custom one provided in this issue.

KEIL erasing does not work either, but with the OFF-ON-OFF-ON -config it at least tries, but does not succeed. KEIL flashing (aka upload) of `hello_world.c`  also fails, to a timeout.

```
Program Size: Code=9748 RO-data=2900 RW-data=8 ZI-data=264268  
"flexspi_nor_debug\hello_world.out" - 0 Error(s), 0 Warning(s).
Build Time Elapsed:  00:00:06
Full Chip Erase Failed!
Flash Erase finished 12:39:01
Load "flexspi_nor_debug\\hello_world.out" 
Erase Failed!
Cannot access Memory
Error: Flash Download failed  -  "Cortex-M7"
Flash Load finished at 12:39:29
```

JanneKiiskila, 2020-02-10:
Ok, so with some help from @jarlamsa - he changed the SW7 back to OFF, ON, ON, OFF and now it boots from the correct place and the SW runs. I didn't realize the OFF, ON, OFF, ON setup was supposed to be used temporarily. Hello world at least now runs correctly.

`pyocd flash` still does not work however. 

```
pyocd flash ~/Downloads/hello_world.bin 
[====================] 100%
0028815:INFO:loader:Erased 262144 bytes (1 sector), programmed 262144 bytes (512 pages), skipped 0 bytes (0 pages) at 9.06 kB/s
```

It flashes successfully, but the app does not start again. In RaaS (test automation setup) we can't keep changing the SW7 modes.

If I then copy over (`cp ~/Downloads/hello_world.bin /media/jankii01/RT1050-EVK/ && sync`) it recovers and app starts again.

Flashing the bigger application (mbed-os-example-sockets) fails with `pyocd flash`, still.

```
(mbed) jankii01@ubuntu:~/mbed/mbed-os-example-sockets$ pyocd flash BUILD/MIMXRT1050_EVK/GCC_ARM/mbed-os-example-sockets.bin 
0007803:ERROR:board:link exception during target disconnect:
Traceback (most recent call last):
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/board/board.py", line 96, in uninit
    self.target.disconnect(resume)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/core/coresight_target.py", line 302, in disconnect
    core.disconnect(resume)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 506, in disconnect
    self.resume()
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 1013, in resume
    if self.get_state() != Target.State.HALTED:
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 974, in get_state
    dhcsr = self.read_memory(CortexM.DHCSR)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 638, in read_memory
    result = self.ap.read_memory(addr, transfer_size, now)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/ap.py", line 142, in _locking
    return func(self, *args, **kwargs)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/ap.py", line 583, in _read_memory
    result = read_mem_cb()
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/ap.py", line 565, in read_mem_cb
    res = result_cb()
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/dap.py", line 412, in read_ap_cb
    result = result_cb()
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/probe/cmsis_dap_probe.py", line 297, in read_ap_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/six.py", line 737, in raise_from
    raise value
TransferError: No ACK received
0007831:CRITICAL:__main__:No ACK received
Traceback (most recent call last):
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/__main__.py", line 344, in run
    self._COMMANDS[self._args.cmd](self)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/__main__.py", line 481, in do_flash
    file_format=self._args.format)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/flash/file_programmer.py", line 157, in program
    self._loader.commit()
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/flash/loader.py", line 168, in commit
    keep_unwritten=self._keep_unwritten)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/flash/builder.py", line 449, in program
    sector_erase_count, page_program_time = self._compute_sector_erase_pages_and_weight(fast_verify)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/flash/builder.py", line 623, in _compute_sector_erase_pages_and_weight
    self._analyze_pages_with_crc32(fast_verify)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/flash/builder.py", line 603, in _analyze_pages_with_crc32
    crc_list = self.flash.compute_crcs(sector_list)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/flash/flash.py", line 318, in compute_crcs
    result = self._call_function_and_wait(self.flash_algo['analyzer_address'], self.begin_data, len(data))
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/flash/flash.py", line 585, in _call_function_and_wait
    return self.wait_for_completion()
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/flash/flash.py", line 533, in wait_for_completion
    while self.target.get_state() == Target.State.RUNNING:
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/core/coresight_target.py", line 390, in get_state
    return self.selected_core.get_state()
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 974, in get_state
    dhcsr = self.read_memory(CortexM.DHCSR)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 638, in read_memory
    result = self.ap.read_memory(addr, transfer_size, now)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/ap.py", line 142, in _locking
    return func(self, *args, **kwargs)
[... cut ...]
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/coresight/dap.py", line 412, in read_ap_cb
    result = result_cb()
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/pyocd/probe/cmsis_dap_probe.py", line 297, in read_ap_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "/home/jankii01/venv/mbed/local/lib/python2.7/site-packages/six.py", line 737, in raise_from
    raise value
TransferError: No ACK received
```

So, we have some issue still.

JanneKiiskila, 2020-02-10:
After you have tried to copy over the `mbed-os-example-sockets`, board dies again. Even flashing the `hello_world.bin` does not seem to bring it back. Is the board allergic to bigger binaries?

Hello world is only 16 kB, while socket example is 143 kB.

jarlamsa, 2020-02-10:
I don't have any problem flashing with drag & drop with the `mbed-os-example-sockets`, which toolchain are you using? Haven't tried with the pyOCD yet.

JanneKiiskila, 2020-02-10:
Toolchain used is; gcc-arm-none-eabi-9-2019-q4-major

jarlamsa, 2020-02-11:
I took a quick glance of the flash algorithms in the daplink-repo, they were changed in https://github.com/ARMmbed/DAPLink/commit/d33bff41de1f567c8504f97456e26271da5d46f9
I took the latest CMSIS-pack from Keil and noticed that the generated flash-algorithm for the hyperflash doesn't match the hyperhlash in the daplink but the spiflash instead. Were those mixed by accident somewhere? And what place has the wrong algorithms now? The pyocd-algorithms would need update also.

jarlamsa, 2020-02-11:
@mmahadevan108 could you double-check the flash-algorithms?

mmahadevan108, 2020-02-13:
I compared the flash blob that is is inside ARMmbed/DAPLink with the internal code base. I don't see a mistake, the file is for HyperFlash.

jarlamsa, 2020-02-14:
Ok, thanks @mmahadevan108

maclobdell, 2020-03-02:
Today we had a meeting to debug pyOCD on I.MXRT1050/2.  

We tried two things.
1) Switch pyOCD to use the flash algo that is delivered with CMSIS pack 

Steps to reproduce: 
```
pyocd pack -i "*RT1052"
```
(wait to complete, ignore errors)
```
pyocd list -t -n rt1052 
pyocd flash -t mimxrt1052xxxxb "C:\Users\user\Downloads\mbed-os-example-blinky.MIMXRT1050_EVK.bin"
```

**Result 1 ** Unfortunately, this fails for a known issue with the CMSIS pack for I.MXRT1052.
```
   pyocd.core.exceptions.TargetSupportError: No boot memory is defined for this device
```

Next we tried...
2) copy the flash algo from Daplink to pyOCD.

Steps to reproduce: 

Install pyOCD from sources

```
git clone https://github.com/mbedmicro/pyocd  
cd pyocd
pip install -e . 
```
copy flash algo and pointers from: 
`daplink/source/family/freescale/mimxrt1050_hyper_flash/flash_blob.c`
to:
`pyocd/target/builtin/target_MIMXRT1052xxxxB.py`

then try to flash: 
```
pyocd flash -t mimxrt1050_hyperflash "C:\Users\user\Downloads\mbed-os-example-blinky.MIMXRT1050_EVK.bin"
```
**Result 2** This seems to fail in the same ways as previously.

```
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferError: No ACK received
```
The board gets into a bad state and needs to be recovered using the SW7 switches and flashing with DAPLink Drag and Drop.  

**Actions** - Ryan Chen from NXP has been contacted to debug pyOCD issues on I.MXRT105x.  

I have ran all Mbed-OS tests on I.MXRT1050 using DAPLink drag-and-drop flashing (with the latest DAPLink binary provided by Mahesh) and testing with the Mbed OS master branch.  This seems to be stable and working.

juhhov, 2020-03-24:
#838 seems to resolve the flash issue, thanks for that. Tested with [head of master](https://github.com/mbedmicro/pyOCD/commit/bf66fdd86568d5011c555ad101ec2c2876dab7ec).
@maclobdell the erase however still seems to fail.
```
pyocd erase -vvv -u 0227000041114e45005d300ac207003692d1000097969900 -t mimxrt1050_hyperflash --pack NXP.MIMXRT1052_DFP.12.1.0.pack --chip
...
Traceback (most recent call last):
  File "pyOCD/py3/lib/python3.5/site-packages/pyocd/board/board.py", line 96, in uninit
    self.target.disconnect(resume)
  File "pyOCD/py3/lib/python3.5/site-packages/pyocd/core/coresight_target.py", line 304, in disconnect
    self.dp.power_down_debug()
  File "pyOCD/py3/lib/python3.5/site-packages/pyocd/coresight/dap.py", line 303, in power_down_debug
    r = self.read_reg(DP_CTRL_STAT)
  File "pyOCD/py3/lib/python3.5/site-packages/pyocd/coresight/dap.py", line 260, in read_reg
    return self.read_dp(addr, now)
  File "pyOCD/py3/lib/python3.5/site-packages/pyocd/coresight/dap.py", line 374, in read_dp
    return read_dp_cb()
  File "pyOCD/py3/lib/python3.5/site-packages/pyocd/coresight/dap.py", line 366, in read_dp_cb
    result = result_cb()
  File "pyOCD/py3/lib/python3.5/site-packages/pyocd/probe/cmsis_dap_probe.py", line 260, in read_dp_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.TransferError: No ACK received

```

flit, 2020-03-24:
cc @jhqian (Big thanks for helping us _finally_ make some real progress on the RT1052!)

@juhhov Thanks for testing. Basic flashing also works for me, and I similarly see an exception in some other cases. The failures may be due to not having code on the device. The "No ACK received" error means that the target is not responding to SWD at all. I'm going to look into it some more.

(Btw, you don't need to include the `--pack` argument in the command line since you are using the builtin `mimxrt1050_hyperflash` target type. The RT1052 pack also has some issues that prevent it from being used by pyOCD.)

ljfischer, 2020-04-18:
Ive been working for 2 12+ hour days, trying to build images that can boot/burn into flash for my imxrt1050-evkb board.  I can download existing images, using both drag and drop and MCUbootutility.  Im missing something simple.  
I would like the linker file and build settings for that hello world program, that is included above.  I can drag and drop that onto my drive, and it is happily burned into flash.  I want to know the settings/linker directives to create such a thing!!

JanneKiiskila, 2020-04-18:
The trials were with Mbed OS, the Mbed OS linker files are in the folders available here: https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_MIMXRT1050/device

- separate folder for each tool chain (ARMC, GCCARM, IAR).

ljfischer, 2020-04-18:
Thank you for the incredibly fast response - Im working in MCUXpresso - will attempt to convert this to that.  

Once the hello world image was built, were there follow-on actions in order to make it bootable.  Im looking for a simple guide on what steps need to be taken.  I have read many things, and somehow havent gotten this to fly.

fredlee12001, 2020-04-20:
Hi All, 

I believe https://github.com/mbedmicro/pyOCD/pull/838 + https://github.com/mbedmicro/pyOCD/pull/845
should have fixed the pyocd flash command issue.  

While the pyocd erase command still have "No ACK received" errors. 

Just FYI.

Thanks.

JanneKiiskila, 2020-04-20:
Sounds like this case could be closed a new one should be raised for the --erase, instead.

juhhov, 2020-04-22:
After few flash rounds the board goes to "No cores were discovered!" state and has to be restored with dip switch altering and erase. This blocks CI usage.

Steps:
1. ``` pyocd flash -Oconnect_mode=under-reset -vvv -u 02270b0341114e45005d300ac207003692d1000097969900 -t mimxrt1050_hyperflash  printer_MIMXRT1050_EVK.hex```
2. ```pyocd flash -Oconnect_mode=under-reset -vvv -u 02270b0341114e45005d300ac207003692d1000097969900 -t mimxrt1050_hyperflash  payload_MIMXRT1050_EVK.hex```
3. ```pyocd flash -Oconnect_mode=under-reset -vvv -u 02270b0341114e45005d300ac207003692d1000097969900 -t mimxrt1050_hyperflash  payload_null_MIMXRT1050_EVK.hex```

Using pyoCD version mbedmicro/pyOCD@fdc69fac102880de4b845bcedf540dbd0c43d7b5.
[daplink.txt](https://github.com/mbedmicro/pyOCD/files/4514347/daplink.txt)
[no_cores_were_discovered_trace.txt](https://github.com/mbedmicro/pyOCD/files/4514348/no_cores_were_discovered_trace.txt)
[binaries.zip](https://github.com/mbedmicro/pyOCD/files/4514388/binaries.zip)

JanneKiiskila, 2020-04-22:
Ok, then this definitely cannot be closed. This is a blocker issue, @mmahadevan108  @maclobdell  - can we get some runtime, please? Test automation simply will not work this kind of behaviour.

juhhov, 2020-04-29:
Created #861 and #862. I'll close this.

closed 2020-04-29 by juhhov