pyocd/pyOCD #809: Failed to install pyOCD in virtual environment

ihos-cypress, 2020-01-22:
I am able to install pyocd into system python, however it fails when I try to install it in virtual environment.

```
$ pip install pyocd
Collecting pyocd
  Using cached pyocd-0.24.1-py2.py3-none-any.whl (6.8 MB)
Processing /Users/ihos/Library/Caches/pip/wheels/08/99/c0/5a5942f5b9567c59c14aac76f95a70bf11dccc71240b91ebf5/intervaltree-3.0.2-cp37-none-any.whl
Processing /Users/ihos/Library/Caches/pip/wheels/e4/76/4d/a95b8dd7b452b69e8ed4f68b69e1b55e12c9c9624dd962b191/PyYAML-5.3-cp37-cp37m-macosx_10_14_x86_64.whl
Processing /Users/ihos/Library/Caches/pip/wheels/80/34/1c/3967380d9676d162cb59513bd9dc862d0584e045a162095606/prettytable-0.7.2-cp37-none-any.whl
Collecting colorama
  Using cached colorama-0.4.3-py2.py3-none-any.whl (15 kB)
Processing /Users/ihos/Library/Caches/pip/wheels/1f/a9/7e/d189b5030ee3a56f9b72c28281bb11d661b8ea312e28de08a5/pyusb-1.0.2-cp37-none-any.whl
Collecting cmsis-pack-manager>=0.2.7
  Using cached cmsis-pack-manager-0.2.9.tar.gz (42 kB)
Collecting pyelftools
  Using cached pyelftools-0.26-py2.py3-none-any.whl (136 kB)
Collecting six<2.0,>=1.0
  Using cached six-1.14.0-py2.py3-none-any.whl (10 kB)
Collecting intelhex<3.0,>=2.0
  Using cached intelhex-2.2.1-py2.py3-none-any.whl (50 kB)
Processing /Users/ihos/Library/Caches/pip/wheels/51/0e/c8/c539e26fe889b92fd4f31333a951aeff601bd63c11c3c6c98a/hidapi-0.7.99.post21-cp37-cp37m-macosx_10_14_x86_64.whl
Collecting sortedcontainers<3.0,>=2.0
  Using cached sortedcontainers-2.1.0-py2.py3-none-any.whl (28 kB)
Collecting appdirs>=1.4
  Using cached appdirs-1.4.3-py2.py3-none-any.whl (12 kB)
Collecting milksnake>=0.1.2
  Using cached milksnake-0.1.5-py2.py3-none-any.whl (9.6 kB)
Requirement already satisfied: setuptools>=19.0 in ./latestpyocd/lib/python3.7/site-packages (from hidapi; platform_system == "Darwin"->pyocd) (45.1.0)
Collecting cffi>=1.6.0
  Using cached cffi-1.13.2-cp37-cp37m-macosx_10_6_intel.whl (258 kB)
Processing /Users/ihos/Library/Caches/pip/wheels/f2/9a/90/de94f8556265ddc9d9c8b271b0f63e57b26fb1d67a45564511/pycparser-2.19-py2.py3-none-any.whl
Building wheels for collected packages: cmsis-pack-manager
  Building wheel for cmsis-pack-manager (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /Users/ihos/venv/latestpyocd/bin/python3.7 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/v1/hz6wqtxj75b74_71ks1__xjc0000gn/T/pip-install-8d303om6/cmsis-pack-manager/setup.py'"'"'; __file__='"'"'/private/var/folders/v1/hz6wqtxj75b74_71ks1__xjc0000gn/T/pip-install-8d303om6/cmsis-pack-manager/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/v1/hz6wqtxj75b74_71ks1__xjc0000gn/T/pip-wheel-9asg9lqz
       cwd: /private/var/folders/v1/hz6wqtxj75b74_71ks1__xjc0000gn/T/pip-install-8d303om6/cmsis-pack-manager/
  Complete output (9 lines):
  running bdist_wheel
  running build
  running build_py
  creating build/lib
  creating build/lib/cmsis_pack_manager
  copying cmsis_pack_manager/_version.py -> build/lib/cmsis_pack_manager
  copying cmsis_pack_manager/__init__.py -> build/lib/cmsis_pack_manager
  copying cmsis_pack_manager/pack_manager.py -> build/lib/cmsis_pack_manager
  error: [Errno 2] No such file or directory: 'cargo': 'cargo'
  ----------------------------------------
  ERROR: Failed building wheel for cmsis-pack-manager
  Running setup.py clean for cmsis-pack-manager
Failed to build cmsis-pack-manager
Installing collected packages: sortedcontainers, intervaltree, pyyaml, prettytable, colorama, pyusb, appdirs, pycparser, cffi, milksnake, cmsis-pack-manager, pyelftools, six, intelhex, hidapi, pyocd
    Running setup.py install for cmsis-pack-manager ... error
    ERROR: Command errored out with exit status 1:
     command: /Users/ihos/venv/latestpyocd/bin/python3.7 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/v1/hz6wqtxj75b74_71ks1__xjc0000gn/T/pip-install-8d303om6/cmsis-pack-manager/setup.py'"'"'; __file__='"'"'/private/var/folders/v1/hz6wqtxj75b74_71ks1__xjc0000gn/T/pip-install-8d303om6/cmsis-pack-manager/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/v1/hz6wqtxj75b74_71ks1__xjc0000gn/T/pip-record-maob9q7c/install-record.txt --single-version-externally-managed --compile --install-headers /Users/ihos/venv/latestpyocd/bin/../include/site/python3.7/cmsis-pack-manager
         cwd: /private/var/folders/v1/hz6wqtxj75b74_71ks1__xjc0000gn/T/pip-install-8d303om6/cmsis-pack-manager/
    Complete output (9 lines):
    running install
    running build
    running build_py
    creating build/lib
    creating build/lib/cmsis_pack_manager
    copying cmsis_pack_manager/_version.py -> build/lib/cmsis_pack_manager
    copying cmsis_pack_manager/__init__.py -> build/lib/cmsis_pack_manager
    copying cmsis_pack_manager/pack_manager.py -> build/lib/cmsis_pack_manager
    error: [Errno 2] No such file or directory: 'cargo': 'cargo'
    ----------------------------------------
ERROR: Command errored out with exit status 1: /Users/ihos/venv/latestpyocd/bin/python3.7 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/v1/hz6wqtxj75b74_71ks1__xjc0000gn/T/pip-install-8d303om6/cmsis-pack-manager/setup.py'"'"'; __file__='"'"'/private/var/folders/v1/hz6wqtxj75b74_71ks1__xjc0000gn/T/pip-install-8d303om6/cmsis-pack-manager/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/v1/hz6wqtxj75b74_71ks1__xjc0000gn/T/pip-record-maob9q7c/install-record.txt --single-version-externally-managed --compile --install-headers /Users/ihos/venv/latestpyocd/bin/../include/site/python3.7/cmsis-pack-manager Check the logs for full command output.
```

juhhov, 2020-01-24:
https://github.com/ARMmbed/cmsis-pack-manager/issues/137
As a workaround downgrade pip to version < 20.

ihos-cypress, 2020-01-28:
> [ARMmbed/cmsis-pack-manager#137](https://github.com/ARMmbed/cmsis-pack-manager/issues/137)
> As a workaround downgrade pip to version < 20.

Thanks! Also, it is already fixed in 20.0.2.

closed 2020-01-28 by ihos-cypress