pyocd/pyOCD #812: NUCLEO_H743ZI2 - pyocd flash does not work (pyocd 0.24.1)

JanneKiiskila, 2020-01-24:
NUCLEO_H743ZI2 cannot be flashed, despite manually updating the CMSIS index/packs.

pip list 
```
mbed) jankii01@ubuntu:~/mbed/mbed-os-example-pelion$ pip list
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Package            Version    
------------------ -----------
appdirs            1.4.3      
asn1ate            0.6.0      
asn1crypto         1.0.1      
beautifulsoup4     4.6.3      
certifi            2019.9.11  
cffi               1.12.3     
chardet            3.0.4      
Click              7.0        
cmsis-pack-manager 0.2.8      
colorama           0.3.9      
cryptography       2.4.2      
ecdsa              0.13.3     
enum34             1.1.6      
fasteners          0.15       
functools32        3.2.3.post2
future             0.16.0     
fuzzywuzzy         0.17.0     
icetea             1.2.4      
idna               2.7        
intelhex           2.2.1      
intervaltree       3.0.2      
ipaddress          1.0.22     
Jinja2             2.10.3     
jsonmerge          1.7.0      
jsonschema         2.6.0      
junit-xml          1.8        
lockfile           0.12.2     
manifest-tool      1.5.2      
MarkupSafe         1.1.1      
mbed-cli           1.10.1     
mbed-cloud-sdk     2.0.8      
mbed-flasher       0.10.1     
mbed-greentea      1.7.2      
mbed-host-tests    1.5.8      
mbed-ls            1.7.8      
mbed-os-tools      0.0.9      
milksnake          0.1.5      
monotonic          1.5        
pip                20.0.1     
prettytable        0.7.2      
protobuf           3.5.2.post1
psutil             5.6.2      
pyasn1             0.2.3      
pyasn1-modules     0.2.7      
pycparser          2.19       
pycryptodome       3.7.3      
pyelftools         0.25       
pyocd              0.24.1     
pyOpenSSL          19.0.0     
pyparsing          2.4.2      
pyserial           3.4        
python-dateutil    2.8.0      
python-dotenv      0.10.3     
pyusb              1.0.2      
PyYAML             5.3        
requests           2.20.1     
semver             2.8.1      
setuptools         41.4.0     
six                1.12.0     
sortedcontainers   2.1.0      
typing             3.7.4.1    
urllib3            1.24.2     
wheel              0.33.6     
WMI                1.4.9      
yattag             1.12.2  
```

So, latest pyocd is used (0.24.1)

Manual updates of packs do not help either,

```
(cmsis) jankii01@ubuntu:~/mbed/mbed-os-example-pelion$ pyocd pack --update
... lots of stuff
(cmsis) jankii01@ubuntu:~/mbed/mbed-os-example-pelion$ pyocd pack --find stm32h7
  Part            Vendor   Pack            Version  
----------------------------------------------------
  STM32H742AGIx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H742AIIx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H742BGTx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H742BITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H742IGKx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H742IGTx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H742IIKx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H742IITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H742VGHx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H742VGTx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H742VIHx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H742VITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H742XGHx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H742XIHx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H742ZGTx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H742ZITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H743AGIx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H743AIIx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H743BGTx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H743BITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H743IGKx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H743IGTx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H743IIKx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H743IITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H743VGHx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H743VGTx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H743VIHx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H743VITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H743XGHx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H743XIHx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H743ZGTx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H743ZITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H745BGTx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H745BITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H745IGKx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H745IGTx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H745IIKx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H745IITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H745XGHx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H745XIHx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H745ZGTx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H745ZITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H747AGIx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H747AIIx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H747BGTx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H747BITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H747IGTx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H747IITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H747XGHx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H747XIHx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H747ZIYx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H750IBKx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H750VBTx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H750XBHx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H753AIIx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H753BITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H753IIKx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H753IITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H753VIHx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H753VITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H753XIHx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H753ZITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H755BITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H755IIKx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H755IITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H755XIHx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H755ZITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H757AIIx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H757BITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H757IITx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H757XIHx   Keil     STM32H7xx_DFP   2.3.1    
  STM32H757ZIYx   Keil     STM32H7xx_DFP   2.3.1    
(cmsis) jankii01@ubuntu:~/mbed/mbed-os-example-pelion$ pyocd pack --install STM32H743ZITx
Downloading packs (press Control-C to cancel):
    Keil::STM32H7xx_DFP::2.3.1
(cmsis) jankii01@ubuntu:~/mbed/mbed-os-example-pelion$ pyocd flash ./BUILD/NUCLEO_H743ZI2/GCC_ARM/mbed-os-example-pelion.bin
0000798:ERROR:dap:Exception reading AP#0 IDR: STLink error (5): No device connected
0000809:ERROR:dap:Exception reading AP#1 IDR: STLink error (5): No device connected
0000818:ERROR:dap:Exception reading AP#2 IDR: STLink error (5): No device connected
0000833:CRITICAL:__main__:No cores were discovered!
Traceback (most recent call last):
  File "/home/jankii01/venv/cmsis/local/lib/python2.7/site-packages/pyocd/__main__.py", line 344, in run
    self._COMMANDS[self._args.cmd](self)
  File "/home/jankii01/venv/cmsis/local/lib/python2.7/site-packages/pyocd/__main__.py", line 474, in do_flash
    with session:
  File "/home/jankii01/venv/cmsis/local/lib/python2.7/site-packages/pyocd/core/session.py", line 291, in __enter__
    self.open()
  File "/home/jankii01/venv/cmsis/local/lib/python2.7/site-packages/pyocd/core/session.py", line 371, in open
    self._board.init()
  File "/home/jankii01/venv/cmsis/local/lib/python2.7/site-packages/pyocd/board/board.py", line 83, in init
    self.target.init()
  File "/home/jankii01/venv/cmsis/local/lib/python2.7/site-packages/pyocd/core/coresight_target.py", line 162, in init
    seq.invoke()
  File "/home/jankii01/venv/cmsis/local/lib/python2.7/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
  File "/home/jankii01/venv/cmsis/local/lib/python2.7/site-packages/pyocd/core/coresight_target.py", line 296, in check_for_cores
    raise exceptions.DebugError("No cores were discovered!")
DebugError: No cores were discovered!
(cmsis) jankii01@ubuntu:~/mbed/mbed-os-example-pelion$ pyocd list
  #   Probe                            Unique ID                 
-----------------------------------------------------------------
  0   NUCLEO-H743ZI2 [stm32h743zitx]   003700213137511239383538  
```

So, despite getting the right pack - it still doesn't work?

Internal ref; IOTPART-9002.

JanneKiiskila, 2020-01-24:
@flit  @MarceloSalazar

flit, 2020-03-01:
The "STLink error (5): No device connected" errors are interesting. There is an odd sequence of events going on here. pyocd has succeeded in communicating with the DP, and seems to have read the AP[0-2] IDRs once each, as it has detected those IDRs. But on the second read of the IDRs, it fails with this error.

@schstm Would you have any special insight about what the above error indicates?

schstm, 2020-03-10:
Hello,
is it possible that pyocd connects initially while the target is running and then goes in sleep mode ? Is it possible to try a connect sequence with target reset (and halt) before flashing ? A more detailed sequence of commands at ST-Link level done by "pyocd flash" could help me understanding the failure

flit, 2020-03-11:
Agreed, please try testing again with `-Oconnect_mode=under-reset` on the command line (or set that option in `pyocd.yaml` config file).

You can get a low-level trace of the STLink USB traffic by adding these lines to `pyocd.yaml`:

```{.yaml}
logging:
  loggers:
    pyocd.probe.stlink.usb.trace:
      level: DEBUG
```

juhhov, 2020-03-19:
Fails with under-reset connect mode as well.
```
$ pyocd --version
0.25.0
$ pyocd flash -vvv -Oconnect_mode=under-reset -u 003700213137511239383538 -t stm32h743zitx --pack Keil.STM32H7xx_DFP.2.3.1.pack mbed_cliapp_NUCLEO_H743ZI2.bin 
```
<details>
<summary>Full trace</summary>

```
$ pyocd flash -vvv -Oconnect_mode=under-reset -u 003700213137511239383538 -t stm32h743zitx --pack Keil.STM32H7xx_DFP.2.3.1.pack mbed_cliapp_NUCLEO_H743ZI2.bin 
0000342:DEBUG:session:Project directory: /home/raas/NUCLEO_H743ZI2
0000369:DEBUG:pack_target:Loading target 'stm32h742agix' from CMSIS-Pack
0000369:DEBUG:pack_target:Loading target 'stm32h742aiix' from CMSIS-Pack
0000369:DEBUG:pack_target:Loading target 'stm32h742bgtx' from CMSIS-Pack
0000369:DEBUG:pack_target:Loading target 'stm32h742bitx' from CMSIS-Pack
0000369:DEBUG:pack_target:Loading target 'stm32h742igkx' from CMSIS-Pack
0000369:DEBUG:pack_target:Loading target 'stm32h742iikx' from CMSIS-Pack
0000370:DEBUG:pack_target:Loading target 'stm32h742igtx' from CMSIS-Pack
0000370:DEBUG:pack_target:Loading target 'stm32h742iitx' from CMSIS-Pack
0000370:DEBUG:pack_target:Loading target 'stm32h742vghx' from CMSIS-Pack
0000370:DEBUG:pack_target:Loading target 'stm32h742vihx' from CMSIS-Pack
0000370:DEBUG:pack_target:Loading target 'stm32h742vgtx' from CMSIS-Pack
0000370:DEBUG:pack_target:Loading target 'stm32h742vitx' from CMSIS-Pack
0000370:DEBUG:pack_target:Loading target 'stm32h742xghx' from CMSIS-Pack
0000370:DEBUG:pack_target:Loading target 'stm32h742xihx' from CMSIS-Pack
0000370:DEBUG:pack_target:Loading target 'stm32h742zgtx' from CMSIS-Pack
0000370:DEBUG:pack_target:Loading target 'stm32h742zitx' from CMSIS-Pack
0000370:DEBUG:pack_target:Loading target 'stm32h743agix' from CMSIS-Pack
0000370:DEBUG:pack_target:Loading target 'stm32h743aiix' from CMSIS-Pack
0000370:DEBUG:pack_target:Loading target 'stm32h743bgtx' from CMSIS-Pack
0000370:DEBUG:pack_target:Loading target 'stm32h743bitx' from CMSIS-Pack
0000370:DEBUG:pack_target:Loading target 'stm32h743igkx' from CMSIS-Pack
0000370:DEBUG:pack_target:Loading target 'stm32h743igtx' from CMSIS-Pack
0000371:DEBUG:pack_target:Loading target 'stm32h743iikx' from CMSIS-Pack
0000371:DEBUG:pack_target:Loading target 'stm32h743iitx' from CMSIS-Pack
0000371:DEBUG:pack_target:Loading target 'stm32h743vghx' from CMSIS-Pack
0000371:DEBUG:pack_target:Loading target 'stm32h743vgtx' from CMSIS-Pack
0000371:DEBUG:pack_target:Loading target 'stm32h743vihx' from CMSIS-Pack
0000371:DEBUG:pack_target:Loading target 'stm32h743vitx' from CMSIS-Pack
0000371:DEBUG:pack_target:Loading target 'stm32h743xghx' from CMSIS-Pack
0000371:DEBUG:pack_target:Loading target 'stm32h743xihx' from CMSIS-Pack
0000371:DEBUG:pack_target:Loading target 'stm32h743zgtx' from CMSIS-Pack
0000371:DEBUG:pack_target:Loading target 'stm32h743zitx' from CMSIS-Pack
0000371:DEBUG:pack_target:Loading target 'stm32h753aiix' from CMSIS-Pack
0000371:DEBUG:pack_target:Loading target 'stm32h753bitx' from CMSIS-Pack
0000371:DEBUG:pack_target:Loading target 'stm32h753iikx' from CMSIS-Pack
0000371:DEBUG:pack_target:Loading target 'stm32h753iitx' from CMSIS-Pack
0000371:DEBUG:pack_target:Loading target 'stm32h753vihx' from CMSIS-Pack
0000371:DEBUG:pack_target:Loading target 'stm32h753vitx' from CMSIS-Pack
0000372:DEBUG:pack_target:Loading target 'stm32h753xihx' from CMSIS-Pack
0000372:DEBUG:pack_target:Loading target 'stm32h753zitx' from CMSIS-Pack
0000372:DEBUG:pack_target:Loading target 'stm32h750ibkx' from CMSIS-Pack
0000372:DEBUG:pack_target:Loading target 'stm32h750vbtx' from CMSIS-Pack
0000372:DEBUG:pack_target:Loading target 'stm32h750xbhx' from CMSIS-Pack
0000372:DEBUG:pack_target:Loading target 'stm32h745bgtx' from CMSIS-Pack
0000372:DEBUG:pack_target:Loading target 'stm32h745bitx' from CMSIS-Pack
0000372:DEBUG:pack_target:Loading target 'stm32h745igkx' from CMSIS-Pack
0000372:DEBUG:pack_target:Loading target 'stm32h745igtx' from CMSIS-Pack
0000372:DEBUG:pack_target:Loading target 'stm32h745iikx' from CMSIS-Pack
0000372:DEBUG:pack_target:Loading target 'stm32h745iitx' from CMSIS-Pack
0000372:DEBUG:pack_target:Loading target 'stm32h745xghx' from CMSIS-Pack
0000372:DEBUG:pack_target:Loading target 'stm32h745xihx' from CMSIS-Pack
0000372:DEBUG:pack_target:Loading target 'stm32h745zgtx' from CMSIS-Pack
0000372:DEBUG:pack_target:Loading target 'stm32h745zitx' from CMSIS-Pack
0000372:DEBUG:pack_target:Loading target 'stm32h755bitx' from CMSIS-Pack
0000372:DEBUG:pack_target:Loading target 'stm32h755iikx' from CMSIS-Pack
0000373:DEBUG:pack_target:Loading target 'stm32h755iitx' from CMSIS-Pack
0000373:DEBUG:pack_target:Loading target 'stm32h755xihx' from CMSIS-Pack
0000373:DEBUG:pack_target:Loading target 'stm32h755zitx' from CMSIS-Pack
[... 57 lines cut ...]
0000871:DEBUG:sequencer:Running task create_components
0000871:DEBUG:sequencer:Running task check_for_cores
0000871:DEBUG:session:uninit session <pyocd.core.session.Session object at 0x7f1d8ced57f0>
0000872:CRITICAL:__main__:No cores were discovered!
Traceback (most recent call last):
  File "/home/raas/py3/lib/python3.5/site-packages/pyocd/__main__.py", line 360, in run
    self._COMMANDS[self._args.cmd](self)
  File "/home/raas/py3/lib/python3.5/site-packages/pyocd/__main__.py", line 516, in do_flash
    with session:
  File "/home/raas/py3/lib/python3.5/site-packages/pyocd/core/session.py", line 299, in __enter__
    self.open()
  File "/home/raas/py3/lib/python3.5/site-packages/pyocd/core/session.py", line 415, in open
[... cut ...]
    self.target.init()
  File "/home/raas/py3/lib/python3.5/site-packages/pyocd/core/coresight_target.py", line 163, in init
    seq.invoke()
  File "/home/raas/py3/lib/python3.5/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
  File "/home/raas/py3/lib/python3.5/site-packages/pyocd/core/coresight_target.py", line 307, in check_for_cores
    raise exceptions.DebugError("No cores were discovered!")
pyocd.core.exceptions.DebugError: No cores were discovered!

```
</details>

PetarZecevic, 2024-12-01:
> Agreed, please try testing again with `-Oconnect_mode=under-reset` on the command line (or set that option in `pyocd.yaml` config file).
> 
> You can get a low-level trace of the STLink USB traffic by adding these lines to `pyocd.yaml`:
> 
> ```
> logging:
>   loggers:
>     pyocd.probe.stlink.usb.trace:
>       level: DEBUG
> ```

This solved the issue for me.