pyocd/pyOCD #813: pyocd pack install fails for STM32G0

erwango, 2020-01-30:
When running pack install command for STM32G0 soc, command fails:
```
$ pyocd pack --install stm32g071rb
0000270:WARNING:__main__:No matching devices. Please make sure the pack index is up to date.
```

Might be due to following previous error:
```
Jan 30 15:13:12.522 ERRO download of "https://www.keil.com/pack/Keil.STM32G0xx_DFP.pdsc" failed: https://www.keil.com/pack/Keil.STM32G0xx_DFP.pdsc: error trying to connect: unexpected end of file

```

cederom, 2020-01-30:
I can confirm this. `nRF52832` installed with no problem :-)

```
(venv36bsd) 0xCFMX4% pyocd pack -u
0000491:INFO:__main__:Updating pack index...
Jan 30 20:55:48.922 ERRO download of "https://www.keil.com/pack/Keil.STM32G0xx_DFP.pdsc" failed: https://www.keil.com/pack/Keil.STM32G0xx_DFP.pdsc: error trying to connect: unexpected end of file
Jan 30 20:55:48.925 ERRO download of "https://www.keil.com/pack/Keil.STM32F7xx_DFP.pdsc" failed: https://www.keil.com/pack/Keil.STM32F7xx_DFP.pdsc: error trying to connect: unexpected end of file
Jan 30 20:55:48.930 ERRO download of "https://www.keil.com/pack/Keil.STM32L4xx_DFP.pdsc" failed: https://www.keil.com/pack/Keil.STM32L4xx_DFP.pdsc: error trying to connect: unexpected end of file
Jan 30 20:55:48.930 ERRO download of "https://www.keil.com/pack/Keil.STM32H7xx_DFP.pdsc" failed: https://www.keil.com/pack/Keil.STM32H7xx_DFP.pdsc: error trying to connect: unexpected end of file
Jan 30 20:55:49.723 ERRO download of "http://www.qorvo.com/extra/keil_pack/Active-Semi.PAC55XX.pdsc" failed: https://www.qorvo.com/extra/keil_pack/Active-Semi.PAC55XX.pdsc: error trying to connect: Connection reset by peer (os error 54)
Jan 30 20:55:49.723 ERRO download of "http://www.qorvo.com/extra/keil_pack/Active-Semi.PAC52XX.pdsc" failed: https://www.qorvo.com/extra/keil_pack/Active-Semi.PAC52XX.pdsc: error trying to connect: Connection reset by peer (os error 54)
Jan 30 20:56:18.974 ERRO download of "http://www.sinowealth.com/ftp/tool/Arm32/Sinowealth.SH32Fxxx_DFP.pdsc" failed: http://www.sinowealth.com/ftp/tool/Arm32/Sinowealth.SH32Fxxx_DFP.pdsc: error trying to connect: failed to lookup address information: Name does not resolve
Jan 30 20:56:19.748 WARN tried to parse element "package" from element "html"
Jan 30 20:56:19.893 WARN Only Comment, CDATA and DOCTYPE nodes can start with a '!'
Jan 30 20:56:20.206 WARN No Core found!, Package: CMSIS, Vendor: ARM
Jan 30 20:56:20.254 WARN invalid digit found in string, Package: TLE984x_DFP, Vendor: Infineon
Jan 30 20:56:20.310 WARN Only Comment, CDATA and DOCTYPE nodes can start with a '!'
Jan 30 20:56:20.514 WARN Unknown file catogory preIncludeLocal, Group: Unity, Class: Test, in: Component, Package: Unity, Vendor: Arm-Packs
(venv36bsd) 0xCFMX4% pyocd pack --install stm32g071rb
0000633:WARNING:__main__:No matching devices. Please make sure the pack index is up to date.
(venv36bsd) 0xCFMX4% pyocd pack --install nrf52832
Downloading packs (press Control-C to cancel):
    NordicSemiconductor::nRF_DeviceFamilyPack::8.30.2
(venv36bsd) 0xCFMX4% pyocd pack -s
  Vendor                Pack                   Version
--------------------------------------------------------
  NordicSemiconductor   nRF_DeviceFamilyPack   8.30.2

(venv36bsd) 0xCFMX4% pyocd --version
0.24.1
(venv36bsd) 0xCFMX4% uname -a
FreeBSD 0xCFMX4 12.1-RELEASE-p1 FreeBSD 12.1-RELEASE-p1 GENERIC  amd64
(venv36bsd) 0xCFMX4% python --version
Python 3.6.9
```

This is strange because opening https://sadevicepacksprodus.blob.core.windows.net/pdsc/Keil.STM32G0xx_DFP.pdsc shows up content correctly in a browser.

```
(venv36bsd) 0xCFMX4% pyocd pack --install stm32g071rb
0000633:WARNING:__main__:No matching devices. Please make sure the pack index is up to date.

(venv36bsd) 0xCFMX4% pyocd pack --install stm32g071
0000705:WARNING:__main__:No matching devices. Please make sure the pack index is up to date.

(venv36bsd) 0xCFMX4% pyocd pack --install stm32g
Downloading packs (press Control-C to cancel):
    Keil::STM32G4xx_DFP::1.1.0

(venv36bsd) 0xCFMX4% pyocd pack -s
  Vendor                Pack                   Version
--------------------------------------------------------
  Keil                  STM32G4xx_DFP          1.1.0
  NordicSemiconductor   nRF_DeviceFamilyPack   8.30.2
```

JanneKiiskila, 2020-01-31:
Likely the root cause is this issue; 
https://github.com/ARMmbed/cmsis-pack-manager/issues/121
@theotherjimmy  @flit

fgr1986, 2020-06-05:
Hi @JanneKiiskila is there any way to manually downliad and later pass the .pdsc file?
Due to this issue, I am also experiencing problems installing packs:
```
pyocd pack --install stm32l476rg
0000384:WARNING:__main__:No matching devices. Please make sure the pack index is up to date.
```
Thanks

flit, 2020-06-07:
@fgr1986 You can download DFPs directly from the [Keil CMSIS-Pack list](https://www.keil.com/dd2/pack/) and use the `--pack` argument to pass them to pyocd.

joectchan, 2020-06-30:
> @fgr1986 You can download DFPs directly from the [Keil CMSIS-Pack list](https://www.keil.com/dd2/pack/) and use the `--pack` argument to pass them to pyocd.

Hello, I use a nucleo_l452re board. I try to follow the above instruction but does not work. I will show what I typed and my results. Please point out what I missed.

I confirm with pyocd that it knows about the board.
```
pyocd list -b | grep 452
  0821   NUCLEO-L452RE                    stm32l452re             None
```
stm32l452re is not a built-in target. I downloaded DFP for STM32L4 from the Keil link. I confirmed the file is fine with `ll` and `strings`.
```
user@3a009e1edd4c:/workdir/zephyrproject$ ll /workdir/Keil.STM32L4xx_DFP.2.4.0.pack
-rw-rw-r-- 1 user user 253902145 Jun 30 03:42 /workdir/Keil.STM32L4xx_DFP.2.4.0.pack

user@3a009e1edd4c:/workdir/zephyrproject$ strings /workdir/Keil.STM32L4xx_DFP.2.4.0.pack | head -n 1
Keil.STM32L4xx_DFP.pdscSDi
```
I use `pyocd cmd -t stm32l452re --pack /workdir/Keil.STM32L4xx_DFP.2.4.0.pack ` to tell pyocd that my target board is stm32l452re and the pack is located at /workdir/Keil.STM32L4xx_DFP.2.4.0.pack. However, I got error message.
```
user@3a009e1edd4c:/workdir/zephyrproject$ pyocd cmd -t stm32l452re --pack /workdir/Keil.STM32L4xx_DFP.2.4.0.pack 
0000526:ERROR:linux:Could not get disk devices by id. This could be because your Linux distribution does not use udev, or does not create /dev/disk/by-id symlinks. Please submit an issue to github.com/armmbed/mbed-ls.
0000798:CRITICAL:__main__:Target type 'stm32l452re' not recognized. Use 'pyocd list --targets' to see currently available target types. See <https://github.com/mbedmicro/pyOCD/blob/master/docs/target_support.md> for how to install additional target support.
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/pyocd/board/board.py", line 52, in __init__
    self.target = TARGET[self._target_type](session)
KeyError: 'stm32l452re'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/pyocd/__main__.py", line 368, in run
    self._COMMANDS[self._args.cmd](self)
  File "/usr/local/lib/python3.6/dist-packages/pyocd/__main__.py", line 702, in do_commander
    PyOCDCommander(self._args, cmds).run()
  File "/usr/local/lib/python3.6/dist-packages/pyocd/tools/pyocd.py", line 692, in run
    if not self.connect():
  File "/usr/local/lib/python3.6/dist-packages/pyocd/tools/pyocd.py", line 795, in connect
    resume_on_disconnect=False,
  File "/usr/local/lib/python3.6/dist-packages/pyocd/core/helpers.py", line 241, in session_with_chosen_probe
    return Session(probe, auto_open=auto_open, options=options, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/pyocd/core/session.py", line 179, in __init__
    or Board(self, self.options.get('target_override'))
  File "/usr/local/lib/python3.6/dist-packages/pyocd/board/board.py", line 58, in __init__
    "for how to install additional target support." % self._target_type), exc)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.TargetSupportError: Target type 'stm32l452re' not recognized. Use 'pyocd list --targets' to see currently available target types. See <https://github.com/mbedmicro/pyOCD/blob/master/docs/target_support.md> for how to install additional target support.
user@3a009e1edd4c:/workdir/zephyrproject$ 
```
I tried a few other potential command line combinations, but still fail.
```
pyocd cmd -t stm32l452re --pack=/workdir/Keil.STM32L4xx_DFP.2.4.0.pack

pyocd pack --install /workdir/Keil.STM32L4xx_DFP.2.4.0.pack -n

pyocd pack --install /workdir/Keil.STM32L4xx_DFP.2.4.0.pack
```
Please show an sample command line that works. The phrase "use the `--pack` argument to pass them to pyocd" is too terse to get a newbie started. Thanks in advance.

joectchan, 2020-06-30:
Further googling shows me [issue 643](https://github.com/mbedmicro/pyOCD/issues/643). I try to follow the instructions over there. I still run into problem.

First, I obtain target_override and probe's unique ID.
```
pyocd list -b | grep 452
  0821   NUCLEO-L452RE                    stm32l452re             None                               
  0829   NUCLEO-L452RE-P                  stm32l452re             None

pyocd list
0000532:ERROR:linux:Could not get disk devices by id. This could be because your Linux distribution does not use udev, or does not create /dev/disk/by-id symlinks. Please submit an issue to github.com/armmbed/mbed-ls.
  #   Probe          Unique ID                 
-----------------------------------------------
  0   STM32 STLink   066DFF515055657867072315  
```
Next, I create this config file /workdir/mypyocd.yaml
```
probes:
  066DFF515055657867072315: # Probe's unique ID.
    target_override:  stm32l452re

# Global options
auto_unlock: false
pack:
  - /workdir/Keil.STM32L4xx_DFP.2.4.0.pack
```
Invoke `pyocd cmd --config=/workdir/mypyocd.yaml`
```
user@3a009e1edd4c:/workdir/zephyrproject$ pyocd cmd --config=/workdir/mypyocd.yaml 
0000571:ERROR:linux:Could not get disk devices by id. This could be because your Linux distribution does not use udev, or does not create /dev/disk/by-id symlinks. Please submit an issue to github.com/armmbed/mbed-ls.
0000854:CRITICAL:__main__:Target type 'stm32l452re' not recognized. Use 'pyocd list --targets' to see currently available target types. See <https://github.com/mbedmicro/pyOCD/blob/master/docs/target_support.md> for how to install additional target support.
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/pyocd/board/board.py", line 52, in __init__
    self.target = TARGET[self._target_type](session)
KeyError: 'stm32l452re'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/pyocd/__main__.py", line 368, in run
    self._COMMANDS[self._args.cmd](self)
  File "/usr/local/lib/python3.6/dist-packages/pyocd/__main__.py", line 702, in do_commander
    PyOCDCommander(self._args, cmds).run()
  File "/usr/local/lib/python3.6/dist-packages/pyocd/tools/pyocd.py", line 692, in run
    if not self.connect():
  File "/usr/local/lib/python3.6/dist-packages/pyocd/tools/pyocd.py", line 795, in connect
    resume_on_disconnect=False,
  File "/usr/local/lib/python3.6/dist-packages/pyocd/core/helpers.py", line 241, in session_with_chosen_probe
    return Session(probe, auto_open=auto_open, options=options, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/pyocd/core/session.py", line 179, in __init__
    or Board(self, self.options.get('target_override'))
  File "/usr/local/lib/python3.6/dist-packages/pyocd/board/board.py", line 58, in __init__
    "for how to install additional target support." % self._target_type), exc)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.TargetSupportError: Target type 'stm32l452re' not recognized. Use 'pyocd list --targets' to see currently available target types. See <https://github.com/mbedmicro/pyOCD/blob/master/docs/target_support.md> for how to install additional target support.
```
I am not able to get the positive result as in issue 643.

I think pyocd has read my yaml, since my command line did not specify any target but Traceback showed `Board(self, self.options.get('target_override'))`. Then, I expect pyocd should know about the absolute path of  /workdir/Keil.STM32L4xx_DFP.2.4.0.pack in the yaml file. I had previously confirmed /workdir/Keil.STM32L4xx_DFP.2.4.0.pack has real content inside. I do not know what should be my next step to troubleshoot this.

joectchan, 2020-06-30:
When I use a disco_l475_iot1 board, pyocd has built-in support for that target and I have seen pyocd working with that board.

I make another /workdir/mypyocd.yaml for my disco_l475_iot1 board.
```
probes:
  0670FF323535474B43132338: # Probe's unique ID.
    target_override:  stm32l475xg

# Global options
auto_unlock: false
frequency: 480000
```
I don't need to specify `pack:` in this yaml because `stm32l475xg` is built-in.

I invoked pyocd with this yaml. It ran fine.
```
user@5d1ead943ded:/workdir$ pyocd cmd --config=/workdir/mypyocd.yaml 
0000563:ERROR:linux:Could not get disk devices by id. This could be because your Linux distribution does not use udev, or does not create /dev/disk/by-id symlinks. Please submit an issue to github.com/armmbed/mbed-ls.
Connected to STM32L475xG [Sleeping]: 0670FF323535474B43132338
>>> show map
 Region  Start       End         Size        Access  Sector      Page       
 flash   0x08000000  0x080fffff  0x00100000  rx      0x00000800  0x00000400 
 sram2   0x10000000  0x10007fff  0x00008000  rwx     -           -          
 sram1   0x20000000  0x20017fff  0x00018000  rwx     -           -          
```

So, built-in target works. Using yaml configuration file works. target_override works. However, I haven't seen `pack:` section in yaml configuration working for me yet, for the case of NUCLEO-L452RE board. What shall I try next?

joectchan, 2020-06-30:
By the way, I know how to bring up stml452re using **openocd**. I know NUCLEO-L452RE board can work.

I just need some assistance to get **pyocd** to use the pack I downloaded and work with NUCLEO-L452RE. I feel I am really close. I don't know what is the missing piece of the puzzle to explore next.

```
src/openocd -s tcl -f interface/stlink.cfg -f  target/slow_stm32l4x.cfg
Open On-Chip Debugger 0.10.0+dev-01204-gc60252ac-dirty (2020-04-28-01:35)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 480 kHz
Info : STLINK V2J36M26 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.272000
Info : stm32l4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32l4x.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
target halted due to debug-request, current mode: Thread 
xPSR: 0x21000000 pc: 0x08001988 msp: 0x20027f60
Info : device idcode = 0x20016462 (STM32L45/L46xx - Rev: Y)
Info : flash size = 512kbytes
Info : flash mode : single-bank
target halted due to debug-request, current mode: Thread 
```

anthrax-0, 2020-06-30:
@joectchan Did you try something like this?
`pyocd list --targets --pack /workdir/Keil.STM32L4xx_DFP.2.4.0.pack`
It may be that the pack refers to your target differently, for example, as `stm32l452reix` or `stm32l452retx`.  If such target works - then probably the Nucleo board built-in description has to be updated.

joectchan, 2020-06-30:
The problem I see happens on my in-house development board. It has an stm32l452reix at 3.3V and stm32l452ceux at 1.8V. I can use openocd and STM32CubeIDE to debug them, i.e. I know the hardware works.

I connect this board to pyocd on a raspberry pi running ubuntu. This is my config yaml.
```
probes:
    066DFF333437414257163919: # Probe's unique ID.
    target_override:  stm32l452reix # 3.3V
    #target_override:  stm32l452ceux # 1.8V

# Global options
auto_unlock: false
frequency: 480000
pack:
  - /home/ubuntu/workspace/Keil.STM32L4xx_DFP.2.4.0.pack
```
When I tell pyocd to use my yaml, it thinks I am using generic target cortex_m.
```
ubuntu@ubuntu:~/workspace$ pyocd cmd --config=/home/ubuntu/workspace/mypyocd.yaml 
0001442:WARNING:board:Generic 'cortex_m' target type is selected; is this intentional? You will be able to debug but not program flash. To set the target type use the '--target' argument or 'target_override' option. Use 'pyocd list --targets' to see available targets types.
Connected to CoreSightTarget [Halted]: 066DFF333437414257163919
>>> quit
^C0075569:ERROR:board:link exception during target disconnect:
Traceback (most recent call last):
  File "/home/ubuntu/.local/lib/python3.7/site-packages/pyocd/board/board.py", line 96, in uninit
    self.target.disconnect(resume)
  File "/home/ubuntu/.local/lib/python3.7/site-packages/pyocd/core/coresight_target.py", line 306, in disconnect
    self.dp.power_down_debug()
  File "/home/ubuntu/.local/lib/python3.7/site-packages/pyocd/coresight/dap.py", line 314, in power_down_debug
    r = self.read_reg(DP_CTRL_STAT)
  File "/home/ubuntu/.local/lib/python3.7/site-packages/pyocd/coresight/dap.py", line 260, in read_reg
    return self.read_dp(addr, now)
  File "/home/ubuntu/.local/lib/python3.7/site-packages/pyocd/coresight/dap.py", line 352, in read_dp
    def read_dp(self, addr, now=True):
KeyboardInterrupt
```
When I tell pyocd to use stm32l452reix as target at command line, it works.
```
ubuntu@ubuntu:~/workspace$ pyocd cmd -t stm32l452reix --pack /home/ubuntu/workspace/Keil.STM32L4xx_DFP.2.4.0.pack 
Connected to STM32L452REIx [Halted]: 066DFF333437414257163919
>>> 
```
I copy my yaml to x86 host. Connect my board to x86 host. Then, pyocd on x86 can read my yaml and work fine.
```
pyocd cmd --config=/workdir/pyocd.yaml
Connected to STM32L452REIx [Halted]: 066DFF333437414257163919
>>> show map
 Region      Start       End         Size        Access  Sector      Page       
 Main_Flash  0x08000000  0x0807ffff  0x00080000  rx      0x00000800  0x00000400 
 IRAM2       0x10000000  0x10007fff  0x00008000  rwx     -           -          
 SRAM        0x20000000  0x2001ffff  0x00020000  rwx     -           -          
 IRAM1       0x20000000  0x2001ffff  0x00020000  rwx     -           -          
>>> 
```
I edit my yaml file to use stm32l452ceux as target. Move my probe to that uC. Using the same command line on x86 host, pyocd knows my target is stm32l452ceux.
```
>>> show target
Target:       STM32L452CEUx
DAP IDCODE:   0x2ba01477
>>> halt
Successfully halted device
>>> reg
     r0: 0x294ba161       r6: 0x00000000      r12: 0x20027f13  
     r1: 0x00000010       r7: 0x20027fc8       sp: 0x20027fc4  
     r2: 0x000003e9       r8: 0x00000000       lr: 0x08001753  
     r3: 0x000001ec       r9: 0x00000000       pc: 0x08001712  
     r4: 0x00000000      r10: 0x00000000     xpsr: 0x21000000  
     r5: 0x00000000      r11: 0x00000000  primask: 0x00000000  
>>> 
```
So, there is something wrong parsing yaml when running on raspberry pi, but the same yaml can work on x86 host.

flit, 2020-06-30:
In the config yaml you show above, the `target_override` key is not indented under the probe unique id. It should look like this:

```
probes:
    066DFF333437414257163919: # Probe's unique ID.
        target_override:  stm32l452reix # 3.3V
```

I _don't_ know why it would work on your host machine. Can you confirm whether the indentation is different on the host?

flit, 2020-06-30:
Ah, I know. On the RasPi, unless you have mounted the STLink USB mass storage volume, pyocd won't be able to auto-detect the board/MCU type. This is because the board ID is stored in an HTML file on the STLink volume. I see the same thing when running pyocd on my RasPi with my STM32L475 connected.

On the PC, the STLink USB volume is automatically mounted and thus the board ID can be detected.

joectchan, 2020-07-01:
@flit Thanks. It was indentation. I manual edit the yaml file as I tested multiple boards and hosts. I screwed up the indentation on the raspberry side.

I also realize the `exit` command in pyocd works. It is just very slow that I thought it hung.

Thanks for explaining the mechanism of auto-detecting the board. No wonder that does not work on raspberry nor my in-house dev board. Good to know I did not do anything wrong to stop it from working.

Now, pyocd understands my corrected yaml on the raspberry. I get the same good behavior as on my x86. This accomplishes my goal of working around #813. I can **manually**  give my yaml config to pyocd to connect to my uCs.

My next step is to figure our how to get `west flash` and `west debug` to work. `west flash` help screen shows me openocd and jlink are the supported `zephyr runner`. I thought pyocd and zephyr has a very close relationship. I am surprised pyocd is not listed.

Anyway, I wonder how to give my pyocd.yaml to `west flash` so that it can pass that yaml down to pyocd so that pyocd can flash binary to L452 targets. Does anyone know a good contact person / reference for that? i will start googling again. Thanks for helping with #813  Appreciated.

```
user@87eadf9b5989:/workdir/zephyrproject$ west flash --context -d bare_metal_nucleo_l452re_build-blinky
-- west flash: rebuilding                                                                                                                             
ninja: no work to do.
build configuration:                                                       
  build directory: bare_metal_nucleo_l452re_build-blinky                                                                                              
  board: nucleo_l452re                                                     
  runners.yaml: /workdir/zephyrproject/bare_metal_nucleo_l452re_build-blinky/zephyr/runners.yaml
zephyr runners which support "west flash":
  blackmagicprobe, bossac, canopen, dediprog, dfu-util, esp32,
  hifive1, intel_s1000, jlink, misc-flasher, nios2, nrfjprog,   
  arc-nsim, openocd, pyocd, stm32flash, mdb      
                                     
  Note: not all may work with this board and build directory.  
  Available runners are listed below.                     
available runners in runners.yaml:                                         
  openocd, jlink                                                           
default runner in runners.yaml:
  openocd                                                                  
```

flit, 2020-07-01:
Glad you're making progress!

That's surprising that west doesn't show pyocd. Probably best to create an issue on the [west project](https://github.com/zephyrproject-rtos/west). (Please cc me if you don't mind so I have a link.) They should basically allow pyocd with any device/board as long as there's a way to set the target type.

For the config file, pyocd will automatically use a `pyocd.yaml` (or .yml) file in the working directory from where it was run. If a project directory is specified, it will use that instead of the working dir. So it would be nice if west could pass the zephyr directory as the pyocd project directory.

It would also be nice if west had a way to pass arbitrary arguments to the runner. (It may already, I don't know.)

joectchan, 2020-07-08:
Following up: I did not contact west project because I figured I could manually do everything I wanted to do with pyocd.

I can flash MCUBoot and signed image in slot 0 partition with
```
pyocd flash nucleo_l452re_build-mcuboot/zephyr/zephyr.elf
pyocd flash nucleo_l452re_build-dfu/zephyr/zephyr.signed.hex
``
I can test dfu with
```
sudo dfu-util --alt 1 --download nucleo_l452re_build-blinky/zephyr/zephyr.signed.bin
```
I successfully tested USB DFU sample on nucleo_l452re by using pyocd manually. It is so easy that I do not need any abstraction from west. I didn't bother filing a complain there.

flit, 2020-08-02:
The pack download issues for the STM32 packs have been worked around on the server. Closing this issue.

closed 2020-08-02 by flit