pyocd/pyOCD #815: pyocd is too specific about seeing CMSIS-DAP in USB iInterface string

buzmeg, 2020-02-20:
Hardware may not be able to redundantly set CMSIS-DAP in the USB iInterface field as well as the "product" string.

Some USB driver software wires the iInterface field to be a specific value depending upon the selection of the "protocol" value.  The Linux USB Gadget interface, for example, is guilty of this.

In addition, if you run PyOCD on OS X, it seems to ignore iInterface not having CMSIS-DAP in the iInterface string.

This should probably be made more lenient and consistent across architectures in that seeing CMSIS-DAP either in product *OR* iInterface should be sufficient to identify as a CMSIS-DAP probe.

flit, 2020-03-01:
I'm working on a fix. See flit/pyOCD@98498fe for the CMSIS-DAPv1 fix. It only checks for "CMSIS-DAP" in the interface name string if there is more than one HID interface (which was the original reason I added the check).

CMSIS-DAPv2 is more problematic, as the interface class is vendor specific. So there's no other way to identify a CMSIS-DAPv2 interface versus other interfaces on the same device. The interface name check could be skipped if there is only one interface, but that's such a rare configuration it's probably of little use (most often, probes have CDC interfaces too).

Btw, the reason you don't see pyocd checking the interface name of macOS is that it uses hidapi instead of the pyusb probe driver. Same for Windows (though pywinusb by default instead of hidapi). The kernel HID drivers on these OSes bind to HID interface and cannot be detached (without hacks).

flit, 2020-03-10:
The just-released version 0.25.0 includes the commit mentioned above.

I'll keep this issue open until a similar fix is implemented for CMSIS-DAPv2. Just didn't have time for 0.25.0.

jiangming1399, 2020-03-16:
Hello, I have a device that set iInterface string to "CMSIS-DAP" but not set the product field correctly, it was not recoginze by pyocd in linux.

Currently, pyocd will check the product field first. If "CMSIS-DAP" is not present in the product field. thie device will be "skip". Maybe should pyocd check this two string in the same time?

flit, 2020-03-16:
@jiangming1399 Sorry about that. My plan is to continue refining this test, including fixing your problem. The change in 0.25.0 was a quick change that I was trying to get in before the release—without breaking anything else.

Btw, which probe is this? I like to be aware of as many probes as possible.

buzmeg, 2020-03-26:
Hi, jiangmin1399.  Please do let us know which probe that is.  The more concrete things I can point to that need a "iInterface" string set the more I can push on the Linux kernel guys to implement that specification.

jiangming1399, 2020-03-26:
Hi, the probe I mention above is actually a custom implementation of CMSIS-DAP. The dap feature of that device is not the main feature, so can't it set "CMSIS-DAP" as the "iInterface" string. Sorry for this late information.