pyocd/pyOCD #826: FreeRTOS detection not working in gdbserver. Could also be erroneous when it does.

haneefdm, 2020-03-01:
Sorry, long:

I want to describe the problem first, but I also have a solution. Please read and see if you can **entertain a PR.** I hope it is not a stupid user error.

**Problem**
I think I have everything right, but my FreeRTOS executable fails to be recognized. I looked at the implementation it may not work depending on the version of GCC/another-compiler you are using or optimization levels used.

I see the pyOCD source code for FreeRTOS detection. It is expecting that the symbols defined in the executable will come in the exact order as in the source. For me, the source matches pyOCD's expectation but GCC is rearranging them and I am not aware of any C/ABI spec that says how these variables have to be arranged. Neither do I know of GCC command-line options to control the behavior. So, I am stuck.

GCC version: I do not have the liberty to switch versions, but should it matter?
```text
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
```
If this is a known issue and there is a workaround as to how to manage the compiler/linker options. *Problem solved.*

But spec. wise it is a fragile implementation to expect the FreeRTOS variables to be ordered exactly right. **Once all the addresses/sizes match the experience is wonderful though. Super nice**

**Solutions**
I have a couple. I implemented both and I wanted to present my preferred one, which I tested it many simulated failure situations.

The root cause is that the gdb server protocol does not provide for a symbol size. Just the address via 'qSymbol' and we try to guess by using symbol address offsets. But if the user supplies the elf file (optional) then we can find out the exact sizes. It is already a command-line option.

Very few lines of code changes

**Notes**
I am very familiar with OpenOCD source code as well and I did not want to implement what they did, but it is a fallback option that I also implemented. It works, but not pretty and difficult for the user to get it right.

I am one of two maintainers of the Cortex-Debug extension for VSCode. Also contributed to VSCode generic debugger

flit, 2020-03-01:
Thanks for reporting the issue!

Could you create a pull request with your fixes? A release is planned for next weekend, so it would be nice to include the fix.

haneefdm, 2020-03-01:
I will do a PR as soon as I can. Tomorrow morning at the latest.

Thanks for getting back to me so fast.

flit, 2020-03-01:
Perfect, thanks a lot! Looking forward to the PR. 😀

haneefdm, 2020-03-02:
I made a draft PR #828 Should be ready for review tomorrow. Anyone can look at it in the meantime and give feedback.

PR #828 (2020-03-03): Fix issue #826. More robust detection of FreeRTOS
Fixes #826

We were relying on the compiler/linker placing the symbols in the order they are declared and packed. I have simple executables that are jumbling up the FreeRTOS symbols we are interested in almost random ways. This makes it impossible to detect proper FreeRTOS and also a chance for improperly identifying the FreeRTOS data.

Since the user has the option of supplying an elf file, we can use it to check symbol sizes. We also make sure the address matches what gdb tells us via the `qSymbol` query as a backup because we want to have on [... cut]
closed 2020-03-03 by flit