pyocd/pyOCD #831: SWO never flushes if target doesn't send timestamps.

Dirbaio, 2020-03-03:
Currently, the SWO parser buffers all events until a timestamp is received: https://github.com/mbedmicro/pyOCD/blob/ee63c7b64b313bad65dfa9871b7acb55fdda055d/pyocd/trace/swo.py#L116-L132

In my current project, I'm just sending printf debugging data over ITM port 0, and I want to receive it from Python. However, it seems no timestamps are being sent, so the data is never flushed, and gets buffered forever.

I'm not sure whether I want these timestamps or not, or what's the best way to enable them (from the firmware or from pyOCD?). Either way, shouldn't pyOCD offer an API to read raw SWO data, and process timestamps in a higher-level layer? (SWV, not SWO?)

flit, 2020-03-03:
Yes, there should be an API (and TCP port) for accessing the raw SWO data. Timestamps need to be processed at the event-generation layer, though. There should really be a `SWOReader` class that manages the basic SWO init and reading, and `SWVReader` would be a downstream.

In my setups, I've always done the SWO setup from pyOCD so the configuration can be changed without requiring firmware changes. But either way works.

For this particular bug, we should have a timeout where, if a timestamp hasn't been received in N seconds and/or N packets are buffered, then the queue is flushed. An additional possibility would be for the `SWOParser` (or the object using it) to introspect the target's configuration to see if timestamps are being sent, and disable buffering if not. Probably both methods are required.

aonsquared, 2020-08-02:
Count my vote for adding this feature, as I'm also just sending printf data over ITM port 0.