pyocd/pyOCD #839: imxrt1010evk seen as 1050evk

bfrog, 2020-03-20:
Currently it appears that running pyocd with a imxrt1010evk connected up, it sees a 1050. This seems like it might be problematic, or not?

```
pyocd list                                                                                                                                                                                                                                                        
  #   Probe                                     Unique ID                                         
--------------------------------------------------------------------------------------------------
  0   MIMXRT1050-EVKB [mimxrt1050_hyperflash]   022700001713429800000000000000000000000097969905  
```

flit, 2020-03-21:
The board ID `0227` (first 4 digits of unique ID) is indeed the RT1050 EVK. The RT1010 EVK has a board ID of `0242`. So I'm not sure what's happened here…

@jhqian Could there have been an early firmware release for the RT1010 EVK with the old board ID?

jhqian, 2020-04-28:
yes, @flit, it seems so. @bfrog , please try the latest firmware which can be found at: https://www.nxp.com/design/microcontrollers-developer-resources/ides-for-kinetis-mcus/opensda-serial-and-debug-adapter:OPENSDA

bfrog, 2020-04-28:
@jhqian thanks, will do

flit, 2020-04-28:
Closing the issue as updated firmware was referenced.

closed 2020-04-28 by flit