pyocd/pyOCD #848: Look up the external flash memory region fail

lgl88911, 2020-04-18:
After 0.24, get_region_by_name be remove from MemoryMap, override the flash algorithm for an external flash memory can't work. Is there an alternative method available
```
def will_connect():
    # Look up the external flash memory region.
    extFlash = target.memory_map.get_region_by_name("flexspi")

    # Set the path to an .FLM flash algorithm.
    extFlash.flm = "MIMXRT105x_QuadSPI_4KB_SEC.FLM"
```
FAIL LOG
```
0000771:CRITICAL:__main__:uncaught exception: 'MemoryMap' object has no attribute 'get_region_by_name'
Traceback (most recent call last):
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/__main__.py", line 360, in run
    self._COMMANDS[self._args.cmd](self)
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/__main__.py", line 516, in do_flash
    with session:
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/core/session.py", line 299, in __enter__
    self.open()
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/core/session.py", line 415, in open
    self._board.init()
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/board/board.py", line 80, in init
    self.delegate.will_connect(board=self)
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/core/session.py", line 460, in __call__
    self._fn(**args)
  File "<string>", line 12, in will_connect
AttributeError: 'MemoryMap' object has no attribute 'get_region_by_name'

```

flit, 2020-04-19:
Apologies that you were using the API, and that I forgot to update the example! `get_region_by_name()` was replaced with a more general `get_first_matching_region()` method. It takes keyword arguments that match against any of the region attributes, including `name`.

So you'd use it like this:

```py
def will_connect():
    # Look up the external flash memory region.
    extFlash = target.memory_map.get_first_matching_region(name="flexspi")

    # Set the path to an .FLM flash algorithm.
    extFlash.flm = "MIMXRT105x_QuadSPI_4KB_SEC.FLM"
```

The example code in user_script.md is fixed in #849.

lgl88911, 2020-04-20:
@flit Thanks for your reply, "Look up the external flash memory region" be fixed, but there have other issue,  I am investigating
```
<FlashRegion@0x7f6196e0bc88 name=flexspi type=MemoryType.FLASH start=0x60000000 end=0x7f7fffff length=0x1f800000 access=rx blocksize=0x1000>
/home/frank/work/project/westz/zephyr/boards/arm/mm_swiftio/burner/MIMXRT105x_QuadSPI_4KB_SEC.FLM
[                    ]   0%0003276:ERROR:board:link exception during target disconnect:
Traceback (most recent call last):
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/probe/cmsis_dap_probe.py", line 294, in read_ap_result_callback
    return result()
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 810, in read_reg_cb
    res = transfer.get_result()
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 141, in get_result
    self.daplink.flush()
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 665, in flush
    self._read_packet()
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 894, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 449, in decode_data
    data = self._decode_transfer_data(data)
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 356, in _decode_transfer_data
    self._check_response(data[2])
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 339, in _check_response
    raise DAPAccessIntf.TransferError("No ACK received")
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferError: No ACK received

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/__main__.py", line 523, in do_flash
    file_format=self._args.format)
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/flash/file_programmer.py", line 157, in program
    self._loader.commit()
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/flash/loader.py", line 168, in commit
    keep_unwritten=self._keep_unwritten)
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/flash/builder.py", line 466, in program
    flash_operation = self._sector_erase_program_double_buffer(progress_cb)
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/flash/builder.py", line 863, in _sector_erase_program_double_buffer
    self.flash.erase_sector(sector.addr)
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/flash/flash.py", line 349, in erase_sector
    result = self._call_function_and_wait(self.flash_algo['pc_erase_sector'], address)
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/flash/flash.py", line 585, in _call_function_and_wait
    return self.wait_for_completion()
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/flash/flash.py", line 533, in wait_for_completion
    while self.target.get_state() == Target.State.RUNNING:
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/core/coresight_target.py", line 401, in get_state
    return self.selected_core.get_state()
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/coresight/cortex_m.py", line 974, in get_state
    dhcsr = self.read_memory(CortexM.DHCSR)
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/coresight/cortex_m.py", line 638, in read_memory
    result = self.ap.read_memory(addr, transfer_size, now)
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/coresight/ap.py", line 142, in _locking
    return func(self, *args, **kwargs)
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/coresight/ap.py", line 583, in _read_memory
    result = read_mem_cb()
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/coresight/ap.py", line 565, in read_mem_cb
    res = result_cb()
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/coresight/dap.py", line 416, in read_ap_cb
    result = result_cb()
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/probe/cmsis_dap_probe.py", line 297, in read_ap_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.TransferError: No ACK received

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/probe/cmsis_dap_probe.py", line 294, in read_ap_result_callback
    return result()
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 810, in read_reg_cb
    res = transfer.get_result()
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 141, in get_result
[... 76 lines cut ...]
    result = self._call_function_and_wait(self.flash_algo['pc_erase_sector'], address)
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/flash/flash.py", line 585, in _call_function_and_wait
    return self.wait_for_completion()
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/flash/flash.py", line 533, in wait_for_completion
[... cut ...]
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/coresight/ap.py", line 565, in read_mem_cb
    res = result_cb()
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/coresight/dap.py", line 416, in read_ap_cb
    result = result_cb()
  File "/home/frank/.local/lib/python3.6/site-packages/pyocd/probe/cmsis_dap_probe.py", line 297, in read_ap_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.TransferError: No ACK received

```

closed 2020-04-21 by lgl88911

flit, 2020-04-21:
The "No ACK received" error indicates that the device is not responding to SWD at all. Usually this means it is powered down or in a deep sleep state.

You can try connecting under reset by adding `-Oconnect_mode=under-reset` to the command line.

lgl88911, 2020-04-22:
@flit  Thanks for your suggestion. HEAD of the master branch haven't this issue,  0.25.1.dev46
 test pass.

flit, 2020-04-22:
cool! thanks a lot for letting me know.