pyocd/pyOCD #857: KeyError exception when pausing execution

maxgerhardt, 2020-04-27:
Setup:
* `pyocd --version`: `0.25.0` (latest release)
* Installed in a Python 3.7.7 base
* STM32L476RG based microcontroller board
* Firmware uses mbed-os (RTX5 RTOS)
* ST-Link V2 connected to it, latest firmware version V2J34M7 installed on it 
* GDB server started with `pyocd gdb -t cortex_m -v`
* Interacted with by CLion (GDB Remote Debug target with `arm-none-eabi-gdb` as client and localhost:3333 as target.

At first, the GDB server is started using `pyocd`. Then, CLion is connected to it (using the Debug button). Then the "Pause" button is pressed. `pyocd` crashes immediately with an error 

```
0033899:DEBUG:gdbserver:GDB received query: [b'Xfer', b'threads', b'read', b'', b'0,7fb#d0']
0033899:DEBUG:gdbserver:GDB query b'threads': offset: 0, size: 2043
0033900:DEBUG:memory:range [200002d0:200002d1] is not cacheable
0033902:DEBUG:memory:range [200002d0:200002d1] is not cacheable
0033905:DEBUG:memory:range [200002d0:200002d1] is not cacheable
0033909:DEBUG:memory:range [200002d0:200002d1] is not cacheable
0033911:ERROR:gdbserver:Unhandled exception in handle_message: 75
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pyocd/gdbserver/gdbserver.py", line 379, in handle_message
    reply = handler(msg[msgStart:])
  File "/usr/local/lib/python3.7/dist-packages/pyocd/gdbserver/gdbserver.py", line 853, in handle_query
    resp = self.handle_query_xml(b'threads', int(data[0], 16), int(data[1].split(b'#')[0], 16))
  File "/usr/local/lib/python3.7/dist-packages/pyocd/gdbserver/gdbserver.py", line 1040, in handle_query_xml
    xml = self.get_threads_xml()
  File "/usr/local/lib/python3.7/dist-packages/pyocd/gdbserver/gdbserver.py", line 1145, in get_threads_xml
    t.text = thread.description
  File "/usr/local/lib/python3.7/dist-packages/pyocd/rtos/rtx5.py", line 281, in description
    return "%s; Priority %d" % (self.STATES[self._state], self.priority)
KeyError: 75
```

Full log:

```
max@PC-LT-23:~/eol$ pyocd gdb -t cortex_m -v
0000438:DEBUG:pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=04ca PID=703c). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000446:DEBUG:pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=046a PID=b090). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000450:DEBUG:pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=045e PID=0745). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000451:DEBUG:pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=058f PID=6331). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000465:DEBUG:pyusb_v2_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=04ca PID=703c). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000471:DEBUG:pyusb_v2_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=046a PID=b090). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000476:DEBUG:pyusb_v2_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=045e PID=0745). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000477:DEBUG:pyusb_v2_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=058f PID=6331). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000510:DEBUG:session:Project directory: /home/max/eol
0000511:INFO:board:Target type is cortex_m
0000511:WARNING:board:Generic 'cortex_m' target type is selected; is this intentional? You will be able to debug but not program flash. To set the target type use the '--target' argument or 'target_override' option. Use 'pyocd list --targets' to see available targets types.
0000517:DEBUG:stlink:STLink probe 54FF6E067880565308420567 firmware version: V2J34M7
0000518:DEBUG:sequencer:Running task load_svd
0000518:DEBUG:sequencer:Running task create_flash
0000518:DEBUG:sequencer:Running task pre_connect
0000518:DEBUG:sequencer:Running task dp_init
0000520:DEBUG:dap:Default wire protocol selected; using SWD
0000521:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
0000521:DEBUG:sequencer:Running task power_up
0000522:DEBUG:sequencer:Running task find_aps
0000523:DEBUG:sequencer:Running task create_aps
0000524:DEBUG:sequencer:Running task create_ap.0
0000524:DEBUG:ap:Using accelerated memory access interface
0000525:INFO:ap:AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
0000525:DEBUG:ap:AP#0 default HPROT=3 HNONSEC=0
0000526:DEBUG:ap:AP#0 implemented HPROT=3 HNONSEC=0
0000527:DEBUG:sequencer:Running task find_components
0000527:DEBUG:sequencer:Running task init_ap.0
0000530:INFO:rom_table:AP#0 ROM table #0 @ 0xe00ff000 (designer=020 part=415)
0000533:INFO:rom_table:[0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0000535:INFO:rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
0000537:INFO:rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
0000538:INFO:rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
[... 259 lines cut ...]
    reply = handler(msg[msgStart:])
  File "/usr/local/lib/python3.7/dist-packages/pyocd/gdbserver/gdbserver.py", line 853, in handle_query
    resp = self.handle_query_xml(b'threads', int(data[0], 16), int(data[1].split(b'#')[0], 16))
  File "/usr/local/lib/python3.7/dist-packages/pyocd/gdbserver/gdbserver.py", line 1040, in handle_query_xml
    xml = self.get_threads_xml()
  File "/usr/local/lib/python3.7/dist-packages/pyocd/gdbserver/gdbserver.py", line 1145, in get_threads_xml
    t.text = thread.description
  File "/usr/local/lib/python3.7/dist-packages/pyocd/rtos/rtx5.py", line 281, in description
    return "%s; Priority %d" % (self.STATES[self._state], self.priority)
KeyError: 75
0033913:DEBUG:gdbserver:GDB breakpoint b'z'0 @ 8007c1a
0033914:DEBUG:manager:remove bkpt at 0x8007c1a
0033915:DEBUG:gdbserver:GDB breakpoint b'z'0 @ 804075c
0033915:DEBUG:manager:remove bkpt at 0x804075c
0038320:INFO:gdbserver:Client detached
0038320:INFO:gdbserver:Client disconnected!
0038321:DEBUG:gdbserver:GDB server cleaning up
0038324:DEBUG:packet_io:GDB packet thread stopping
0038616:INFO:server:Semihost server stopped
0038616:DEBUG:session:uninit session <pyocd.core.session.Session object at 0x7f100b7acbd0>
0038616:DEBUG:board:uninit board <pyocd.board.board.Board object at 0x7f100adf6650>
0038620:DEBUG:cortex_m:resuming core 0
0038621:DEBUG:manager:added=[] removed=[]
0038621:DEBUG:manager:bps after flush={}
```

flit, 2020-04-27:
Thanks for the very detailed bug report! Looks like this should be a straightforward fix.

(Btw, the reported "Unable to print the message and arguments" errors have been fixed on master.)

maxgerhardt, 2020-04-27:
Hm actually this might be an invalid bug report. I've cloned the latest version from the repository and retried the same procedure, now the error is **not** occurring. Either I can't reproduce an edge case here or the error has already been fixed in master. Maybe some safeguards were added to make sure `self.STATES[self._state]` exists?  

If you can't reproduce it and the good looks good in current master, I'm closing this.

```
$ pyocd gdb -t cortex_m -v
0002578:DEBUG:pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=04ca PID=703c). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0002590:DEBUG:pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=046a PID=b090). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0002594:DEBUG:pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=045e PID=0745). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0002594:DEBUG:pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=058f PID=6331). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0002612:DEBUG:pyusb_v2_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=04ca PID=703c). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0002630:DEBUG:pyusb_v2_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=046a PID=b090). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0002635:DEBUG:pyusb_v2_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=045e PID=0745). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0002636:DEBUG:pyusb_v2_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=058f PID=6331). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0002823:DEBUG:session:Project directory: /home/max/eol
0002824:INFO:board:Target type is cortex_m
0002824:WARNING:board:Generic 'cortex_m' target type is selected; is this intentional? You will be able to debug but not program flash. To set the target type use the '--target' argument or 'target_override' option. Use 'pyocd list --targets' to see available targets types.
0002830:DEBUG:stlink:STLink probe 54FF6E067880565308420567 firmware version: V2J34M7
0002831:DEBUG:sequencer:Running task load_svd
0002831:DEBUG:sequencer:Running task create_flash
0002831:DEBUG:sequencer:Running task pre_connect
0002831:DEBUG:sequencer:Running task dp_init
0002831:DEBUG:sequencer:Running task connect
0002833:DEBUG:dap:Default wire protocol selected; using SWD
0002834:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
0002834:DEBUG:sequencer:Running task clear_sticky_err
0002834:DEBUG:sequencer:Running task power_up_debug
0002836:DEBUG:sequencer:Running task create_discoverer
0002836:DEBUG:sequencer:Running task discovery
0002836:DEBUG:sequencer:Running task find_aps
0002837:DEBUG:sequencer:Running task create_aps
0002837:DEBUG:sequencer:Running task create_ap.0
0002839:DEBUG:ap:Using accelerated memory access interface
0002839:INFO:ap:AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
0002842:DEBUG:ap:AP#0 default HPROT=3 HNONSEC=0
0002843:DEBUG:ap:AP#0 implemented HPROT=3 HNONSEC=0
0002844:DEBUG:sequencer:Running task find_components
0002845:DEBUG:sequencer:Running task init_ap.0
0002852:INFO:rom_table:AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=020 part=415)
0002855:INFO:rom_table:[0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0002858:INFO:rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
0002860:INFO:rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
0002863:INFO:rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
0002865:INFO:rom_table:[4]<e0040000:TPIU-M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=ca1:0:0>
0002868:INFO:rom_table:[5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
0002868:DEBUG:sequencer:Running task create_cores
0002869:DEBUG:discovery:Creating SCS-M4 component
0002873:INFO:cortex_m:CPU core #0 is Cortex-M4 r0p1
0002879:INFO:cortex_m:FPU present: FPv4-SP
0002881:DEBUG:sequencer:Running task create_components
0002881:DEBUG:discovery:Creating DWT component
0002883:INFO:dwt:4 hardware watchpoints
0002886:DEBUG:discovery:Creating FPB component
0002887:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0002888:DEBUG:fpb:fpb has been disabled
0002898:DEBUG:discovery:Creating ITM component
0002902:DEBUG:discovery:Creating TPIU-M4 component
0002903:DEBUG:sequencer:Running task check_for_cores
0002903:DEBUG:sequencer:Running task halt_on_connect
0002904:DEBUG:cortex_m:halting core 0
0002908:DEBUG:sequencer:Running task post_connect
0002909:DEBUG:sequencer:Running task post_connect_hook
0002909:DEBUG:sequencer:Running task notify
[... cut ...]
0009804:DEBUG:gdbserver:GDB query b'threads': offset: 0, size: 2043
0009810:DEBUG:memory:range [200002d0:200002d1] is not cacheable
0009847:DEBUG:gdbserver:GDB breakpoint b'z'0 @ 8007c1a
0009847:DEBUG:manager:remove bkpt at 0x8007c1a
0009848:DEBUG:gdbserver:GDB breakpoint b'z'0 @ 804075c
0009848:DEBUG:manager:remove bkpt at 0x804075c
0009853:DEBUG:gdbserver:GDB breakpoint b'z'0 @ 804076c
0009853:DEBUG:manager:remove bkpt at 0x804076c
0009855:DEBUG:memory:range [802591c:8025920] is not cacheable
```

PR #858 (2020-04-27): Handle invalid RTX5 thread state in description property.
Fixes #857.
commit 719afa80 (2020-04-27): Handle invalid RTX5 thread state in description property.

Fixes #857.
closed 2020-04-27 by flit