pyocd/pyOCD #861: MIMXRT1050_EVK "No ACK received" error on disconnect

juhhov, 2020-04-29:
```
Traceback (most recent call last):
  File "/home/juhhov01/projects/pyOCD/py3/lib/python3.6/site-packages/pyocd/board/board.py", line 96, in uninit
    self.target.disconnect(resume)
  File "/home/juhhov01/projects/pyOCD/py3/lib/python3.6/site-packages/pyocd/core/coresight_target.py", line 306, in disconnect
    self.dp.power_down_debug()
  File "/home/juhhov01/projects/pyOCD/py3/lib/python3.6/site-packages/pyocd/coresight/dap.py", line 303, in power_down_debug
    r = self.read_reg(DP_CTRL_STAT)
  File "/home/juhhov01/projects/pyOCD/py3/lib/python3.6/site-packages/pyocd/coresight/dap.py", line 260, in read_reg
    return self.read_dp(addr, now)
  File "/home/juhhov01/projects/pyOCD/py3/lib/python3.6/site-packages/pyocd/coresight/dap.py", line 374, in read_dp
    return read_dp_cb()
  File "/home/juhhov01/projects/pyOCD/py3/lib/python3.6/site-packages/pyocd/coresight/dap.py", line 366, in read_dp_cb
    result = result_cb()
  File "/home/juhhov01/projects/pyOCD/py3/lib/python3.6/site-packages/pyocd/probe/cmsis_dap_probe.py", line 265, in read_dp_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.TransferError: No ACK received
```
Seems that the erase itself succeeds and the disconnect raises this.

pyOCD @ mbedmicro/pyOCD@860b61ced6948dca4cb0cfacf94ed8825d047271
[daplink.txt](https://github.com/mbedmicro/pyOCD/files/4549891/daplink.txt)
[full_trace.txt](https://github.com/mbedmicro/pyOCD/files/4549892/full_trace.txt)

cc: @mmahadevan108 @maclobdell

maclobdell, 2020-05-04:
I spoke with Chris Reed and he believes this is a known condition that simply can be ignored. So if you have a python script that is issuing pyocd commands, perhaps you can trap this error and ignore it.  
@flit - please confirm

juhhov, 2020-05-05:
It doesn't raise exception outside of pyOCD code and also pyOCD cli returns 0, so this is minor indeed. @flit please close this if you think this is working as intended.