pyocd/pyOCD #862: MIMXRT1050_EVK goes to "No cores were disovered!" state

juhhov, 2020-04-29:
After few flash rounds the board goes to "No cores were discovered!" state and has to be restored with DIP switch altering and erase.

Steps:
1. ``` pyocd flash -Oconnect_mode=under-reset -vvv -u 02270b0341114e45005d300ac207003692d1000097969900 -t mimxrt1050_hyperflash  printer_MIMXRT1050_EVK.hex```
2. ```pyocd flash -Oconnect_mode=under-reset -vvv -u 02270b0341114e45005d300ac207003692d1000097969900 -t mimxrt1050_hyperflash  payload_MIMXRT1050_EVK.hex```
3. ```pyocd flash -Oconnect_mode=under-reset -vvv -u 02270b0341114e45005d300ac207003692d1000097969900 -t mimxrt1050_hyperflash  payload_null_MIMXRT1050_EVK.hex```
4. "No cores were discovered!" occurs and board is unusable without manual restore.

Using pyoCD version mbedmicro/pyOCD@fdc69fac102880de4b845bcedf540dbd0c43d7b5.
[binaries.zip](https://github.com/mbedmicro/pyOCD/files/4550083/binaries.zip)
[daplink.txt](https://github.com/mbedmicro/pyOCD/files/4550084/daplink.txt)
[no_cores_were_discovered_trace.txt](https://github.com/mbedmicro/pyOCD/files/4550085/no_cores_were_discovered_trace.txt)

cc: @mmahadevan108 @maclobdell

maclobdell, 2020-04-30:
I am able to recreate this problem (or at least a similar one).  I have a workaround to propose. 

I noticed that the application in the RAAS acceptance test (private repo) is using **wait(0.2)** in the main thread. Please note that wait() will block the thread and put the device to sleep if no other threads are running.  There is a known issue on this target that if the application goes to sleep too quickly upon startup, then the debugger cannot connect.

**Workaround Proposal**
  Please change **wait(0.2)** to **wait_us(200*1000)** in the main loop of the acceptance test.  **wait_us** will block and not let the board go to sleep. 

**Real fix needed** - I think NXP should investigate a fix to pyOCD that will ensure the debugger can always connect - even if the board is asleep. In DAPLink, this was accomplished by issuing a hard reset.  need to double check status of https://github.com/ARMmbed/mbed-os/issues/12376.

Steps I used to recreate this problem.  I created several binaries from the blinky application. The first few binaries use **wait_us()**.  The last binary uses **wait()**.  Everything works fine, until I flash the last binary. Then I see the same problem.

maclobdell, 2020-05-01:
**Alternative Possible Workaround**
You can prevent an Mbed OS application from going to deep sleep by defining the MBED_DEBUG macro.  
https://github.com/ARMmbed/mbed-os/blob/af4c8a94f3913ceda9069ec79cba81a793d695fe/platform/source/mbed_power_mgmt.c#L233

flit, 2020-05-03:
@maclobdell Please try using the `connect_mode` option if you think a pre-reset or connecting under reset would help.

RomanSaveljev, 2020-05-04:
Defect originator uses `--connect_mode=under-reset`

maclobdell, 2020-05-04:
@flit - as Roman mentions, we tried that and it doesn't work. NXP has said that reconnecting to a device that is asleep requires a hard reset. so perhaps connecting under reset is not enough.

maclobdell, 2020-05-04:
Here are the files I used to test this. 
[i.mxrt1050_pyocd_flashing_debug.zip](https://github.com/mbedmicro/pyOCD/files/4575472/i.mxrt1050_pyocd_flashing_debug.zip)

flit, 2020-05-06:
I don't know if this is good or bad, but I tried both the binaries attached to the original description of this ticket, as well as Mac's from a couple days ago. In all cases, I'm able to successfully connect, both via commander as well as to program flash again. This was with version 0.26.0.

juhhov, 2020-05-07:
Retested with 0.26.0 using the steps in description. On first round step 3 the board fails and the error reproduces.

maclobdell, 2020-05-15:
@flit - did you try to flash the binaries in the same sequence? Also, can you share the revision of the board you have, or any details that might explain why you see different behavior?

flit, 2020-05-17:
Sorry, haven't had time yet. Will try to get to it soon.

flit, 2020-05-20:
Ok! I'm able to reproduce it! 😄

maclobdell, 2020-05-23:
Based on further communication, we believe this issue is related to letting the system go to sleep too early in the main loop. It needs more time before going to sleep to allow the debugger to connect.

maclobdell, 2020-06-29:
This might provide additional protection. Needs to be tested. https://github.com/mbedmicro/pyOCD/pull/882