pyocd/pyOCD #863: DISCO-F746NG - "No cores were discovered!"

georgejourneaux, 2020-04-30:
Sorry if this is the wrong place for this, first experience with pyOCD. 
Using it through Mbed Studio on the DISCO-F746NG and I get the following:
```
0000379:INFO:board:Target type is stm32f746ng
0000385:INFO:coresight_target:Asserting reset prior to connect
0000388:INFO:dap:DP IDR = 0x5ba02477 (v2 rev5)
0000397:INFO:ap:AP#0 IDR = 0x74770001 (AHB-AP var0 rev7)
0000405:INFO:rom_table:AP#0 ROM table #0 @ 0xe00fd000 (designer=020 part=449)
0000410:INFO:rom_table:[0]
0000410:INFO:rom_table:  AP#0 ROM table #1 @ 0xe00fe000 (designer=43b part=4c8)
0000414:INFO:rom_table:  [0]
0000414:INFO:rom_table:    AP#0 ROM table #2 @ 0xe00ff000 (designer=43b part=4c7)
0000420:ERROR:ap:Transfer error while reading AP#0 ROM table: SWD/JTAG Transfer Fault @ 0xe000efbc-0xe000f043
0000421:CRITICAL:__main__:No cores were discovered!
Traceback (most recent call last):
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\__main__.py", line 343, in run
    self._COMMANDS[self._args.cmd](self)
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\__main__.py", line 571, in do_gdbserver
    with session:
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\core\session.py", line 287, in __enter__
    self.open()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\core\session.py", line 367, in open
    self._board.init()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\board\board.py", line 83, in init
    self.target.init()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\core\coresight_target.py", line 164, in init
    seq.invoke()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\utility\sequencer.py", line 208, in invoke
    resultSequence = call()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\core\coresight_target.py", line 298, in check_for_cores
    raise exceptions.DebugError("No cores were discovered!")
DebugError: No cores were discovered!
```
Any ideas?

Thanks!

jhd25, 2020-06-14:
Exactly the same error on a stm32f746zg

> 0000571:INFO:board:Target type is stm32f746zg
> 0000667:INFO:coresight_target:Asserting reset prior to connect
> 0000728:INFO:dap:DP IDR = 0x5ba02477 (v2 rev5)
> 0000885:INFO:ap:AP#0 IDR = 0x74770001 (AHB-AP var0 rev7)
> 0000897:INFO:rom_table:AP#0 ROM table #0 @ 0xe00fd000 (designer=020 part=449)
> 0000899:INFO:rom_table:[0]<e00fe000:ROM class=1 designer=43b part=4c8>
> 0000899:INFO:rom_table:  AP#0 ROM table #1 @ 0xe00fe000 (designer=43b part=4c8)
> 0000902:INFO:rom_table:  [0]<e00ff000:ROM class=1 designer=43b part=4c7> []
> 0000903:INFO:rom_table:    AP#0 ROM table #2 @ 0xe00ff000 (designer=43b part=4c7) []
> 0000906:ERROR:ap:Transfer error while reading AP#0 ROM table: SWD/JTAG Transfer Fault @ 0xe000efbc-0xe000f043 []
> 0000907:CRITICAL:__main__:No cores were discovered!

flit, 2020-06-14:
This is a known issue with the STM32F7 series. This series (and the H7 series) requires some special handling to recover from low power modes that other STM32 device don't. @bartek-w created a solution for the F767 that is merged into master. This might be usable for the F746.

Suggestions:
- Try using the current HEAD of the master branch (commit d59bd88 as of this writing) and set the target type to 'stm32f767zi' (`-t` or `--target` argument).
- Make sure you are using a debug build with Mbed OS so the application doesn't put the chip to sleep.
- For debug builds of your application, configure the `DBGMCU_CR` register immediately after it starts running.

jhd25, 2020-06-14:
Using -t stm32f767zi on my NUCLEO-F746ZG : erase , flash and gdbserver all seem to work - not tested extensively.
This error:
0000609:ERROR:target_STM32F767xx:Dual bank configuration detected [FLASH_OPTCR=0xc0ffaafd]. Currently only single bank configuration is fully supported!
is raised but operation continues.
Thanks !!

bartek-w, 2020-06-14:
You man take a look into https://www.st.com/resource/en/application_note/dm00266999-stm32f7-series-flash-memory-dual-bank-mode-stmicroelectronics.pdf

```
First configuration: Project configured in the single bank mode and nDBANK = 0
As the project is configured in the single bank mode, the user must erase 1 sector
(32 Kbytes), but with nDBANK = 0 the Flash interface is configured in the dual bank
mode, therefore 1 sector = 16 Kbytes. The result is that only 16 Kbytes are erased and
4 Kbytes are missed.
```

So overall be careful - probably you are not erasing enough sectors before programming. You can solve it in one of 3 ways:

- always do chip erase instead of sector erase
- change (with other programming software, e.g. STM32CubeProgrammer) configuration bits to single bank,
- Add dual bank programming support to pyocd :)

stephenstarkie, 2021-03-24:
Any news on this - it seems to have been around for a long time and is quite painful!