pyocd/pyOCD #865: [pyocd 0.26] pyocd gdbserver exceptions

Hoohaha, 2020-05-07:
**tested boards**
1. twrk22f12, daplink version 0247
2. frdmk66f, daplink version 0243

**description**
1. At first, run command `pyocd erase -c -u 0231000026294e450024000e58a900543ee1000097969900` to perform a chip erase.
2. Exceptions will occured on gdb command "load" is sent.

*used gdb commands*
```
target remote localhost: {gdbport}
load
monitor reset
q
```

**note**
Verfied above steps works very well on version 0.14

**pyocd gdbserver**
```
$ pyocd gdbserver --trust-crc --uid 0231000026294e450024000e58a900543ee1000097969900 --port 7100 
0000675:WARNING:common:STLink and CMSIS-DAPv2 probes are not supported because no libusb library was found.
0001093:INFO:board:Target type is k22f
0001108:INFO:coresight_target:Performing connect pre-reset
0001342:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
0001367:INFO:ap:AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
0001386:INFO:ap:AP#1 IDR = 0x001c0000 (proprietary)
0003197:WARNING:target_kinetis:K22F in secure state: will try to unlock via mass erase
0003426:WARNING:target_kinetis:K22F secure state: unlocked successfully
0003440:INFO:rom_table:AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=43b part=4c4)
0003451:INFO:rom_table:[0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0003459:INFO:rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
0003467:INFO:rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
0003476:INFO:rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
0003484:INFO:rom_table:[4]<e0040000:TPIU-M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=ca0:0:0>
0003488:INFO:cortex_m:CPU core #0 is Cortex-M4 r0p1
0003501:INFO:dwt:4 hardware watchpoints
0003506:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0003538:INFO:server:Semihost server started on port 4444 (core 0)
0003539:INFO:gdbserver:GDB server started on port 7100 (core 0)
0004069:INFO:gdbserver:One client connected!
0004102:ERROR:gdbserver:Unhandled exception in handle_message: type object 'CortexM' has no attribute 'DFSR_PMU'
Traceback (most recent call last):
  File "c:\python27\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 381, in handle_message
    reply = handler()
  File "c:\python27\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 516, in stop_reason_query
    return self.create_rsp_packet(self.get_t_response())
  File "c:\python27\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 1116, in get_t_response
    response = self.target_facade.get_t_response(forceSignal)
  File "c:\python27\lib\site-packages\pyocd\gdbserver\context_facade.py", line 138, in get_t_response
    response = six.b('T' + conversion.byte_to_hex2(self.get_signal_value()))
  File "c:\python27\lib\site-packages\pyocd\gdbserver\context_facade.py", line 152, in get_signal_value
    if self._context.core.is_vector_catch():
  File "c:\python27\lib\site-packages\pyocd\coresight\cortex_m.py", line 1335, in is_vector_catch
    return self.get_halt_reason() == Target.HaltReason.VECTOR_CATCH
  File "c:\python27\lib\site-packages\pyocd\coresight\cortex_m.py", line 1349, in get_halt_reason
    elif dfsr & CortexM.DFSR_PMU:
AttributeError: type object 'CortexM' has no attribute 'DFSR_PMU'
0008320:ERROR:gdbserver:Unhandled exception in handle_message: init error: 536870913
Traceback (most recent call last):
  File "c:\python27\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 383, in handle_message
    reply = handler(msg[msgStart:])
  File "c:\python27\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 602, in v_command
    return self.flash_op(data)
  File "c:\python27\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 718, in flash_op
    self.flash_loader.commit()
  File "c:\python27\lib\site-packages\pyocd\flash\loader.py", line 168, in commit
    keep_unwritten=self._keep_unwritten)
  File "c:\python27\lib\site-packages\pyocd\flash\builder.py", line 417, in program
    self._build_sectors_and_pages(keep_unwritten)
  File "c:\python27\lib\site-packages\pyocd\flash\builder.py", line 315, in _build_sectors_and_pages
    fill_end_of_page_gap()
  File "c:\python27\lib\site-packages\pyocd\flash\builder.py", line 259, in fill_end_of_page_gap
    self._enable_read_access()
  File "c:\python27\lib\site-packages\pyocd\flash\builder.py", line 221, in _enable_read_access
    self.flash.init(self.flash.Operation.VERIFY)
  File "c:\python27\lib\site-packages\pyocd\flash\flash.py", line 250, in init
    raise FlashFailure('init error: %i' % result, result_code=result)
FlashFailure: init error: 536870913
0008342:INFO:gdbserver:Client detached
0008342:INFO:gdbserver:Client disconnected!
0008546:INFO:server:Semihost server stopped
```

flit, 2020-05-09:
Thanks @Hoohaha for the detailed bug report.

Hoohaha, 2020-05-11:
@flit Another exception `FlashFailure: init error: 536870913`, what does that mean?

Hoohaha, 2020-05-15:
seems add under-reset it can improve the stability for kinetis.

flit, 2020-05-16:
Fyi, I'm able to reproduce the flash failure. It's failing when trying to init the flash algo in "verify" mode.

flit, 2020-05-16:
Tried with the K22FN512VDC12 CMSIS-Pack and got a similar failure but different error code.

The built-in flash algo reports error 0x20000001 (536870913).
The pack flash algo reports error 0x1fff0201 (536805889).

Neither of these make sense to me…

flit, 2020-05-16:
@Hoohaha I ended up just working around the verify init failures. If it gets an exception during a verify init, it will retry to init in erase mode. That works at least in the case here.

closed 2020-05-17 by flit

Hoohaha, 2020-05-19:
Thanks flit!

PR #873 (2020-05-17): Several flash related fixes
1. Work around cases where initing the flash algo in VERIFY mode fails by attempting to reinit in ERASE mode.
2. If the gdbserver gets an exception while programming flash, it deletes the `FlashLoader` instance it was using. This prevents conflicts when trying to add data to the stale `FlashLoader` if another attempt is made to program flash.
3. Moved the `create_flash` init task to the end of the init sequence, just prior to the post-connect notification. This allows the memory map to be modified or replaced during initialization without losing existing flash insta [... cut]