pyocd/pyOCD #868: Unable to connect to Cypress PSOC 62

marcuschangarm, 2020-05-11:
I'm getting the error below when I try to connect to CY8CPROTO-062-4343W.

pyOCD: 0.26.0
Python: 3.6.9
OS: Ubuntu 18.04.4 LTS

```

$ pyocd gdb -t CY8C6xxA
0001233:INFO:board:Target type is cy8c6xxa
0001285:INFO:dap:DP IDR = 0x6ba02477 (v2 rev6)
0001299:INFO:ap:AP#0 IDR = 0x84770001 (AHB-AP var0 rev8)
0001310:INFO:ap:AP#1 IDR = 0x84770001 (AHB-AP var0 rev8)
0001323:INFO:ap:AP#2 IDR = 0x24770011 (AHB-AP var1 rev2)
0001337:INFO:rom_table:AP#0 Class 0x1 ROM table #0 @ 0xf1000000 (designer=034 part=102)
0001352:INFO:rom_table:AP#1 Class 0x1 ROM table #0 @ 0xf0000000 (designer=034 part=102)
0001360:INFO:rom_table:[0]<e00ff000:ROM class=1 designer=43b part=4c0>
0001360:INFO:rom_table:  AP#1 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b part=4c0)
0001368:INFO:rom_table:  [0]<e000e000:SCS-M0+ class=14 designer=43b part=008>
0001373:INFO:rom_table:  [1]<e0001000:DWT-M0+ class=14 designer=43b part=00a>
0001379:INFO:rom_table:  [2]<e0002000:BPU class=14 designer=43b part=00b>
0001384:INFO:rom_table:[1]<f0002000:CTI class=9 designer=43b part=9a6 devtype=14 archid=1a14 devid=1040800:0:0>
0001388:INFO:rom_table:[2]<f0003000:MTB-M0+ class=9 designer=43b part=932 devtype=31 archid=0a31 devid=0:0:0>
0001397:INFO:rom_table:AP#2 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=034 part=102)
0001404:INFO:rom_table:[0]<e0080000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=40800:0:0>
0001410:INFO:rom_table:[3]<e008e000:TPIU-M3 class=9 designer=43b part=923 devtype=11 archid=0000 devid=ca1:0:0>
0001416:INFO:rom_table:[4]<e007f000:ROM class=1 designer=034 part=102>
0001416:INFO:rom_table:  AP#2 Class 0x1 ROM table #1 @ 0xe007f000 (designer=034 part=102)
0001423:INFO:rom_table:  [0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0001428:INFO:rom_table:  [1]<e0001000:DWT class=14 designer=43b part=002>
0001434:INFO:rom_table:  [2]<e0002000:FPB class=14 designer=43b part=003>
0001439:INFO:rom_table:  [3]<e0000000:ITM class=14 designer=43b part=001>
0001445:INFO:rom_table:  [4]<e0042000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=40800:0:0>
0001450:INFO:rom_table:  [5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
0001453:INFO:cortex_m:CPU core #0 is Cortex-M0+ r0p1
0001457:INFO:cortex_m:CPU core #1 is Cortex-M4 r0p1
0001466:INFO:cortex_m:FPU present: FPv4-SP
0001472:INFO:dwt:2 hardware watchpoints
0001477:INFO:fpb:4 hardware breakpoints, 0 literal comparators
0001487:INFO:dwt:4 hardware watchpoints
0001490:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0001518:INFO:server:Semihost server started on port 4444 (core 0)
0001518:INFO:gdbserver:GDB server started on port 3333 (core 0)
Exception in thread gdb-server-core0:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/gdbserver/gdbserver.py", line 272, in run
    self.target.halt()
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/coresight/cortex_m.py", line 683, in halt
    self.write_memory(CortexM.DHCSR, CortexM.DBGKEY | CortexM.C_DEBUGEN | CortexM.C_HALT)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/coresight/cortex_m.py", line 640, in write_memory
    self.ap.write_memory(addr, value, transfer_size)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/coresight/ap.py", line 142, in _locking
    return func(self, *args, **kwargs)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/coresight/ap.py", line 540, in _write_memory
    self.write_reg(MEM_AP_TAR, addr)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/coresight/ap.py", line 142, in _locking
    return func(self, *args, **kwargs)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/coresight/ap.py", line 512, in write_reg
    super(MEM_AP, self).write_reg(addr, data)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/coresight/ap.py", line 142, in _locking
    return func(self, *args, **kwargs)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/coresight/ap.py", line 241, in write_reg
    self.dp.write_ap((self.ap_num << APSEL_SHIFT) | addr, data)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/coresight/dap.py", line 402, in write_ap
    self.probe.write_ap(addr, data)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/probe/cmsis_dap_probe.py", line 325, in write_ap
    self._select_ap(addr)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/probe/cmsis_dap_probe.py", line 295, in _select_ap
    self.write_dp(self.DP_SELECT, select)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/probe/cmsis_dap_probe.py", line 280, in write_dp
    self._link.write_reg(reg_id, data)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 786, in write_reg
    self._write(dap_index, 1, request, [value])
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 964, in _write
    size = cmd.get_request_space(size_to_transfer, transfer_request, dap_index)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 217, in get_request_space
    assert self._data_encoded is False
AssertionError

0001525:INFO:server:Semihost server started on port 4445 (core 1)
0001526:INFO:gdbserver:GDB server started on port 3334 (core 1)
Exception in thread gdb-server-core1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/gdbserver/gdbserver.py", line 272, in run
    self.target.halt()
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/coresight/cortex_m.py", line 683, in halt
    self.write_memory(CortexM.DHCSR, CortexM.DBGKEY | CortexM.C_DEBUGEN | CortexM.C_HALT)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/coresight/cortex_m.py", line 640, in write_memory
    self.ap.write_memory(addr, value, transfer_size)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/coresight/ap.py", line 142, in _locking
    return func(self, *args, **kwargs)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/coresight/ap.py", line 540, in _write_memory
    self.write_reg(MEM_AP_TAR, addr)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/coresight/ap.py", line 142, in _locking
    return func(self, *args, **kwargs)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/coresight/ap.py", line 512, in write_reg
    super(MEM_AP, self).write_reg(addr, data)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/coresight/ap.py", line 142, in _locking
    return func(self, *args, **kwargs)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/coresight/ap.py", line 241, in write_reg
    self.dp.write_ap((self.ap_num << APSEL_SHIFT) | addr, data)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/coresight/dap.py", line 402, in write_ap
    self.probe.write_ap(addr, data)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/probe/cmsis_dap_probe.py", line 325, in write_ap
    self._select_ap(addr)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/probe/cmsis_dap_probe.py", line 295, in _select_ap
    self.write_dp(self.DP_SELECT, select)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/probe/cmsis_dap_probe.py", line 280, in write_dp
    self._link.write_reg(reg_id, data)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 786, in write_reg
    self._write(dap_index, 1, request, [value])
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 964, in _write
    size = cmd.get_request_space(size_to_transfer, transfer_request, dap_index)
  File "/home/marcus/python/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 217, in get_request_space
    assert self._data_encoded is False
AssertionError
```

maclobdell, 2020-05-11:
I have Python 3.7.4 , pyOCD 0.26 and I see the same thing. 

Could it be something related to the dual-core?

Quincy-Zh, 2020-05-15:
I'm having similar issues.
pyOCD: 0.26.0
Python: 3.7.3
OS: Windows 10 1909
Target: LPC55S69
Probe: DAPLink like

pyocd output:
```
pyocd gdb -t lpc55s69
0000470:INFO:board:Target type is lpc55s69
0000492:INFO:dap:DP IDR = 0x6ba02477 (v2 rev6)
0000502:INFO:ap:AP#0 IDR = 0x84770001 (AHB-AP var0 rev8)
0000509:INFO:ap:AP#1 IDR = 0x84770001 (AHB-AP var0 rev8)
0000518:INFO:ap:AP#2 IDR = 0x002a0000 (proprietary)
0000524:INFO:rom_table:AP#0 Class 0x1 ROM table #0 @ 0xe00fe000 (designer=015 part=000)
0000528:INFO:rom_table:[0]<e00ff000:ROM class=1 designer=43b part=4c9>
0000528:INFO:rom_table:  AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b part=4c9)
0000533:INFO:rom_table:  [0]<e000e000:SCS-M33 class=9 designer=43b part=d21 devtype=00 archid=2a04 devid=0:0:0>
0000554:INFO:rom_table:  [1]<e0001000:DWT class=9 designer=43b part=d21 devtype=00 archid=1a02 devid=0:0:0>
0000559:INFO:rom_table:  [2]<e0002000:BPU class=9 designer=43b part=d21 devtype=00 archid=1a03 devid=0:0:0>
0000563:INFO:rom_table:  [3]<e0000000:ITM class=9 designer=43b part=d21 devtype=43 archid=1a01 devid=0:0:0>
0000567:INFO:rom_table:[1]<e0040000:TPIU-M33 class=9 designer=43b part=d21 devtype=11 archid=0000 devid=ca0:0:0>
0000617:INFO:rom_table:AP#1 Class 0x1 ROM table #0 @ 0xe00fe000 (designer=015 part=000)
0000660:INFO:rom_table:[0]<e00ff000:ROM class=1 designer=43b part=4c9>
0000666:INFO:rom_table:  AP#1 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b part=4c9)
0000680:INFO:rom_table:  [0]<e000e000:SCS-M33 class=9 designer=43b part=d21 devtype=00 archid=2a04 devid=0:0:0>
0000682:INFO:rom_table:  [1]<e0001000:DWT class=9 designer=43b part=d21 devtype=00 archid=1a02 devid=0:0:0>
0000684:INFO:rom_table:  [2]<e0002000:BPU class=9 designer=43b part=d21 devtype=00 archid=1a03 devid=0:0:0>
0000686:INFO:rom_table:  [3]<e0000000:ITM class=9 designer=43b part=d21 devtype=43 archid=1a01 devid=0:0:0>
0000688:INFO:rom_table:[1]<e0040000:TPIU-M33 class=9 designer=43b part=d21 devtype=11 archid=0000 devid=ca0:0:0>
0000689:INFO:cortex_m_v8m:CPU core #0 is Cortex-M33 r0p3 (security ext present)
0000691:INFO:cortex_m:FPU present: FPv5-SP
0000694:INFO:cortex_m_v8m:CPU core #1 is Cortex-M33 r0p3
0000698:INFO:dwt:4 hardware watchpoints
0000699:INFO:fpb:8 hardware breakpoints, 1 literal comparators
0000704:INFO:dwt:4 hardware watchpoints
0000705:INFO:fpb:8 hardware breakpoints, 1 literal comparators
0000715:INFO:server:Semihost server started on port 4444 (core 0)
0000716:INFO:gdbserver:GDB server started on port 3333 (core 0)
Exception in thread gdb-server-core0:
Traceback (most recent call last):
  File "C:\python37\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "C:\python37\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 272, in run
    self.target.halt()
  File "C:\python37\lib\site-packages\pyocd\coresight\cortex_m.py", line 683, in halt
    self.write_memory(CortexM.DHCSR, CortexM.DBGKEY | CortexM.C_DEBUGEN | CortexM.C_HALT)
  File "C:\python37\lib\site-packages\pyocd\coresight\cortex_m.py", line 640, in write_memory
    self.ap.write_memory(addr, value, transfer_size)
  File "C:\python37\lib\site-packages\pyocd\coresight\ap.py", line 142, in _locking
    return func(self, *args, **kwargs)
  File "C:\python37\lib\site-packages\pyocd\coresight\ap.py", line 540, in _write_memory
    self.write_reg(MEM_AP_TAR, addr)
  File "C:\python37\lib\site-packages\pyocd\coresight\ap.py", line 142, in _locking
    return func(self, *args, **kwargs)
  File "C:\python37\lib\site-packages\pyocd\coresight\ap.py", line 512, in write_reg
    super(MEM_AP, self).write_reg(addr, data)
  File "C:\python37\lib\site-packages\pyocd\coresight\ap.py", line 142, in _locking
    return func(self, *args, **kwargs)
  File "C:\python37\lib\site-packages\pyocd\coresight\ap.py", line 241, in write_reg
    self.dp.write_ap((self.ap_num << APSEL_SHIFT) | addr, data)
  File "C:\python37\lib\site-packages\pyocd\coresight\dap.py", line 402, in write_ap
    self.probe.write_ap(addr, data)
  File "C:\python37\lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 325, in write_ap
    self._select_ap(addr)
  File "C:\python37\lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 295, in _select_ap
    self.write_dp(self.DP_SELECT, select)
  File "C:\python37\lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 280, in write_dp
    self._link.write_reg(reg_id, data)
[... 23 lines cut ...]
    self.write_reg(MEM_AP_TAR, addr)
  File "C:\python37\lib\site-packages\pyocd\coresight\ap.py", line 142, in _locking
    return func(self, *args, **kwargs)
  File "C:\python37\lib\site-packages\pyocd\coresight\ap.py", line 512, in write_reg
    super(MEM_AP, self).write_reg(addr, data)
  File "C:\python37\lib\site-packages\pyocd\coresight\ap.py", line 142, in _locking
    return func(self, *args, **kwargs)
  File "C:\python37\lib\site-packages\pyocd\coresight\ap.py", line 241, in write_reg
    self.dp.write_ap((self.ap_num << APSEL_SHIFT) | addr, data)
  File "C:\python37\lib\site-packages\pyocd\coresight\dap.py", line 402, in write_ap
    self.probe.write_ap(addr, data)
  File "C:\python37\lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 325, in write_ap
    self._select_ap(addr)
  File "C:\python37\lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 295, in _select_ap
    self.write_dp(self.DP_SELECT, select)
  File "C:\python37\lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 280, in write_dp
    self._link.write_reg(reg_id, data)
[... cut ...] \site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 786, in write_reg
    self._write(dap_index, 1, request, [value])
  File "C:\python37\lib\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 964, in _write
    size = cmd.get_request_space(size_to_transfer, transfer_request, dap_index)
  File "C:\python37\lib\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 217, in get_request_space
    assert self._data_encoded is False
AssertionError
```

RcColes, 2020-05-26:
I can confirm that I have also had this issue on the Musca s1, and that it works correctly after a downgrade to pyocd 0.25.0

maclobdell, 2020-06-01:
@flit can you investigate or recommend the right person to investigate? Looks like there is a problem reported on multiple targets on release 0.26.0.

bohdan-tymkiv, 2021-06-23:
@marcuschangarm @Quincy-Zh @maclobdell 
Is this issue still relevant? I've tested with different PSoC6 flavors and I can not reproduce it with latest pyOCD master.

marcuschangarm, 2021-06-23:
@bohdan-tymkiv I haven't used the PSOC in awhile. Let's close it if you can't reproduce it.

closed 2021-06-23 by marcuschangarm