pyocd/pyOCD #874: [0.26.1] target kv11z75 FlashEraseFailure: erase_sector(0x400) error

Hoohaha, 2020-05-19:
**Reproduce steps**

- Flash a program(hello_world) to board hvpkv11z75m with GDB.
- Reprogram any another application to board, the gdb server will raise error `FlashEraseFailure: erase_sector(0x400) error: 105`

gdb server:
`pyocd gdbserver --trust-crc --port 3800 -u A020000028634e4500585002673a003fa101000097969900 -v -t kv11z7 -O connect_mode=under-reset -O allow_no_cores=True`

**Note**
- This issue happens very regularly in second flash programming.
- Tested on 0.14, it works fine.

```
$ pyocd gdbserver --trust-crc --port 3800 -u A020000028634e4500585002673a003fa101000097969900 -v -t kv11z7 -O connect_mode=under-reset -O allow_no_cores=True

0001286:INFO:board:Target type is kv11z7
0001286:DEBUG:dap_access_cmsis_dap:CMSIS-DAP probe A020000028634e4500585002673a003fa101000097969900 firmware version: 1.0
0001302:DEBUG:sequencer:Running task load_svd
0001302:DEBUG:sequencer:Running task pre_connect
0001302:INFO:coresight_target:Asserting reset prior to connect
0001302:DEBUG:sequencer:Running task dp_init
0001302:DEBUG:sequencer:Running task connect
0001318:DEBUG:dap:Default wire protocol selected; using SWD
0001318:DEBUG:swj:Sending deprecated SWJ sequence to select SWD
0001334:INFO:dap:DP IDR = 0x0bc11477 (v1 MINDP rev0)
0001334:DEBUG:sequencer:Running task clear_sticky_err
0001334:DEBUG:sequencer:Running task power_up_debug
0001334:DEBUG:sequencer:Running task create_discoverer
0001334:DEBUG:sequencer:Running task discovery
0001334:DEBUG:sequencer:Running task find_aps
0001348:DEBUG:sequencer:Running task create_aps
0001348:DEBUG:sequencer:Running task create_ap.0
0001348:INFO:ap:AP#0 IDR = 0x04770031 (AHB-AP var3 rev0)
0001348:DEBUG:ap:AP#0 default HPROT=3 HNONSEC=0
0001348:DEBUG:ap:AP#0 implemented HPROT=3 HNONSEC=0
0001365:DEBUG:sequencer:Running task create_ap.1
0001365:INFO:ap:AP#1 IDR = 0x001c0020 (proprietary)
0001365:DEBUG:sequencer:Running task check_mdm_ap_idr
0001365:DEBUG:target_kinetis:MDM-AP version 2
0001365:DEBUG:sequencer:Running task check_flash_security
0001411:INFO:target_kinetis:KV11Z7 not in secure state
0001411:DEBUG:sequencer:Running task find_components
0001411:DEBUG:sequencer:Running task init_ap.0
0001427:INFO:rom_table:AP#0 Class 0x1 ROM table #0 @ 0xf0002000 (designer=00e part=000)
0001443:INFO:rom_table:[0]<f0000000:MTB-M0+ class=9 designer=43b part=932 devtype=31 archid=0a31 devid=0:0:0>
0001443:INFO:rom_table:[1]<f0001000:MTBDWT class=9 designer=00e part=000 devtype=04 archid=0000 devid=0:0:0>
0001459:INFO:rom_table:[2]<e00ff000:ROM class=1 designer=43b part=4c0>
0001459:INFO:rom_table:  AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b part=4c0)
0001459:INFO:rom_table:  [0]<e000e000:SCS-M0+ class=14 designer=43b part=008>
0001473:INFO:rom_table:  [1]<e0001000:DWT-M0+ class=14 designer=43b part=00a>
0001473:INFO:rom_table:  [2]<e0002000:BPU class=14 designer=43b part=00b>
0001473:DEBUG:sequencer:Running task create_cores
0001473:DEBUG:discovery:Creating SCS-M0+ component
0001490:INFO:cortex_m:CPU core #0 is Cortex-M0+ r0p1
0001490:DEBUG:sequencer:Running task create_components
0001490:DEBUG:discovery:Creating DWT-M0+ component
0001490:INFO:dwt:2 hardware watchpoints
0001490:DEBUG:discovery:Creating BPU component
0001490:INFO:fpb:2 hardware breakpoints, 0 literal comparators
0001490:DEBUG:fpb:fpb has been disabled
0001490:DEBUG:sequencer:Running task check_for_cores
0001490:DEBUG:sequencer:Running task halt_on_connect
0001490:INFO:target_kinetis:Configuring MDM-AP to halt when coming out of reset
0001505:DEBUG:sequencer:Running task post_connect
0001505:INFO:target_kinetis:Deasserting reset post connect
0001505:DEBUG:sequencer:Running task post_connect_hook
0001505:DEBUG:sequencer:Running task create_flash
0001505:DEBUG:sequencer:Running task notify
0001521:DEBUG:cortex_m:Setting vector catch to 0x00000001
0001521:INFO:server:Semihost server started on port 4444 (core 0)
[... 85 lines cut ...]
    return self.flash_op(data)
  File "c:\python27\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 719, in flash_op
    self.flash_loader.commit()
  File "c:\python27\lib\site-packages\pyocd\flash\loader.py", line 168, in commit
    keep_unwritten=self._keep_unwritten)
  File "c:\python27\lib\site-packages\pyocd\flash\builder.py", line 470, in program
    flash_operation = self._sector_erase_program_double_buffer(progress_cb)
  File "c:\python27\lib\site-packages\pyocd\flash\builder.py", line 867, in _sector_erase_program_double_buffer
    self.flash.erase_sector(sector.addr)
  File "c:\python27\lib\site-packages\pyocd\flash\flash.py", line 353, in erase_sector
    raise FlashEraseFailure('erase_sector(0x%x) error: %i' % (address, result), address, result)
FlashEraseFailure: erase_sector(0x400) error: 105
0002628:INFO:gdbserver:Client detached
0002628:INFO:gdbserver:Client disconnected!
0002628:DEBUG:gdbserver:GDB server cleaning up
0002628:DEBUG:packet_io:GDB packet thread: other side closed connection
0002628:DEBUG:packet_io:GDB packet thread stopping
0003065:INFO:server:Semihost server stopped
0003081:DEBUG:session:uninit session <pyocd.core.session.Session object at 0x02749D50>
0003081:DEBUG:board:uninit board <pyocd.board.mbed_board.MbedBoard object at 0x036ABFD0>
0003081:DEBUG:cortex_m:resuming core 0
0003081:DEBUG:manager:added=[] removed=[]
0003081:DEBUG:manager:bps after flush={}
0003097:DEBUG:pywinusb_backend:closing interface
```

flit, 2020-05-21:
Hi @Hoohaha 

The error 105 returned from erase_sector is `kStatus_FLASH_CommandFailure` from the fsl_flash driver. Looking at the driver code, this error can only be returned if `FTFx_FSTAT_MGSTAT0_MASK` is set. When this error is reported, it basically means the flash is worn out (the verify portion of the erase command failed).

Can you please try with another KV11 board? Unfortunately I don't have one of these. (One of the very few Kinetis board I don't have!)

flit, 2020-05-21:
I'm curious if the KV11 CMSIS-Pack behaves differently, since it probably has a newer flash algo than the built-in pyocd KV11 target. Could you please try installing the pack with `pyocd pack -i mkv11`, then using `-tMKV11Z128VFM7`?

Hoohaha, 2020-05-28:
@flit Tried with cmsis pack, has another failures.
But issue is gone if pass chip_erase=auto

flit, 2020-05-29:
Interesting. That implies that the chip erase worked while sector erase didn't. That might make sense even with worn out flash, given that chip and sector erase use different algorithms in the flash controller.

Were you able to test with another board?