pyocd/pyOCD #878: [0.26.1] FlashFailure: init error: 536870913

Hoohaha, 2020-05-28:
Hi @flit ,

FlashFailure still exists with on latest 0.26.1;
Confirmed this issue will happen on version >= 0.22.0

pyocd flash frdmk64f.out -u 0240000041114e4500463003b60f0033a6e1000097969900 --format elf -vvv
```
0003051:DEBUG:session:Project directory: C:\Dpc
0003051:INFO:board:Target type is k64f
0003051:DEBUG:dap_access_cmsis_dap:CMSIS-DAP probe 0240000041114e4500463003b60f0033a6e1000097969900 firmware version: 1.10
0003051:DEBUG:sequencer:Running task load_svd
0003051:DEBUG:sequencer:Running task pre_connect
0003067:DEBUG:sequencer:Running task dp_init
0003067:DEBUG:sequencer:Running task connect
0003067:DEBUG:dap:Default wire protocol selected; using SWD
0003067:DEBUG:swj:Sending deprecated SWJ sequence to select SWD
0003082:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
0003082:DEBUG:sequencer:Running task clear_sticky_err
0003082:DEBUG:sequencer:Running task power_up_debug
0003098:DEBUG:sequencer:Running task create_discoverer
0003098:DEBUG:sequencer:Running task discovery
0003098:DEBUG:sequencer:Running task find_aps
0003098:DEBUG:sequencer:Running task create_aps
0003098:DEBUG:sequencer:Running task create_ap.0
0003115:INFO:ap:AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
0003115:DEBUG:ap:AP#0 default HPROT=3 HNONSEC=0
0003115:DEBUG:ap:AP#0 implemented HPROT=3 HNONSEC=0
0003115:DEBUG:sequencer:Running task create_ap.1
0003129:INFO:ap:AP#1 IDR = 0x001c0000 (proprietary)
0003129:DEBUG:sequencer:Running task check_mdm_ap_idr
0003129:DEBUG:target_kinetis:MDM-AP version 0
0003129:DEBUG:sequencer:Running task check_flash_security
0003316:INFO:target_kinetis:K64F not in secure state
0003316:DEBUG:sequencer:Running task find_components
0003316:DEBUG:sequencer:Running task init_ap.0
0003316:INFO:rom_table:AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=43b part=4c4)
0003332:INFO:rom_table:[0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0003348:INFO:rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
0003348:INFO:rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
0003348:INFO:rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
0003363:INFO:rom_table:[4]<e0040000:TPIU-M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=ca1:0:0>
0003363:INFO:rom_table:[5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
0003379:INFO:rom_table:[6]<e0042000:ETB class=9 designer=43b part=907 devtype=21 archid=0000 devid=0:0:0>
0003379:INFO:rom_table:[7]<e0043000:CSTF class=9 designer=43b part=908 devtype=12 archid=0000 devid=28:0:0>
0003379:DEBUG:sequencer:Running task create_cores
0003379:DEBUG:discovery:Creating SCS-M4 component
0003394:INFO:cortex_m:CPU core #0 is Cortex-M4 r0p1
0003394:DEBUG:sequencer:Running task create_components
0003394:DEBUG:discovery:Creating DWT component
0003394:INFO:dwt:4 hardware watchpoints
0003411:DEBUG:discovery:Creating FPB component
0003411:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0003411:DEBUG:fpb:fpb has been disabled
0003411:DEBUG:discovery:Creating ITM component
0003426:DEBUG:discovery:Creating TPIU-M4 component
0003426:DEBUG:sequencer:Running task check_for_cores
0003426:DEBUG:sequencer:Running task halt_on_connect
0003426:DEBUG:cortex_m:halting core 0
0003426:DEBUG:sequencer:Running task post_connect
0003426:DEBUG:sequencer:Running task post_connect_hook
0003426:DEBUG:sequencer:Running task create_flash
0003426:DEBUG:sequencer:Running task notify
0003426:DEBUG:file_programmer:Writing segment LMA:0x00000000, VMA:0x00000000, size 9180
0003426:DEBUG:cortex_m:halting core 0
0003441:DEBUG:cortex_m:set reset catch, core 0
0003441:DEBUG:cortex_m:halting core 0
0003441:DEBUG:cortex_m:reset, core 0, type=SW_SYSTEM
0007467:DEBUG:cortex_m:clear reset catch, core 0
0007528:DEBUG:cortex_m:cannot resume: target not halted
0007528:DEBUG:cortex_m:halting core 0
0007560:DEBUG:cortex_m:cannot resume: target not halted
0007575:DEBUG:session:uninit session <pyocd.core.session.Session object at 0x03916070>
0007575:DEBUG:board:uninit board <pyocd.board.mbed_board.MbedBoard object at 0x0390CD30>
0007575:DEBUG:cortex_m:cannot resume: target not halted
0007592:DEBUG:pywinusb_backend:closing interface
0007592:CRITICAL:__main__:init error: 536870913
Traceback (most recent call last):
  File "c:\python27\lib\site-packages\pyocd\__main__.py", line 368, in run
    self._COMMANDS[self._args.cmd](self)
  File "c:\python27\lib\site-packages\pyocd\__main__.py", line 532, in do_flash
    file_format=self._args.format)
  File "c:\python27\lib\site-packages\pyocd\flash\file_programmer.py", line 158, in program
    self._loader.commit()
  File "c:\python27\lib\site-packages\pyocd\flash\loader.py", line 168, in commit
    keep_unwritten=self._keep_unwritten)
  File "c:\python27\lib\site-packages\pyocd\flash\builder.py", line 421, in program
    self._build_sectors_and_pages(keep_unwritten)
  File "c:\python27\lib\site-packages\pyocd\flash\builder.py", line 319, in _build_sectors_and_pages
    fill_end_of_page_gap()
  File "c:\python27\lib\site-packages\pyocd\flash\builder.py", line 263, in fill_end_of_page_gap
    self._enable_read_access()
  File "c:\python27\lib\site-packages\pyocd\flash\builder.py", line 225, in _enable_read_access
    self.flash.init(self.flash.Operation.ERASE)
  File "c:\python27\lib\site-packages\pyocd\flash\flash.py", line 250, in init
    raise FlashFailure('init error: %i' % result, result_code=result)
FlashFailure: init error: 536870913

```

Hoohaha, 2020-05-28:
https://github.com/mbedmicro/pyOCD/issues/865

utzig, 2020-08-14:
I am experiencing similar issues, and adding `--connect attach` to the command line seems to fix it.