pyocd/pyOCD #893: Likely libusb fault on Windows 10

buzmeg, 2020-06-15:
Tried to install pyocd on Windows 10 today:

```
(venv) C:\Users\andrewl\Desktop\python\pyocdvenv\pyOCD>pyocd list
Exception ignored in: <function _AutoFinalizedObjectBase.__del__ at 0x000002DF16228AF0>
Traceback (most recent call last):
  File "C:\Users\andrewl\Desktop\python\pyocdvenv\venv\lib\site-packages\usb\_objfinalizer.py", line 84, in __del__
    self.finalize()
  File "C:\Users\andrewl\Desktop\python\pyocdvenv\venv\lib\site-packages\usb\_objfinalizer.py", line 144, in finalize
    self._finalizer()
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.1008.0_x64__qbz5n2kfra8p0\lib\weakref.py", line 566, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
  File "C:\Users\andrewl\Desktop\python\pyocdvenv\venv\lib\site-packages\usb\_objfinalizer.py", line 104, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "C:\Users\andrewl\Desktop\python\pyocdvenv\venv\lib\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "C:\Users\andrewl\Desktop\python\pyocdvenv\venv\lib\site-packages\usb\backend\libusb1.py", line 604, in _finalize_object
    _lib.libusb_unref_device(self.devid)
OSError: exception: access violation writing 0x0000000000000024
  #   Probe                                               Unique ID
---------------------------------------------------------------------
  0   Thirdwayv DAPB Thirdwayv Bulk CMSIS-DAP Interface   AAAAAAAA

```

Presumably the fact that you're ignoring it means that you have seen it before.  I believe that this is a libusb bug and it's biting a lot of people.

tdjastrzebski, 2020-08-23:
I experience a similar issue with version 0.27.2 on Windows 10  

```
C:\Users\Thomas>pyocd list
  #   Probe       Unique ID
--------------------------------------------
  0   STLINK-V3   003000363137510D33333639
Exception ignored in: <bound method _AutoFinalizedObjectBase.__del__ of <usb.backend.libusb1._Device object at 0x000002A2C8CDE780>>
Traceback (most recent call last):
  File "C:\Users\Thomas\AppData\Roaming\Python\Python36\site-packages\usb\_objfinalizer.py", line 84, in __del__
    self.finalize()
  File "C:\Users\Thomas\AppData\Roaming\Python\Python36\site-packages\usb\_objfinalizer.py", line 144, in finalize
    self._finalizer()
  File "c:\program files\python36\lib\weakref.py", line 548, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
  File "C:\Users\Thomas\AppData\Roaming\Python\Python36\site-packages\usb\_objfinalizer.py", line 104, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "C:\Users\Thomas\AppData\Roaming\Python\Python36\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "C:\Users\Thomas\AppData\Roaming\Python\Python36\site-packages\usb\backend\libusb1.py", line 604, in _finalize_object
    _lib.libusb_unref_device(self.devid)
OSError: exception: access violation writing 0x0000000000000024
```

flit, 2020-10-19:
This should be addressed now. The problem was a long-standing bug in the `pyusb` package that surfaced after libusb changed something related to allocation management. pyusb 1.1.0 was recently released that should fix it. (Frustratingly, the fix was available in the pyusb repo for a long time, but never released until recently due to maintainership issues.)

closed 2020-10-19 by flit

tdjastrzebski, 2020-10-19:
@flit thank you, so the fix will be available in the next release?

flit, 2020-10-19:
The fix (and bug) was in pyusb, not pyocd. So if you run `pip install -U pyusb` it will update to the latest pyusb 1.1.0 that has the fix.

Fwiw, you can instead run `pip install -U --upgrade-strategy=eager pyocd` to update all pyocd's dependencies to the latest versions.

tdjastrzebski, 2020-10-19:
@flit I confirm it helped, the issue appears to be fixed

flit, 2020-10-19:
Excellent, thanks for confirming it works on your system!