pyocd/pyOCD #895: stm32l4r5zitx - Loadable section "RW_IRAM1" outside of ELF segments

pilcherd, 2020-06-16:
Hi

I'm using mbed studio and when trying to debug on a stm32l4r5zitx with a STLink.

I'm getting the following log when trying to debug:

`
Selected port 50000 for debugging
0000297:INFO:board:Target type is stm32l4s5vitx
0000428:INFO:coresight_target:Asserting reset prior to connect
0000581:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
0000640:INFO:ap:AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
0000643:INFO:rom_table:AP#0 ROM table #0 @ 0xe00ff000 (designer=020 part=470)
0000645:INFO:rom_table:[0]
0000646:INFO:rom_table:[1]
0000647:INFO:rom_table:[2]
0000648:INFO:rom_table:[3]
0000649:INFO:rom_table:[4]
0000651:INFO:rom_table:[5]
0000652:INFO:cortex_m:CPU core #0 is Cortex-M4 r0p1
0000657:INFO:dwt:4 hardware watchpoints
0000660:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0000668:INFO:coresight_target:Deasserting reset post connect
0000674:INFO:server:Semihost server started on port 4444
0000675:INFO:gdbserver:GDB server started on port 50000
Reading symbols from c:\Users\User\source\repos\arm-mbed\BUILD\MY_BOARD\ARMC6\arm-mbed.elf...
warning: 
**Loadable section "RW_IRAM1" outside of ELF segments**
done.
0000789:INFO:gdbserver:One client connected!
**warning: Loadable section "RW_IRAM1" outside of ELF segments
warning: Overlapping regions in memory map: ignoring**
0x080003d4 in Reset_Handler ()
0000812:INFO:gdbserver:Attempting to load Argon
0000812:INFO:gdbserver:Attempting to load FreeRTOS
0000812:INFO:gdbserver:Attempting to load Zephyr
0000813:INFO:gdbserver:Attempting to load RTX5
0000813:INFO:gdbserver:RTX5 loaded successfully
Attached to debugger on port 50000
Image loaded: c:\Users\User\source\repos\arm-mbed\BUILD\MY_BOARD\ARMC6\arm-mbed.elf
0001809:ERROR:gdbserver:Unhandled exception in handle_message: individual memory accesses must not cross memory region boundaries
Traceback (most recent call last):
  File "c:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 549, in handle_message
    reply = handler(msg[msgStart:])
  File "c:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 776, in v_command
    return self.v_cont(cmd)
  File "c:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 791, in v_cont
    if self.is_threading_enabled():
  File "c:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 1320, in is_threading_enabled
    return (self.thread_provider is not None) and self.thread_provider.is_enabled \
  File "c:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\rtos\rtx5.py", line 408, in is_enabled
    return self.get_kernel_state() != 0 and not self._target.in_thread_mode_on_main_stack()
  File "c:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\rtos\rtx5.py", line 446, in get_kernel_state
    return self._target_context.read8(self._os_rtx_info + RTX5ThreadProvider.KERNEL_STATE_OFFSET)
  File "c:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\core\memory_interface.py", line 64, in read8
    return self.read_memory(addr, 8, now)
  File "c:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\debug\cache.py", line 516, in read_memory
    return self._memcache.read_memory(addr, transfer_size, now)
  File "c:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\debug\cache.py", line 422, in read_memory
    data = self.read_memory_block8(addr, 1)[0]
  File "c:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\debug\cache.py", line 442, in read_memory_block8
    if not self._check_regions(addr, size):
  File "c:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\debug\cache.py", line 414, in _check_regions
    raise MemoryAccessError("individual memory accesses must not cross memory region boundaries")
pyocd.debug.cache.MemoryAccessError: individual memory accesses must not cross memory region boundaries
**"0001809:ERROR:gdbserver:Unhandled exception in handle_message: individual memory accesses must not cross memory region boundaries"
warning: Remote failure reply: E01**
0001809:ERROR:gdbserver:Unhandled exception in handle_message: individual memory accesses must not cross memory region boundaries`

In the map file, the RAM looks ok:

    Execution Region RW_IRAM1 (Base: 0x200001c0, Size: 0x00002cc8, Max: 0x0009fa40, ABSOLUTE)
    Execution Region ARM_LIB_STACK (Base: 0x2009fc00, Size: 0x00000400, Max: 0x00000400, ABSOLUTE)

Processor datasheets say:

> The STM32L4S5xx, STM32L4S7xx and STM32L4S9xx devices feature 640 Kbytes of
> embedded SRAM. This SRAM is split into three blocks:
> • 192 Kbytes mapped at address 0x2000 0000 (SRAM1).
> • 64 Kbytes located at address 0x1000 0000 with hardware parity check (SRAM2).
> This memory is also mapped at address 0x2003 0000 offering a contiguous address
> space with the SRAM1.
> This block is accessed through the ICode/DCode buses for maximum performance.
> These 64 Kbytes SRAM can also be retained in Standby mode.
> The SRAM2 can be write-protected with 1 Kbyte granularity.
> • 384 Kbytes mapped at address 0x2004 0000 - (SRAM3).

So it does sound like memory is 0x2000 0000 - 0x200A 0000 which matches the map file's use of IRAM. So why does pyOCD say the ELF file is wrong? 

Thank you.

pilcherd, 2020-06-16:
Doing an objdump on the elf gives:

arm-mbed.elf:     file format elf32-littlearm

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 ER_IROM1      0000e9c0  08000000  08000000  00000034  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 RW_IRAM1      0000010c  200001c0  200001c0  0000e9f4  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  2 RW_IRAM1      00002bbc  200002cc  200002cc  0000eb00  2**3
                  ALLOC
  3 ARM_LIB_STACK 00000400  2009fc00  2009fc00  0000eb00  2**2
                  ALLOC
  4 .debug_abbrev 0000f180  00000000  00000000  0000eb00  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
  5 .debug_frame  0000da24  00000000  00000000  0001dc80  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
  6 .debug_info   000cf6c2  00000000  00000000  0002b6a4  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
  7 .debug_line   00035ec1  00000000  00000000  000fad66  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
  8 .debug_loc    00022e38  00000000  00000000  00130c27  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
  9 .debug_str    000abe8c  00000000  00000000  00153a5f  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 10 .debug_ranges 00005ce8  00000000  00000000  001ff8eb  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 11 .note         00000020  00000000  00000000  0021bb98  2**2
                  CONTENTS, READONLY
 12 .comment      00003fa0  00000000  00000000  0021bbb8  2**0
                  CONTENTS, READONLY

This matches the map and the CPU spec as far as I can see. Although there are 2 RW_IRAM1 sections.

pilcherd, 2020-06-16:
Keil pack file for STM324L looks to match too:

`        
        <device Dname="STM32L4S5ZITx">
          <memory name="Main_Flash" access="rx"  id="IROM1"     start="0x08000000" size="0x00200000" startup="1" default="1" />
          <memory name="SRAM"       access="rwx" id="IRAM1"     start="0x20000000" size="0x000A0000" init="0" default="1" />
          <algorithm name="CMSIS/Flash/STM32L4Rx_2048_Dual.FLM" start="0x08000000" size="0x00200000"             default="1" />
          <feature type="QFP" n="144"/>
        </device>`

flit, 2020-06-16:
Hi @pilcherd, thanks for all the details!

The warnings about the RW_IRAM1 ELF section and overlapping memory regions are actually from gdb, not pyocd (the output is mixed in Mbed Studio, I guess).

I think the RW_IRAM1 warning from gdb is complaining that the section is marked loadable (LOAD flag in objdump output), but it's not included in an ELF segment (called program headers in objdump, viewable via `-p`). The armcc linker produces files that are sometimes problematic for gdb. cc @thegecko @arekzaluski 

The reported exception is indeed from pyOCD. The memory cache currently reports this exception if a read or write crosses the bounds of memory regions. It's not entirely surprising to me to see this. I was already planning to relax that requirement. 😄 Fixing this should at least solve the pyOCD issue.

pilcherd, 2020-06-17:
Thanks for the reply @flit.

My headers shows this:
```

arm-mbed.elf:     file format elf32-littlearm

Program Header:
    LOAD off    0x00000034 vaddr 0x08000000 paddr 0x08000000 align 2**3
         filesz 0x0000eacc memsz 0x00011a88 flags rwx 80000000
private flags = 5000002: [Version5 EABI]<Unrecognised flag bits set>

```

This is going to head OT a bit, but...

Originally I tried using VS Code with GCC in WSL and then debugging via OpenOCD on the desktop,  While it all built and ran on the board I found OpenOCD didn't support RTX5, so interactive debugging was not possible as the thread info was not known.  Mbed Studio sounded possibly 
a quick way to go but do you know if pyOCD works with VSCode using the arm-cortex extension?

If I try and use the pyOCD as part of the Mbed Studio package in  %localappdata%\Mbed Studio\mbed-studio-tools\python\Scripts, I need the pack working for this MCU.  But when I run pyOCD pack -s   I expect to see a list of packs which are installed, but it shows none.  There are a bunch in %localappdata%\Mbed Studio\mbed-studio-tools\cmsis-packs.  How do I use those?  I ask because when I tried to download from Keil the pack, the one for this processor (STM32L4) had invalid XML in it.

```

C:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\Scripts>pyocd pack -u
0000175:INFO:__main__:Updating pack index...
Jun 16 20:00:10.186 ERRO download of "https://www.keil.com/pack/Keil.STM32F7xx_DFP.pdsc" failed: https://www.keil.com/pack/Keil.STM32F7xx_DFP.pdsc: error trying to connect: unexpected end of file
Jun 16 20:00:10.223 ERRO download of "https://www.keil.com/pack/Keil.STM32G0xx_DFP.pdsc" failed: https://www.keil.com/pack/Keil.STM32G0xx_DFP.pdsc: error trying to connect: unexpected end of file
Jun 16 20:00:10.235 ERRO download of "https://www.keil.com/pack/Keil.STM32H7xx_DFP.pdsc" failed: https://www.keil.com/pack/Keil.STM32H7xx_DFP.pdsc: error trying to connect: unexpected end of file
**Jun 16 20:00:10.282 ERRO download of "https://www.keil.com/pack/Keil.STM32L4xx_DFP.pdsc" failed: https://www.keil.com/pack/Keil.STM32L4xx_DFP.pdsc: error trying to connect: unexpected end of file**
Jun 16 20:00:10.851 ERRO download of "http://www.qorvo.com/extra/keil_pack/Active-Semi.PAC52XX.pdsc" failed: https://www.qorvo.com/extra/keil_pack/Active-Semi.PAC52XX.pdsc: error trying to connect: An existing connection was forcibly closed by the remote host. (os error 10054)
Jun 16 20:00:10.853 ERRO download of "http://www.qorvo.com/extra/keil_pack/Active-Semi.PAC55XX.pdsc" failed: https://www.qorvo.com/extra/keil_pack/Active-Semi.PAC55XX.pdsc: error trying to connect: An existing connection was forcibly closed by the remote host. (os error 10054)
Jun 16 20:00:11.327 WARN tried to parse element "package" from element "html"
Jun 16 20:00:11.330 WARN tried to parse element "package" from element "html"
Jun 16 20:00:11.653 WARN error while parsing attribute at position 13: Attribute value must start with a quote
Jun 16 20:00:11.901 WARN the XML is invalid, an element was wrongly closed
Jun 16 20:00:11.924 WARN the XML is invalid, an element was wrongly closed
Jun 16 20:00:11.955 WARN the XML is invalid, an element was wrongly closed
Jun 16 20:00:11.973 WARN the XML is invalid, an element was wrongly closed
Jun 16 20:00:12.090 WARN tried to parse element "package" from element "html"
Jun 16 20:00:12.092 WARN tried to parse element "package" from element "html"
Jun 16 20:00:12.096 WARN tried to parse element "package" from element "html"
Jun 16 20:00:12.098 WARN tried to parse element "package" from element "html"
Jun 16 20:00:12.102 WARN tried to parse element "package" from element "html"
Jun 16 20:00:12.105 WARN tried to parse element "package" from element "html"
Jun 16 20:00:12.107 WARN tried to parse element "package" from element "html"
Jun 16 20:00:12.109 WARN tried to parse element "package" from element "html"
Jun 16 20:00:12.111 WARN tried to parse element "package" from element "html"
Jun 16 20:00:12.113 WARN tried to parse element "package" from element "html"
Jun 16 20:00:12.115 WARN tried to parse element "package" from element "html"
Jun 16 20:00:12.160 WARN the XML is invalid, an element was wrongly closed
Jun 16 20:00:12.168 WARN the XML is invalid, an element was wrongly closed
Jun 16 20:00:12.175 WARN the XML is invalid, an element was wrongly closed
Jun 16 20:00:12.180 WARN the XML is invalid, an element was wrongly closed
Jun 16 20:00:12.187 WARN the XML is invalid, an element was wrongly closed
Jun 16 20:00:12.224 WARN the XML is invalid, an element was wrongly closed
Jun 16 20:00:12.393 WARN Only Comment, CDATA and DOCTYPE nodes can start with a '!'
Jun 16 20:00:12.536 WARN the XML is invalid, an element was wrongly closed
Jun 16 20:00:12.561 WARN the XML is invalid, an element was wrongly closed
Jun 16 20:00:12.567 WARN the XML is invalid, an element was wrongly closed
Jun 16 20:00:12.763 WARN the XML is invalid, an element was wrongly closed
Jun 16 20:00:12.769 WARN the XML is invalid, an element was wrongly closed
Jun 16 20:00:12.779 WARN the XML is invalid, an element was wrongly closed
Jun 16 20:00:12.787 WARN the XML is invalid, an element was wrongly closed
Jun 16 20:00:12.794 WARN the XML is invalid, an element was wrongly closed
Jun 16 20:00:12.801 WARN the XML is invalid, an element was wrongly closed
Jun 16 20:00:12.809 WARN the XML is invalid, an element was wrongly closed
[... cut ...]
Jun 16 20:00:16.037 WARN cannot parse integer from empty string, Package: RA_DFP, Vendor: Renesas
Jun 16 20:00:16.042 WARN cannot parse integer from empty string, Package: RA_DFP, Vendor: Renesas
Jun 16 20:00:16.047 WARN cannot parse integer from empty string, Package: RA_DFP, Vendor: Renesas

C:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\Scripts>

```

This pack should be local... mbed studio has it and uses it, but I don't see a command to select the pack folder in the help.  Do you know how I can specify the pack folder?

Many thanks.

thegecko, 2020-06-17:
The packs shipped with Mbed Studio are stripped down and only contain the debug data.

You can specify a pack to use with PyOCD using the `--pack` switch. 

Look at the logs in Mbed Studio after a session for an example. 

I assume you have tried the custom targets approach in Mbed Studio to try a different debug target?

pilcherd, 2020-06-17:
> The packs shipped with Mbed Studio are stripped down and only contain the debug data.
>
> You can specify a pack to use with PyOCD using the `--pack` switch.
>
> Look at the logs in Mbed Studio after a [...]

Yes. I have a custom target, a custom board. Can build from the mbed cli from WSL/ubuntu and deploy using OpenOCD but not from Mbed Studio using pyOCD.  Tried switching Mbed Studio to GCC but got stuck [here](https://forums.mbed.com/t/could-not-find-executable-for-gcc-arm/8917/2).

Unfortunately I cannot find any logs in mbed studio showing pyocd options in use.  The only logs in tabs are shown above, and no log files appear to exist in the BUILD folders.

thegecko, 2020-06-17:
> Unfortunately I cannot find any logs in mbed studio

Help > Report an Issue > Access Log

pilcherd, 2020-06-17:
> > Unfortunately I cannot find any logs in mbed studio
> 
> Help > Report an Issue > Access Log

Thank you. I see that cmd accepts a --pack "command" which is different from "pack".

pilcherd, 2020-06-18:
@flit 

I'm now just using vscode with pyOCD and gcc as the compiler.  The elf now looks ok as the LOAD section is in the program headers.

I still get a warning about overlapping reasons and the pyOCD exception.

```

arm-mbed.elf:     file format elf32-littlearm

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         0000d688  08000000  08000000  000000c0  2**6
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .ARM.exidx    00000008  0800d688  0800d688  0000d748  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  2 .data         00000a10  200001c0  0800d690  0000d750  2**3
                  CONTENTS, ALLOC, LOAD, DATA
  3 .bss          000027e0  20000bd0  0800e0a0  0000e160  2**3
                  ALLOC
  4 .heap         0009c850  200033b0  200033b0  0000e160  2**0
                  CONTENTS
  5 .ARM.attributes 0000002e  00000000  00000000  000aa9b0  2**0
                  CONTENTS, READONLY
  6 .comment      0000004c  00000000  00000000  000aa9de  2**0
                  CONTENTS, READONLY
  7 .debug_info   000fd1a7  00000000  00000000  000aaa2a  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
  8 .debug_abbrev 0001a8d4  00000000  00000000  001a7bd1  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
  9 .debug_aranges 000035e0  00000000  00000000  001c24a8  2**3
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 10 .debug_line   0003bbd5  00000000  00000000  001c5a88  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 11 .debug_str    00030f13  00000000  00000000  0020165d  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 12 .debug_ranges 00006e58  00000000  00000000  00232570  2**3
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 13 .debug_frame  0000a250  00000000  00000000  002393c8  2**2
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 14 .debug_loc    00033183  00000000  00000000  00243618  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 15 .stabstr      00000076  00000000  00000000  0027679b  2**0
                  CONTENTS, READONLY, DEBUGGING

```

```
arm-mbed.elf:     file format elf32-littlearm

Program Header:
0x70000001 off    0x0000d748 vaddr 0x0800d688 paddr 0x0800d688 align 2**2
         filesz 0x00000008 memsz 0x00000008 flags r--
    LOAD off    0x000000c0 vaddr 0x08000000 paddr 0x08000000 align 2**6
         filesz 0x0000d690 memsz 0x0000d690 flags r-x
    LOAD off    0x0000d750 vaddr 0x200001c0 paddr 0x0800d690 align 2**3
         filesz 0x00000a10 memsz 0x000031f0 flags rw-
private flags = 5000200: [Version5 EABI] [soft-float ABI]
```

```
0000000449: GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"log","content":"warning: Overlapping regions in memory map: ignoring\n"}]}
0000000449: warning: Overlapping regions in memory map: ignoring
0000000544: GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"0x0800c74c in ?? ()\n"}]}
```

```
>>> gdbserver start
>>> 0457795:ERROR:gdbserver:Unhandled exception in handle_message: individual memory accesses must not cross memory region boundaries
Traceback (most recent call last):
  File "C:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 549, in handle_message
    reply = handler(msg[msgStart:])
  File "C:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 1023, in handle_query
    resp = self.handle_query_xml(b'threads', int(data[0], 16), int(data[1].split(b'#')[0], 16))
  File "C:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 1210, in handle_query_xml
    xml = self.get_threads_xml()
  File "C:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 1300, in get_threads_xml
    if not self.is_threading_enabled():
  File "C:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 1320, in is_threading_enabled
    return (self.thread_provider is not None) and self.thread_provider.is_enabled \
  File "C:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\rtos\rtx5.py", line 408, in is_enabled
    return self.get_kernel_state() != 0 and not self._target.in_thread_mode_on_main_stack()
  File "C:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\rtos\rtx5.py", line 446, in get_kernel_state
    return self._target_context.read8(self._os_rtx_info + RTX5ThreadProvider.KERNEL_STATE_OFFSET)
  File "C:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\core\memory_interface.py", line 64, in read8
    return self.read_memory(addr, 8, now)
  File "C:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\debug\cache.py", line 516, in read_memory
    return self._memcache.read_memory(addr, transfer_size, now)
  File "C:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\debug\cache.py", line 422, in read_memory
    data = self.read_memory_block8(addr, 1)[0]
  File "C:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\debug\cache.py", line 442, in read_memory_block8
    if not self._check_regions(addr, size):
  File "C:\Users\User\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\debug\cache.py", line 414, in _check_regions
    raise MemoryAccessError("individual memory accesses must not cross memory region boundaries")
pyocd.debug.cache.MemoryAccessError: individual memory accesses must not cross memory region boundaries
```

I cannot see the overlapping memory in the ELF sections as the addresses appear to be sequential blocks.  Can you offer any insight as to what is causing this now? 

Many thanks!

pilcherd, 2020-06-19:
Looked a little more.  The memory map has these two overlapping items in it.  Where do they come from?

```
<RamRegion@0x2511c867240 name=IRAM1 type=MemoryType.RAM start=0x20000000 end=0x2009ffff length=0xa0000 access=rwx>, 

<RamRegion@0x2511c8671d0 name=SRAM type=MemoryType.RAM start=0x20000000 end=0x2009ffff length=0xa0000 access=rwx>]

```

AshUK, 2020-07-01:
Also experiencing this with mbed studio and STM32L4 on a custom board

```
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/gdbserver/gdbserver.py", line 549, in handle_message
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/gdbserver/gdbserver.py", line 776, in v_command
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/gdbserver/gdbserver.py", line 791, in v_cont
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/gdbserver/gdbserver.py", line 1320, in is_threading_enabled
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/rtos/rtx5.py", line 408, in is_enabled
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/rtos/rtx5.py", line 446, in get_kernel_state
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/core/memory_interface.py", line 64, in read8
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/debug/cache.py", line 516, in read_memory
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/debug/cache.py", line 422, in read_memory
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/debug/cache.py", line 442, in read_memory_block8
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/debug/cache.py", line 414, in _check_regions
pyocd.debug.cache.MemoryAccessError: individual memory accesses must not cross memory region boundaries []
```

flit, 2020-08-02:
(Really sorry I didn't notice your replies!)

The reason there are the overlapping RAM regions is because that's how they are defined in the CMSIS pack for the L4R5. It looks like this in the PDSC (omitting irrelevant elements):

```xml
<memory id="IRAM1"                                 start="0x20000000" size="0x000A0000" init="0" default="1" />

<device Dname="STM32L4R5AGIx">
  <memory name="Main_Flash" access="rx"  id="IROM1"   start="0x08000000" size="0x00100000" startup="1" default="1" />
  <memory name="SRAM"       access="rwx" id="IRAM1"  start="0x20000000" size="0x000A0000" init="0" default="1" />
</device>
```

It looks like the author intended the inner "IRAM1" to override the outer "IRAM1" (for some reason, even though they are the same address range). The problem is, the documentation for CMSIS Packs does not clearly define rules for overriding outer elements. It states that the `id` attribute is deprecated in favour of `name`. So pyOCD uses `name` when it sees it.

There is a pyOCD bug here, for sure. It should handle this more intelligently. I'll work on a fix. I'm also going to try to get the CMSIS Pack docs improved to remove the ambiguity that led to this.

sschatzkin, 2020-11-23:
FYi , I have the same issue with the b-l4s5i-iot01a st proto brd. I get the same RW-IRAM1 out segment issue.  Got it to the point can connect the debugger, but that is about it.  When the fix is in that will be great .

Thanks

noutram, 2020-12-01:
Same for my Nucleo board (STM32F429ZI) - frequently I'm not able to debug. Using Mbed Studio 1.3.1

heuzen, 2021-01-16:
Dear pyOCD Team,

we are having the same issue with the STM Nucleo L476RG.

"0002485:ERROR:gdbserver:Unhandled exception in handle_message: individual memory accesses must not cross memory region boundaries"

Following this source (https://forums.mbed.com/t/cannot-debug-stm32l4-build/8941) I tried to comment out the line of the concerned Subfamily without any success.

**File:** Keil.STM32L4xx_DFP.pdsc (within archive Keil.STM32L4xx_DFP.2.3.0-small.pack)
**Lines:** 1590 - 1617