pyocd/pyOCD #896: "No cores were discovered!" with STM32H743ZI and Mbed studio

fabiencomte, 2020-06-16:
Hello,
When I try to debug a STM32H743ZI CPU with Mbed Studio, it fails.

0000700:INFO:ap:AP#2 IDR = 0x54770002 (APB-AP var0 rev5)
0000703:INFO:rom_table:AP#0 ROM table #0 @ 0xe00fe000 (designer=020 part=450)
0000705:INFO:rom_table:[0]
0000705:INFO:rom_table:  AP#0 ROM table #1 @ 0xe00ff000 (designer=43b part=4c7)
0000707:ERROR:ap:Transfer error while reading AP#0 ROM table: SWD/JTAG Transfer Fault @ 0xe000efbc-0xe000f043
0000708:INFO:rom_table:AP#2 ROM table #0 @ 0xe00e0000 (designer=020 part=450)
0000710:WARNING:rom_table:Invalid coresight component, cidr=0x0
0000710:INFO:rom_table:[2]
0000717:WARNING:rom_table:Invalid coresight component, cidr=0x0
0000717:INFO:rom_table:[3]
0000718:INFO:rom_table:[4]
0000720:ERROR:ap:Transfer error while reading AP#2 ROM table: SWD/JTAG Transfer Fault @ 0xe00f0fbc-0xe00f1043
0000721:CRITICAL:__main__:No cores were discovered!
Traceback (most recent call last):
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\__main__.py", line 343, in run
    self._COMMANDS[self._args.cmd](self)
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\__main__.py", line 571, in do_gdbserver
    with session:
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\core\session.py", line 287, in __enter__
    self.open()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\core\session.py", line 367, in open
    self._board.init()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\board\board.py", line 83, in init
    self.target.init()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\core\coresight_target.py", line 164, in init
    seq.invoke()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\utility\sequencer.py", line 208, in invoke
    resultSequence = call()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\core\coresight_target.py", line 298, in check_for_cores
    raise exceptions.DebugError("No cores were discovered!")
pyocd.core.exceptions.DebugError: No cores were discovered!
"GDB server stopped unexpectedly with exit code 1"

If I do basic checks outside, it looks correct:

bin\openocd.exe --version
xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-11:28)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html

bin\openocd.exe -f scripts/interface/stlink.cfg -c "adapter_khz 1000; transport select hla_swd" -f scripts/target/stm32h7x.cfg
xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-11:28)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1800 kHz
Info : STLINK V2J36S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.247431
Info : stm32h7x.cpu: hardware has 71 breakpoints, 0 watchpoints
Info : Listening on port 3333 for gdb connections

Configuration: Windows 10 x64, Mbed Studio 1.0.0, xpack-openocd-0.10.0-13-win32-x64

whj4674672, 2020-06-18:
I met the same problem as you. Have you solved it？

MohamedZaheri, 2020-10-05:
Any solution?

mikkleini, 2020-11-13:
Joining the club

Razario, 2021-04-04:
A similar problem. Who can decide? Ready to donate

jbrover, 2021-07-19:
Same problem!