pyocd/pyOCD #910: Too many exceptions from pyocd command

Hoohaha, 2020-07-15:
version: 0.27

There are many exceptions(255) displayed in terminal  when pyocd cannot read aps.
It's very unfirendly. Can we change the level to debug or disable the traceback or another way to fix that?

```
0006193:ERROR:discovery:Exception reading AP#0 IDR: SWD/JTAG Transfer Fault
Traceback (most recent call last):
  File "c:\python27\lib\site-packages\pyocd\coresight\discovery.py", line 158, in _create_1_ap
    ap = AccessPort.create(self.dp, ap_address)
  File "c:\python27\lib\site-packages\pyocd\coresight\ap.py", line 342, in create
    ap.init()
  File "c:\python27\lib\site-packages\pyocd\utility\concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "c:\python27\lib\site-packages\pyocd\coresight\ap.py", line 528, in init
    self._init_hprot()
  File "c:\python27\lib\site-packages\pyocd\coresight\ap.py", line 579, in _init_hprot
    csw = AccessPort.read_reg(self, self._reg_offset + MEM_AP_CSW)
  File "c:\python27\lib\site-packages\pyocd\utility\concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "c:\python27\lib\site-packages\pyocd\coresight\ap.py", line 404, in read_reg
    return self.dp.read_ap(self.address.address + addr, now)
  File "c:\python27\lib\site-packages\pyocd\coresight\dap.py", line 615, in read_ap
    return read_ap_cb()
  File "c:\python27\lib\site-packages\pyocd\coresight\dap.py", line 603, in read_ap_cb
    result = result_cb()
  File "c:\python27\lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 287, in read_ap_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "c:\python27\lib\site-packages\six.py", line 737, in raise_from
    raise value
TransferFaultError: SWD/JTAG Transfer Fault
0006207:ERROR:discovery:Exception reading AP#1 IDR: SWD/JTAG Transfer Fault
Traceback (most recent call last):
  File "c:\python27\lib\site-packages\pyocd\coresight\discovery.py", line 158, in _create_1_ap
    ap = AccessPort.create(self.dp, ap_address)
  File "c:\python27\lib\site-packages\pyocd\coresight\ap.py", line 342, in create
    ap.init()
  File "c:\python27\lib\site-packages\pyocd\utility\concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "c:\python27\lib\site-packages\pyocd\coresight\ap.py", line 528, in init
    self._init_hprot()
  File "c:\python27\lib\site-packages\pyocd\coresight\ap.py", line 579, in _init_hprot
    csw = AccessPort.read_reg(self, self._reg_offset + MEM_AP_CSW)
  File "c:\python27\lib\site-packages\pyocd\utility\concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "c:\python27\lib\site-packages\pyocd\coresight\ap.py", line 404, in read_reg
    return self.dp.read_ap(self.address.address + addr, now)
  File "c:\python27\lib\site-packages\pyocd\coresight\dap.py", line 615, in read_ap
    return read_ap_cb()
  File "c:\python27\lib\site-packages\pyocd\coresight\dap.py", line 603, in read_ap_cb
    result = result_cb()
  File "c:\python27\lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 287, in read_ap_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "c:\python27\lib\site-packages\six.py", line 737, in raise_from
    raise value
TransferFaultError: SWD/JTAG Transfer Fault
0006223:ERROR:discovery:Exception reading AP#2 IDR: SWD/JTAG Transfer Fault
Traceback (most recent call last):
  File "c:\python27\lib\site-packages\pyocd\coresight\discovery.py", line 158, in _create_1_ap
    ap = AccessPort.create(self.dp, ap_address)
  File "c:\python27\lib\site-packages\pyocd\coresight\ap.py", line 342, in create
    ap.init()
  File "c:\python27\lib\site-packages\pyocd\utility\concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "c:\python27\lib\site-packages\pyocd\coresight\ap.py", line 528, in init
    self._init_hprot()
  File "c:\python27\lib\site-packages\pyocd\coresight\ap.py", line 579, in _init_hprot
    csw = AccessPort.read_reg(self, self._reg_offset + MEM_AP_CSW)
  File "c:\python27\lib\site-packages\pyocd\utility\concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
[... 2088 lines cut ...]
  File "c:\python27\lib\site-packages\pyocd\coresight\discovery.py", line 158, in _create_1_ap
    ap = AccessPort.create(self.dp, ap_address)
  File "c:\python27\lib\site-packages\pyocd\coresight\ap.py", line 342, in create
    ap.init()
  File "c:\python27\lib\site-packages\pyocd\utility\concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "c:\python27\lib\site-packages\pyocd\coresight\ap.py", line 528, in init
    self._init_hprot()
  File "c:\python27\lib\site-packages\pyocd\coresight\ap.py", line 579, in _init_hprot
    csw = AccessPort.read_reg(self, self._reg_offset + MEM_AP_CSW)
  File "c:\python27\lib\site-packages\pyocd\utility\concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "c:\python27\lib\site-packages\pyocd\coresight\ap.py", line 404, in read_reg
    return self.dp.read_ap(self.address.address + addr, now)
  File "c:\python27\lib\site-packages\pyocd\coresight\dap.py", line 615, in read_ap
    return read_ap_cb()
  File "c:\python27\lib\site-packages\pyocd\coresight\dap.py", line 603, in read_ap_cb
    result = result_cb()
  File "c:\python27\lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 287, in read_ap_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "c:\python27\lib\site-packages\six.py", line 737, in raise_from
    raise value
TransferFaultError: SWD/JTAG Transfer Fault

```

flit, 2020-10-19:
Looks like a strange combination of issues. The logic is supposed to terminate scanning APs after it gets 2 invalid AP ID register reads in a row (ID == 0). So reading the AP's ID must have returned a non-zero value? That's very odd. In fact, the ID seems to have indicated the APs are all MEM-APs, otherwise you wouldn't see the `_init_hprot()` call in the backtraces.

I'm not sure what the right thing to do here is. If pyocd sees a non-zero AP ID, it should examine the AP.

Can you tell me more about the system? We can talk via email.

Hoohaha, 2020-10-21:
@flit 
os: windows 7 + kinetis frdmk66f.
This is low priority. 
I will continue to monitor this issue on new version of pyocd.

flit, 2020-10-21:
I did _not_ expect K66F! That's surprising. Are there any special conditions that might have triggered this, such as low power states?