pyocd/pyOCD #918: pyOCD failed to flash - STLink error (29)

Dorsel89, 2020-07-25:
Hey,

im trying to flash my nordic chip via mbed studio and pyocd. Everything worked fine until i pressed the "update all tools" button. Unfortunately I did not found any information to this error code. Would be nice if you have any advice for me. 

Thanks in advance!

Here part of my mbed-studio.log

[2020-07-25T18:32:20.055] [DEBUG] root/process - Starting raw process: c:\Use[2020-07-25T18:32:22.010] [DEBUG] root/process - Process 3592 has exited with code 1. [ 'c:\\Users\\Dominik\\AppData\\Local\\Mbed Studio\\mbed-studio-tools\\python\\Scripts\\pyocd.exe',
rs\Dominik\AppData\Local\Mbed Studio\mbed-studio-tools\python\Scripts\pyocd.exe, with args: flash --frequency 1800000 -O connect_mode=under-reset --uid 56ff6d064975485724502087 --target nRF52832_xxAA --pack c:\Users\Dominik\AppData\Local\Mbed Studio\mbed-studio-tools\cmsis-packs\NordicSemiconductor.nRF_DeviceFamilyPack.8.32.1-small.pack --erase=chip c:\Users\Dominik\Mbed Programs\phyphoxCORE\BUILD\NRF52_DK\ARMC6\phyphoxCORE.hex,  with options: {"cwd":"c:\\Users\\Dominik\\AppData\\Local\\Mbed Studio\\mbed-studio-tools\\python\\Scripts"} []
  [ 'flash',
    '--frequency',
    '1800000',
    '-O',
    'connect_mode=under-reset',
    '--uid',
    '56ff6d064975485724502087',
    '--target',
    'nRF52832_xxAA',
    '--pack',
    'c:\\Users\\Dominik\\AppData\\Local\\Mbed Studio\\mbed-studio-tools\\cmsis-packs\\NordicSemiconductor.nRF_DeviceFamilyPack.8.32.1-small.pack',
    '--erase=chip',
    'c:\\Users\\Dominik\\Mbed Programs\\phyphoxCORE\\BUILD\\NRF52_DK\\ARMC6\\phyphoxCORE.hex' ] ]
[2020-07-25T18:32:22.011] [DEBUG] root/process - Unregistering process. [ID: 19] []
[2020-07-25T18:32:22.011] [DEBUG] root/process - The process was successfully unregistered. [ID: 19] []
[2020-07-25T18:32:22.011] [ERROR] root/mbs-deploy - 0000906:CRITICAL:__main__:STLink error (29): Unknown error
Traceback (most recent call last):
  File "c:\Users\Dominik\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\__main__.py", line 362, in run
    self._COMMANDS[self._args.cmd](self)
  File "c:\Users\Dominik\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\__main__.py", line 520, in do_flash
    with session:
  File "c:\Users\Dominik\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\core\session.py", line 302, in __enter__
    self.open()
  File "c:\Users\Dominik\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\core\session.py", line 420, in open
    self._board.init()
  File "c:\Users\Dominik\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\board\board.py", line 85, in init
    self.target.init()
  File "c:\Users\Dominik\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\core\coresight_target.py", line 160, in init
    seq.invoke()
  File "c:\Users\Dominik\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\utility\sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "c:\Users\Dominik\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\utility\sequencer.py", line 208, in invoke
    resultSequence = call()
  File "c:\Users\Dominik\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\target\family\target_nRF52.py", line 97, in check_flash_security
    if self.is_locked():
  File "c:\Users\Dominik\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\target\family\target_nRF52.py", line 114, in is_locked
    status = self.ctrl_ap.read_reg(CTRL_AP_APPROTECTSTATUS)
  File "c:\Users\Dominik\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\utility\concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "c:\Users\Dominik\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\coresight\ap.py", line 404, in read_reg
    return self.dp.read_ap(self.address.address + addr, now)
  File "c:\Users\Dominik\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\coresight\dap.py", line 589, in read_ap
    result_cb = self.probe.read_ap(addr, now=False)
  File "c:\Users\Dominik\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\probe\stlink_probe.py", line 183, in read_ap
    result = self._link.read_dap_register(apsel, addr & 0xffff)
  File "c:\Users\Dominik\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\probe\stlink\stlink.py", line 436, in read_dap_register
    self._check_status(response[:2])
  File "c:\Users\Dominik\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\probe\stlink\stlink.py", line 307, in _check_status
    raise exceptions.ProbeError(error_message)
pyocd.core.exceptions.ProbeError: STLink error (29): Unknown error
 []
[2020-07-25T18:32:22.011] [ERROR] MbedStudio - Request deploy failed with error: PyOCD failed to flash the device, please check the logs for more information [ 'Error: PyOCD failed to flash the device, please check the logs for more information\n    at PyocdDeployer.<anonymous> (C:\\Mbed Studio\\resources\\app\\node_modules\\@mbed\\mbs-deploy\\lib\\backend-electron\\deployers\\pyocd-deployer.js:168:31)\n    at step (C:\\Mbed Studio\\resources\\app\\node_modules\\@mbed\\mbs-deploy\\lib\\backend-electron\\deployers\\pyocd-deployer.js:48:23)\n    at Object.throw (C:\\Mbed Studio\\resources\\app\\node_modules\\@mbed\\mbs-deploy\\lib\\backend-electron\\deployers\\pyocd-deployer.js:29:53)\n    at rejected (C:\\Mbed Studio\\resources\\app\\node_modules\\@mbed\\mbs-deploy\\lib\\backend-electron\\deployers\\pyocd-deployer.js:21:65)' ]
[2020-07-25T18:32:22.018] [INFO] root/mbs-deploy - Request deploy failed with message: PyOCD failed to flash the device, please check the logs for more information []
[2020-07-25T18:32:22.018] [INFO] root/mbs-core - Updating CoreMbedApplicationState with updateType: RUN_SESSION, update: false []

flit, 2020-07-26:
You're using an STLink probe with an nRF device?

flit, 2020-07-26:
cc @thegecko

Dorsel89, 2020-07-26:
Yes, I am using the ST-LINK V2 for flashing/debuging (SWD) an nRF52832.

flit, 2020-07-26:
I believe the STLink firmware checks that it is connected to a STM32 device and refuses for operate if it sees another vendor's device.

Dorsel89, 2020-07-27:
I just checked once again my ST-LINK firmware. I used V2.J37.S7 before and just flashed V2.J34.S7. With this version it looks like everthing is working again! =)

closed 2020-07-27 by Dorsel89