pyocd/pyOCD #921: Fails to start

diggit, 2020-08-02:
Hi,
When I try to execute it, exception happens. Installed all packages using Arch AUR instead of pip.
- pyocd 0.27.2
- system is Archlinux
- pylink version 0.6.1

This is the exception. I don't even know if the problem is in pyocd or elsewhere.
```
# pyocd

Traceback (most recent call last):
  File "/usr/bin/pyocd", line 33, in <module>
    sys.exit(load_entry_point('pyocd==0.27.2', 'console_scripts', 'pyocd')())
  File "/usr/bin/pyocd", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.8/importlib/metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3.8/site-packages/pyocd/__init__.py", line 21, in <module>
    from . import gdbserver
  File "/usr/lib/python3.8/site-packages/pyocd/gdbserver/__init__.py", line 17, in <module>
    from .gdbserver import GDBServer
  File "/usr/lib/python3.8/site-packages/pyocd/gdbserver/gdbserver.py", line 36, in <module>
    from .syscall import GDBSyscallIOHandler
  File "/usr/lib/python3.8/site-packages/pyocd/gdbserver/syscall.py", line 18, in <module>
    from ..debug.semihost import SemihostIOHandler
  File "/usr/lib/python3.8/site-packages/pyocd/debug/semihost.py", line 25, in <module>
    from ..core import (exceptions, session)
  File "/usr/lib/python3.8/site-packages/pyocd/core/session.py", line 31, in <module>
    from ..board.board import Board
  File "/usr/lib/python3.8/site-packages/pyocd/board/board.py", line 18, in <module>
    from ..target import TARGET
  File "/usr/lib/python3.8/site-packages/pyocd/target/__init__.py", line 17, in <module>
    from .builtin import BUILTIN_TARGETS
  File "/usr/lib/python3.8/site-packages/pyocd/target/builtin/__init__.py", line 17, in <module>
    from ...core.coresight_target import CoreSightTarget
  File "/usr/lib/python3.8/site-packages/pyocd/core/coresight_target.py", line 25, in <module>
    from ..debug.svd.loader import (SVDFile, SVDLoader)
  File "/usr/lib/python3.8/site-packages/pyocd/debug/svd/loader.py", line 19, in <module>
    import pkg_resources
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3257, in <module>
    def _initialize_master_working_set():
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3240, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3269, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 582, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 899, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 785, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'future' distribution was not found and is required by pylink-square

```

flit, 2020-08-20:
Hi @diggit! It seems AUR didn't install a package required by a dependency of pyocd. Unfortunately, distribution-provided packages are the responsibility of the distribution; there's not much I can do directly.

Can you try installing via pip? Or you can check out the pyocd repo and install by running `setup.py`.

diggit, 2020-08-20:
I'd like to avoid mixing pip and distribution packages. I know you that's my problem :). Do you recognize what is missing?

I'll try to test is with pip and isolated environment.