pyocd/pyOCD #923: pyocd pack -u error trying to connect

xia-rj, 2020-08-03:
I want to using pack -u update cmsis_pack, but there have an erro. When I used curl, I find that this page has moved https://sadevicepacksprodus.blob.core.windows.net/idxfile/index.pidx. How to fix the problem.

boraozgen, 2020-08-06:
Same

`Aug 06 11:32:59.769 ERRO https://www.keil.com/pack/index.pidx: error trying to connect: unexpected end of file, uri: http://www.keil.com/pack/index.pidx`

flit, 2020-08-09:
Thanks for letting me know—I wasn't aware of this. This issue will have to be worked around on the server.

flit, 2020-08-10:
The server should be fixed. Can you please test and confirm?

(Note that there are known issues downloading the STM32 F7xx, L0xx, and G4xx pack descriptors. These failures are being looked at.)

tarj, 2020-08-10:
Tried pyocd pack -u
0001896:INFO:__main__:Updating pack index...
Aug 10 20:04:57.840 ERRO download of "http://www.keil.com/pack/ARM.minar.pdsc" failed: unexpected end of file
Aug 10 20:04:57.846 ERRO download of "http://www.keil.com/pack/Keil.Kinetis_K60_DFP.pdsc" failed: unexpected end of file
...
Aug 10 20:08:07.274 WARN tried to parse element "package" from element "html"
Aug 10 20:08:07.281 WARN tried to parse element "package" from element "html"
Aug 10 20:08:07.342 WARN error while parsing attribute at position 26: Got 2 '=' tokens
...
Complete log:
[pyocd_pack_update.log](https://github.com/mbedmicro/pyOCD/files/5052731/pyocd_pack_update.log)

flit, 2020-08-10:
Thanks for testing and the log. I'll keep working on it.

flit, 2020-08-10:
What's really strange is that only a handful of STM32 packs fail for me. I also don't see the Maxim failures. (The Qorvo failures are known; there's nothing we can do about it.)

tarj, 2020-08-11:
I don't know if is it matter, but I reinstalled pyocd==0.27.1.dev45 (from source - setup.py install) and ran it again:
pyocd pack -u
0001673:INFO:__main__:Updating pack index...
Aug 11 09:28:34.188 ERRO expecting </meta> found </HEAD>, uri: http://www.keil.com/pack/index.pidx

flit, 2020-08-11:
Ok, it _should_ be fixed completely now. Please retest and confirm. (Really sorry for the trouble!)

flit, 2020-08-11:
@tarj I think you might have happened to test just as the server was being modified, which could account for seeing the failure again.

flit, 2020-09-08:
Closing. Everything is working now. If this pops up again please reopen.

frmdstryr, 2020-10-08:
Just hit this while trying to setup a new system today:

```
(base) user@hp:~$ pyocd pack -u
0000647:INFO:__main__:Updating pack index...
Oct 08 19:32:49.077 ERRO https://www.keil.com/pack/index.pidx: error trying to connect: unexpected end of file, uri: http://www.keil.com/pack/index.pidx
Could not open file "/home/user/.local/share/cmsis-pack-manager/aliases.json"
```

flit, 2020-10-10:
@frmdstryr Thanks for reporting. I see that now too. Something changed on the keil.com server. I've raised it internally so hopefully it will be fixed soon.

As a temporary workaround you can follow the instructions regarding [manual pack usage](https://github.com/pyocd/pyOCD/blob/master/docs/target_support.md#manual-pack-usage) from the pyocd target support docs.

flit, 2020-10-15:
Fixed (again).

closed 2020-10-15 by flit