pyocd/pyOCD #924: program flash using cmsis fail

heartacker, 2020-08-03:
版本: 1.47.3 (system setup)
提交: 91899dcef7b8110878ea59626991a18c8a6a1b3e
日期: 2020-07-23T13:12:49.994Z
Electron: 7.3.2
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Windows_NT x64 10.0.19041
python: 64 3.7.8

```shell
$ python -m pyocd flash --config e:\github\AC781X\AC781X\ac7811_eide\.eide\pyocd.yaml -t cortex_m -f 4M e:\github\AC781X\AC781X\ac7811_eide\out\AC5\ac7811_eide.hex
```
```log
0000905:CRITICAL:__main__:uncaught exception: 'str' object has no attribute 'pop'
Traceback (most recent call last):
  File "D:\Program Files\Python37\lib\site-packages\pyocd\__main__.py", line 362, in run
    self._COMMANDS[self._args.cmd](self)
  File "D:\Program Files\Python37\lib\site-packages\pyocd\__main__.py", line 516, in do_flash
    options=convert_session_options(self._args.options))
  File "D:\Program Files\Python37\lib\site-packages\pyocd\core\helpers.py", line 242, in session_with_chosen_probe
    return Session(probe, auto_open=auto_open, options=options, **kwargs)
  File "D:\Program Files\Python37\lib\site-packages\pyocd\core\session.py", line 153, in __init__
    probesConfig = config.pop('probes', None)
AttributeError: 'str' object has no attribute 'pop'
终端进程“C:\WINDOWS\system32\cmd.exe /C python -m pyocd flash --config e:\github\AC781X\AC781X\ac7811_eide\.eide\pyocd.yaml -t cortex_m -f 4M e:\github\AC781X\AC781X\ac7811_eide\out\AC5\ac7811_eide.hex”已终止，退出代码: 1。
```

flit, 2020-08-03:
Hi @heartacker,

The problem is that you are specifying the target type to be "cortex_m" (the `-t cortex_m` argument). This target is generic and doesn't have any memory regions defined.

Instead, set the target type to one matching your chip. In your case, you will be using a target defined in the AC781x_DFP CMSIS pack. To see which targets are available, you can run `pyocd list --targets --config e:\github\AC781X\AC781X\ac7811_eide\.eide\pyocd.yaml --name ac781`. It will look into the pack you have set in your config file.

heartacker, 2020-08-03:
> Hi @heartacker,
>
> The problem is that you are specifying the target type to be "cortex_m" (the `-t cortex_m` argument). This target is generic and doesn't have any memory regions defined.
>
> Instead, [...]

thanks for help

closed 2020-08-03 by heartacker