pyocd/pyOCD #925: wrong ELF class: ELFCLASS32

UweBonnes, 2020-08-06:
Hello,

I recently updated Opensuse to 15.2. OpenSuse15.2 switched to python3. Probably since then, OpenOCD no longer works. With the OpenSuse system tools already pip fails. So I tried with a user venv. And now I followed the steps in docs/developers_guide.md:

~/devel/pyOCD> python3 -mvirtualenv venv3
[venv3] hertz:~/devel/pyOCD> source venv3/bin/activate.csh
[venv3] hertz:~/devel/pyOCD> pip install -r dev-requirements.txt
[venv3] hertz:~/devel/pyOCD> python setup.py install
[venv3] hertz:~/devel/pyOCD> venv3/bin/pyocd gdb
0000676:CRITICAL:__main__:uncaught exception: /tmp/tmp1gi3hg44.so: wrong ELF class: ELFCLASS32
Traceback (most recent call last):
  File "/devel/pyOCD/venv3/lib/python3.6/site-packages/pyocd-0.27.1.dev28-py3.6.egg/pyocd/__main__.py", line 362, in run
    self._COMMANDS[self._args.cmd](self)
  File "/devel/pyOCD/venv3/lib/python3.6/site-packages/pyocd-0.27.1.dev28-py3.6.egg/pyocd/__main__.py", line 676, in do_gdbserver
    options=sessionOptions)
  File "/devel/pyOCD/venv3/lib/python3.6/site-packages/pyocd-0.27.1.dev28-py3.6.egg/pyocd/core/helpers.py", line 237, in session_with_chosen_probe
    unique_id=unique_id,
...
  File "/devel/pyOCD/venv3/lib/python3.6/site-packages/pylink/jlink.py", line 261, in __init__
    lib = library.Library()
  File "/devel/pyOCD/venv3/lib/python3.6/site-packages/pylink/library.py", line 261, in __init__
    self.load_default()
  File "/devel/pyOCD/venv3/lib/python3.6/site-packages/pylink/library.py", line 299, in load_default
    return self.load(path)
  File "/devel/pyOCD/venv3/lib/python3.6/site-packages/pylink/library.py", line 353, in load
    self._lib = ctypes.cdll.LoadLibrary(tf.name)
  File "/usr/lib64/python3.6/ctypes/__init__.py", line 426, in LoadLibrary
    return self._dlltype(name)
  File "/usr/lib64/python3.6/ctypes/__init__.py", line 348, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /tmp/tmp1gi3hg44.so: wrong ELF class: ELFCLASS32

Any help to fix my system appreciated!

flit, 2020-08-09:
Hi @UweBonnes, the switch to python3 is probably not the root cause. The failure happens when it's trying to load the J-Link library. (More specifically, when the pylink-square package used by pyOCD is trying to load the J-Link lib.)

Maybe reinstalling the J-Link driver package would fix it?

I'm not sure what the "wrong ELF class: ELFCLASS32" error is really complaining about, why it would be different after the update. It sounds related to 32-/64-bit library compatibility. Could the OpenSuse update have switched from a 32-bit python2 to 64-bit python3, or some other 32/64 change.

hariharan235, 2022-07-02:
> Hi @UweBonnes, the switch to python3 is probably not the root cause. The failure happens when it's trying to load the J-Link library. (More specifically, when the pylink-square package used by pyOCD [...]

Thanks. This fixed this issue for me after I uninstalled jlink

closed 2022-07-03 by flit

Teslafly, 2024-05-20:
This isn't really closed. by uninstalling jlink, jlink support in pyocd was completely disabled. which doesn't help if you are trying to use a jlink.

I think this issue is passively related to https://github.com/square/pylink/issues/33
where the jlink libraries that pylink-square relies on are not being imported properly on linux systems like the raspberry pi. if 23 or 64 bit libraries are installed don't seem to make a difference. 

This happens when using pylink standalone too.
```$ pylink info
Traceback (most recent call last):
  File "/home/Proactive/test_fixture_dev/.venv/bin/pylink", line 33, in <module>
    sys.exit(load_entry_point('pylink-square==1.2.0', 'console_scripts', 'pylink')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Proactive/test_fixture_dev/.venv/lib/python3.11/site-packages/pylink_square-1.2.0-py3.11.egg/pylink/__main__.py", line 600, in main
  File "/home/Proactive/test_fixture_dev/.venv/lib/python3.11/site-packages/pylink_square-1.2.0-py3.11.egg/pylink/__main__.py", line 352, in run
  File "/home/Proactive/test_fixture_dev/.venv/lib/python3.11/site-packages/pylink_square-1.2.0-py3.11.egg/pylink/__main__.py", line 76, in create_jlink
  File "/home/Proactive/test_fixture_dev/.venv/lib/python3.11/site-packages/pylink_square-1.2.0-py3.11.egg/pylink/jlink.py", line 293, in __init__
  File "/home/Proactive/test_fixture_dev/.venv/lib/python3.11/site-packages/pylink_square-1.2.0-py3.11.egg/pylink/library.py", line 293, in __init__
  File "/home/Proactive/test_fixture_dev/.venv/lib/python3.11/site-packages/pylink_square-1.2.0-py3.11.egg/pylink/library.py", line 351, in load_default
  File "/home/Proactive/test_fixture_dev/.venv/lib/python3.11/site-packages/pylink_square-1.2.0-py3.11.egg/pylink/library.py", line 410, in load
  File "/usr/lib/python3.11/ctypes/__init__.py", line 454, in LoadLibrary
    return self._dlltype(name)
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/ctypes/__init__.py", line 376, in __init__
    self._handle = _dlopen(self._name, mode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: /tmp/tmpk3c9r2zv.so: wrong ELF class: ELFCLASS32```