pyocd/pyOCD #930: pyOCD fails to detect Cortex-M1 CPU core

henrikbrixandersen, 2020-08-11:
I am using the ARM Cortex-M1 DesignStart FPGA for Xilinx (https://developer.arm.com/ip-products/designstart/fpga) reference design on a Digilent Arty A7 (https://store.digilentinc.com/arty) along with an ARM V2C-DAPLINK-035A (https://developer.arm.com/tools-and-software/development-boards/designstart-daplink-board).

I am trying to get pyOCD to connect to the Cortex-M1 CPU core through the CMSIS-DAP adapter present on the V2C-DAPLINK shield, but pyOCD fails to detect the Cortex-M1 CPU core:

```
$ pyocd --version
0.27.2

$ pyocd commander
0001055:WARNING:mbed_board:Board ID 5008 is not recognized, using generic cortex_m target.
0001056:WARNING:board:Generic 'cortex_m' target type is selected by default; is this intentional? You will be able to debug most devices, but not program  flash. To set the target type use the '--target' argument or 'target_override' option. Use 'pyocd list --targets' to see available targets types.
0001193:WARNING:rom_table:Invalid coresight component, cidr=0x0
0001203:WARNING:rom_table:Invalid coresight component, cidr=0x0
0001211:WARNING:rom_table:Invalid coresight component, cidr=0x0
Exception while initing board: No cores were discovered!
Traceback (most recent call last):
  File "/home/brix/.local/lib/python3.8/site-packages/pyocd/tools/pyocd.py", line 824, in connect
    self.session.open(init_board=not self.args.no_init)
  File "/home/brix/.local/lib/python3.8/site-packages/pyocd/core/session.py", line 420, in open
    self._board.init()
  File "/home/brix/.local/lib/python3.8/site-packages/pyocd/board/board.py", line 85, in init
    self.target.init()
  File "/home/brix/.local/lib/python3.8/site-packages/pyocd/core/coresight_target.py", line 160, in init
    seq.invoke()
  File "/home/brix/.local/lib/python3.8/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
  File "/home/brix/.local/lib/python3.8/site-packages/pyocd/core/coresight_target.py", line 299, in check_for_cores
    raise exceptions.DebugError("No cores were discovered!")
pyocd.core.exceptions.DebugError: No cores were discovered!
```

The CMSIS-DAP works with other debugging tools (e.g. openocd) which are able to detects the Cortex-M1 CPU core, so the hardware design and setup is functioning correctly.

henrikbrixandersen, 2020-08-12:
Verbose output from pyOCD failing to detect the core:

```
$ pyocd commander -v -v
0000330:DEBUG:pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=0e0f PID=0003). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000351:DEBUG:pyusb_v2_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=0e0f PID=0003). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000993:DEBUG:session:Project directory: /home/brix/Projects/zephyrproject/zephyr/samples/hello_world/build
0000994:WARNING:mbed_board:Board ID 5008 is not recognized, using generic cortex_m target.
0000994:WARNING:board:Generic 'cortex_m' target type is selected by default; is this intentional? You will be able to debug most devices, but not program  flash. To set the target type use the '--target' argument or 'target_override' option. Use 'pyocd list --targets' to see available targets types.
0000994:INFO:board:Target type is cortex_m
0001019:DEBUG:dap_access_cmsis_dap:CMSIS-DAP probe 50080000022de80a00000000000000000000000097969902 firmware version: 1.0
0001030:DEBUG:sequencer:Running task load_svd
0001030:DEBUG:sequencer:Running task pre_connect
0001030:DEBUG:sequencer:Running task dp_init
0001030:DEBUG:sequencer:Running task get_probe_capabilities
0001030:DEBUG:sequencer:Running task connect
0001042:DEBUG:dap:Default wire protocol selected; using SWD
0001042:DEBUG:swj:Sending deprecated SWJ sequence to select SWD
0001058:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
0001058:DEBUG:sequencer:Running task clear_sticky_err
0001059:DEBUG:sequencer:Running task power_up_debug
0001062:DEBUG:sequencer:Running task check_version
0001062:DEBUG:sequencer:Running task create_discoverer
0001062:DEBUG:sequencer:Running task discovery
0001062:DEBUG:sequencer:Running task find_aps
0001070:DEBUG:sequencer:Running task create_aps
0001070:DEBUG:sequencer:Running task create_ap.0
0001074:INFO:ap:AHB-AP#0 IDR = 0x44770001 (AHB-AP var0 rev4)
0001083:DEBUG:ap:AHB-AP#0 default HPROT=3 HNONSEC=0
0001086:DEBUG:ap:AHB-AP#0 implemented HPROT=f HNONSEC=1
0001093:DEBUG:sequencer:Running task find_components
0001093:DEBUG:sequencer:Running task init_ap.0
0001106:INFO:rom_table:AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=43b part=470)
0001117:WARNING:rom_table:Invalid coresight component, cidr=0x0
0001117:INFO:rom_table:[0]<e000e000: cidr=0, pidr=0, component invalid>
0001125:WARNING:rom_table:Invalid coresight component, cidr=0x0
0001125:INFO:rom_table:[1]<e0001000: cidr=0, pidr=0, component invalid>
0001134:WARNING:rom_table:Invalid coresight component, cidr=0x0
0001134:INFO:rom_table:[2]<e0002000: cidr=0, pidr=0, component invalid>
0001134:DEBUG:sequencer:Running task create_cores
0001134:DEBUG:sequencer:Running task create_components
0001134:DEBUG:sequencer:Running task check_for_cores
Exception while initing board: No cores were discovered!
Traceback (most recent call last):
  File "/home/brix/.local/lib/python3.8/site-packages/pyocd/tools/pyocd.py", line 824, in connect
    self.session.open(init_board=not self.args.no_init)
  File "/home/brix/.local/lib/python3.8/site-packages/pyocd/core/session.py", line 420, in open
    self._board.init()
  File "/home/brix/.local/lib/python3.8/site-packages/pyocd/board/board.py", line 85, in init
    self.target.init()
  File "/home/brix/.local/lib/python3.8/site-packages/pyocd/core/coresight_target.py", line 160, in init
    seq.invoke()
  File "/home/brix/.local/lib/python3.8/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
  File "/home/brix/.local/lib/python3.8/site-packages/pyocd/core/coresight_target.py", line 299, in check_for_cores
    raise exceptions.DebugError("No cores were discovered!")
pyocd.core.exceptions.DebugError: No cores were discovered!
0001139:DEBUG:session:uninit session <pyocd.core.session.Session object at 0x7f68e2c52b20>
0001145:DEBUG:pyusb_backend:closing interface

```

flit, 2020-08-20:
When you get invalid CoreSight components like this (they're all zeroes, it seems), it usually means that a power domain is not powered or some similar situation. Unfortunately I'm not familiar with the debug configuration on the CM1 DesignStart, so I can't immediately help. I'll see if I can look into though.