pyocd/pyOCD #931: Unable to set breakpoints in vscode

farrenv, 2020-08-14:
I am developing in a virtual machine passing through DAPlink running on K20DX as debugger connected to an nRF52840 on the Agora target.  I am able to flash binaries to the Agora, and I am able to start and run a debugging session.  However, I am unable to set a breakpoint anywhere other than the first line in main().

Here is the debug output that is generated starting a debugging session until it reaches main()
```
1: (267) LaunchOptions<LocalLaunchOptions xmlns='http://schemas.microsoft.com/vstudio/MDDDebuggerOptions/2014'
1: (300) LaunchOptions  ExePath='/home/vagrant/Desktop/Projects/Examples/mbed-os-example-blinky/BUILD/EP_AGORA/GCC_ARM-DEBUG/mbed-os-example-blinky.elf'
1: (300) LaunchOptions  WorkingDirectory='/home/vagrant/Desktop/Projects/Examples/mbed-os-example-blinky'
1: (300) LaunchOptions  ExeArguments=''
1: (301) LaunchOptions  MIMode='gdb'
1: (301) LaunchOptions  MIDebuggerPath='/tools/gcc-arm-none-eabi-9-2020-q2/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gdb'
1: (301) LaunchOptions  WaitDynamicLibLoad='false'
1: (301) LaunchOptions  DebugServer='/usr/local/bin/pyocd-gdbserver'
1: (301) LaunchOptions  DebugServerArgs=''
1: (301) LaunchOptions  ServerStarted='GDB\ server\ started'
1: (301) LaunchOptions  FilterStderr='true'
1: (301) LaunchOptions  ServerLaunchTimeout='20000'
1: (301) LaunchOptions>
1: (301) LaunchOptions    <SetupCommands>
1: (301) LaunchOptions        <Command IgnoreFailures='false' Description='connect to target'>-target-select remote localhost:3333</Command>
1: (301) LaunchOptions        <Command IgnoreFailures='false' Description='load file'>-file-exec-and-symbols /home/vagrant/Desktop/Projects/Examples/mbed-os-example-blinky/BUILD/EP_AGORA/GCC_ARM-DEBUG/mbed-os-example-blinky.elf</Command>
1: (301) LaunchOptions        <Command IgnoreFailures='false' Description='reset monitor'>-interpreter-exec console "monitor reset"</Command>
1: (301) LaunchOptions        <Command IgnoreFailures='false' Description='halt monitor'>-interpreter-exec console "monitor halt"</Command>
1: (302) LaunchOptions        <Command IgnoreFailures='false' Description='flash target'>-target-download</Command>
1: (302) LaunchOptions    </SetupCommands>
1: (302) LaunchOptions</LocalLaunchOptions>
1: (419) Starting: "/usr/local/bin/pyocd-gdbserver"
1: (2237) Starting: "/tools/gcc-arm-none-eabi-9-2020-q2/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gdb" --interpreter=mi
1: (2242) DebuggerPid=29833
1: (2306) ->=thread-group-added,id="i1"
1: (2309) ->~"GNU gdb (GNU Arm Embedded Toolchain 9-2020-q2-update) 8.3.1.20191211-git\n"
1: (2309) ->~"Copyright (C) 2019 Free Software Foundation, Inc.\n"
1: (2315) ->~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
1: (2315) ->~"\nType \"show copying\" and \"show warranty\" for details.\n"
1: (2315) ->~"This GDB was configured as \"--host=x86_64-linux-gnu --target=arm-none-eabi\".\n"
1: (2316) ->~"Type \"show configuration\" for configuration details.\n"
1: (2317) ->~"For bug reporting instructions, please see:\n"
1: (2317) ->~"<http://www.gnu.org/software/gdb/bugs/>.\n"
1: (2317) ->~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
1: (2317) ->~"\n\n"
1: (2317) ->~"For help, type \"help\".\n"
1: (2318) ->~"Type \"apropos word\" to search for commands related to \"word\".\n"
1: (2318) ->(gdb)
1: (2323) <-1001-gdb-set target-async on
1: (2325) ->1001^done
1: (2325) ->(gdb)
1: (2328) 1001: elapsed time 6
1: (2338) <-1002-target-select remote localhost:3333
1: (2499) ->=thread-group-started,id="i1",pid="42000"
1: (2499) ->&"warning: No executable has been specified and target does not support\ndetermining executable automatically.  Try using the \"file\" command."
1: (2499) ->&"\n"
1: (2499) ->=thread-created,id="1",group-id="i1"
1: (2508) <-1003-thread-info 1
1: (2805) ->~"0x000055b4 in ?? ()\n"
1: (2805) ->*stopped,frame={addr="0x000055b4",func="??",args=[],arch="arm"},thread-id="1",stopped-threads="all"
1: (2805) ->1002^connected
1: (2805) ->(gdb)
1: (2816) 1002: elapsed time 477
1: (2817) <-1004-file-exec-and-symbols /home/vagrant/Desktop/Projects/Examples/mbed-os-example-blinky/BUILD/EP_AGORA/GCC_ARM-DEBUG/mbed-os-example-blinky.elf
1: (2832) ->1003^done,threads=[{id="1",target-id="Thread 1",frame={level="0",addr="0x000055b4",func="??",args=[],arch="arm"},state="stopped"}]
1: (2832) ->(gdb)
1: (2835) 1003: elapsed time 326
1: (3688) ->1004^done
1: (3688) ->(gdb)
1: (3689) 1004: elapsed time 871
1: (3689) <-1005-interpreter-exec console "monitor reset"
1: (3744) ->1005^done
1: (3744) ->(gdb)
1: (3744) 1005: elapsed time 55
1: (3744) <-1006-interpreter-exec console "monitor halt"
1: (3749) ->1006^done
[... 183 lines cut ...]
<--   C (variables-13): {"command":"variables","arguments":{"variablesReference":1000},"type":"request","seq":13}

```

Here is the output generated when I attempt to add a breakpoint to line 19:
[image]

```
<--   C (setBreakpoints-15): {"command":"setBreakpoints","arguments":{"source":{"name":"main.cpp","path":"/home/vagrant/Desktop/Projects/Examples/mbed-os-example-blinky/main.cpp","sources":[],"checksums":[]},"lines":[19],"breakpoints":[{"line":19}],"sourceModified":false},"type":"request","seq":15}
1: (419579) <-1027-break-insert -f main.cpp:19
1: (419580) ->&"No line 19 in file \"main.cpp\".\n"
1: (419580) ->1027^done,bkpt={number="2",type="breakpoint",disp="keep",enabled="y",addr="<PENDING>",pending="main.cpp:19",times="0",original-location="main.cpp:19"}
1: (419580) ->(gdb)
1: (419581) 1027: elapsed time 1
1: (419585) Send Event AD7BreakpointErrorEvent
```

I've searched for why gdb would report "No line x in file".  I tried adding -ggdb to the list of compiler flags but that had no effect.  I reduced the compiler optimization from -Og to -O0.  This actually allowed me to set breakpoints, but mbed crashes after one pass through the while loop.  I think there may be an issue with sleep and the 0 optimization.  

Host OS: Linux KDE Plasma Virtual Machine
Embedded OS: mbed-os-6.2.0
IDE: vscode (1.47.3)
Debugger: DAPLink on Flidor development board
Target: Agora
pyocd: 0.27.2

farrenv, 2020-08-14:
@flit @trowbridgec @AGlass0fMilk @maclobdell

farrenv, 2020-08-14:
While attempting to get Jlink to work I added a printf to the code.  I was running arm-none-eabi-gdb manually when I set a breakpoint at line 19 as before.  The debugger became trapped at line 21 where I had added the printf.  I went back and tried pyocd and it was happy to set and find the breakpoint on line 21.  I believe this issue can be closed. Thanks!
[image]

closed 2020-08-14 by farrenv

loverdeg, 2020-08-18:
Use a container