pyocd/pyOCD #932: Can't start debug sesions anymore with Mbed Studio 1.1.0 on Windows 10

gmgallo, 2020-08-14:
Hi, I’m using Mbed Studio 1.1.0 for Windows (10) with the CY8CPROTO-062-434W development board and Mbed OS 6.2. Over the last two weeks I successfully loaded, modified and programmed several examples that worked fine. But, after a couple of short debug sessions,  the debugger stop working.  First it didn't start and I got a message saying that it was a timeout waiting for the debug server to start. After restarting the computer the debug session crashes more catastrophically. I uninstalled and then reinstalled Mbed Studio 1.1.0, but the problem persists. here is the long trace output printed on the debug console:
[--------------------------------------------------------------------------------------------------------------]
`Selected port 50000 for debugging
Exception ignored in: 
Traceback (most recent call last):
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\usb\_objfinalizer.py", line 84, in __del__
    self.finalize()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\usb\_objfinalizer.py", line 144, in finalize
    self._finalizer()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\weakref.py", line 548, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\usb\_objfinalizer.py", line 104, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\usb\backend\libusb1.py", line 604, in _finalize_object
    _lib.libusb_unref_device(self.devid)
OSError: exception: access violation writing 0x0000000000000024
0000679:INFO:board:Target type is cy8c6xxa
0000755:INFO:dap:DP IDR = 0x6ba02477 (v2 rev6)
0000762:INFO:ap:AHB-AP#0 IDR = 0x84770001 (AHB-AP var0 rev8)
0000767:INFO:ap:AHB-AP#1 IDR = 0x84770001 (AHB-AP var0 rev8)
0000779:INFO:ap:AHB-AP#2 IDR = 0x24770011 (AHB-AP var1 rev2)
Exception ignored in: 
Traceback (most recent call last):
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\usb\_objfinalizer.py", line 84, in __del__
    self.finalize()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\usb\_objfinalizer.py", line 144, in finalize
    self._finalizer()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\weakref.py", line 548, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\usb\_objfinalizer.py", line 104, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\usb\backend\libusb1.py", line 604, in _finalize_object
    _lib.libusb_unref_device(self.devid)
OSError: exception: access violation writing 0x0000000000000024
0000803:CRITICAL:__main__:uncaught exception: 
Traceback (most recent call last):
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\__main__.py", line 362, in run
    self._COMMANDS[self._args.cmd](self)
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\__main__.py", line 680, in do_gdbserver
    with session:
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\core\session.py", line 302, in __enter__
    self.open()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\core\session.py", line 420, in open
    self._board.init()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\board\board.py", line 85, in init
    self.target.init()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\core\coresight_target.py", line 160, in init
    seq.invoke()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\utility\sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\utility\sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\utility\sequencer.py", line 208, in invoke
    resultSequence = call()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\coresight\ap.py", line 1021, in find_components
    super(AHB_AP, self).find_components()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\utility\concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\coresight\ap.py", line 649, in find_components
    cmpid.read_id_registers()
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\coresight\rom_table.py", line 111, in read_id_registers
    regs = self.ap.read_memory_block32(self.top_address + self.IDR_READ_START, self.IDR_READ_COUNT)
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\utility\concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\coresight\ap.py", line 972, in _read_memory_block32
    resp += self._read_block32_page(addr, n//4)
  File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\coresight\ap.py", line 930, in _read_block32_page
    resp = self.dp.read_ap_multiple(self.address.address + self._reg_offset + MEM_AP_DRW, size)
      File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\coresight\dap.py", line 664, in read_ap_multiple
        return read_ap_multiple_cb()
      File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\coresight\dap.py", line 654, in read_ap_multiple_cb
        return result_cb()
      File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 315, in read_ap_repeat_callback
        return result()
      File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 871, in reg_read_repeat_cb
        res = transfer.get_result()
      File "c:\ProgramData\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 142, in get_result
        assert not self.daplink._crnt_cmd.get_empty()
    AssertionError
"GDB server stopped unexpectedly with exit code 1"
`

flit, 2020-08-20:
cc @thegecko

gmgallo, 2020-08-20:
Is there any registry or settings somewhere to manually cleanup to start with a clean installation again? I really need this debugger now!

dingyifei, 2020-09-06:
I have the same issue when running `pyocd list`. My PyOCD version is 0.27.3 and it's running on Anaconda python 3.7 with `libusb-1.0.dll` in the anaconda3 folder.

This issue does not occur when my debug probe is removed.

Output sample
```
(base) PS C:\Users\13676> pyocd list
  #   Probe       Unique ID
--------------------------------------------
  0   STLINK-V3   001C00393137510A39383538
Exception ignored in: <function _AutoFinalizedObjectBase.__del__ at 0x000002AF49673E58>
Traceback (most recent call last):
  File "C:\Users\13676\AppData\Roaming\Python\Python37\site-packages\usb\_objfinalizer.py", line 84, in __del__
    self.finalize()
  File "C:\Users\13676\AppData\Roaming\Python\Python37\site-packages\usb\_objfinalizer.py", line 144, in finalize
    self._finalizer()
  File "C:\ProgramData\Anaconda3\lib\weakref.py", line 572, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
  File "C:\Users\13676\AppData\Roaming\Python\Python37\site-packages\usb\_objfinalizer.py", line 104, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "C:\Users\13676\AppData\Roaming\Python\Python37\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "C:\Users\13676\AppData\Roaming\Python\Python37\site-packages\usb\backend\libusb1.py", line 604, in _finalize_object
    _lib.libusb_unref_device(self.devid)
OSError: exception: access violation writing 0x0000000000000024
(base) PS C:\Users\13676> pyocd --version
0.27.3
(base) PS C:\Users\13676> pyocd list
No available debug probes are connected
(base) PS C:\Users\13676> pip --version
pip 20.1.1 from C:\ProgramData\Anaconda3\lib\site-packages\pip (python 3.7)
```