pyocd/pyOCD #938: debug pyOCD with pdb/pudb before install

cederom, 2020-08-20:
Hello world :-)

I would like to try and debug my local changes to pyOCD even before package is created and installed with `setup.py`. How can I do that? :-)

Any hints welcome :-)

cederom, 2020-08-20:
Okay, I had to create a helper script in top level pyOCD source directory where I can specify what module I want to start:

```
#!/usr/bin/env python
import pyocd.tools.pyocd
p=pyocd.tools.pyocd.PyOCDTool()
p.run()
```

Then I call that script with `pudb3 debug.py some_parameters`.

I can now put `import pudb; pudb.set_trace()` wherever I want to inervene and I just jump into the soruce code and variables browser provided by `pudb` :-)

If you know any better way please let me know :-)

flit, 2020-08-20:
Hi @cederom 😄 

I've been using VSCode for debugging pyOCD with great success. It's so much more efficient (for me, at least) than command line debugging.

Are you trying to debug the install process? I guess not since you're not debugging `setup.py`. So the question is why you need to debug before the package is installed?

If you're not already, use an editable install: `pip install -e .` (from the pyocd working copy root)

Here's an example config for VSCode `launch.json`. This assumes an editable install into a virtualenv.

```json
        {
            "name": "pyocd gdb",
            "type": "python",
            "request": "launch",
            "module": "pyocd.__main__",
            "console": "integratedTerminal",
            "stopOnEntry": false,
            "pythonPath": "${workspaceFolder}/sandbox/bin/python3",
            "showReturnValue": true,
            "args": ["gdb", "-Svv"]
        },
```

In case where pyOCD is being called by another tool, you can use the technique presented in #471. This then lets you attach VSCode to a running instance.

(I intend to add all this to the docs at some point!)

cederom, 2020-08-21:
Thank you @flit :-) I avoid Microsoft crap by any means necessary :-) I only use BSD Unix, ZSH, TMux, Vim, some GNU stuff, and basically only Open-Source utilities :-)

I just wanted to quickly debug the USB HID driver detach issue step by step before install.. and it worked :-) With time and your hints I will probably get a smarter more versatile approach. When its done I can put that into the documentation if you like. Thanks again :-)

cederom, 2021-05-12:
Hello world :-) I have came back again here regarding https://github.com/pyocd/pyOCD/issues/1144. I have created and provided simple helper script to allow pyOCD debug directly in Unix shell - even in a situation when exception occurs during pyocd import (simple `pudb3 pyocd` does not work in this case). Not to waste time a simple shell script can be launched to jump into the curses based python debugger :-)

Here is the solution: https://github.com/pyocd/pyOCD/pull/1151

cederom, 2021-05-12:
Whoops. Internals of pyOCD changed a lot recently. I have updated the script. With this kind of reorganization why not a major number bump (i.e. `1.0` instead `0.30`)? :-)