pyocd/pyOCD #945: Error with CMSIS DAP DIY

sabas1080, 2020-09-02:
Hi 

I am making a programmer with CH552 and I am testing your great tool but I am have the next error, in openocd work fine, and with `pyocd list` 

[image]

the error

```
➜  ~ pyocd erase -t nrf52840 --chip
0001536:CRITICAL:__main__:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/cmsis_dap_probe.py", line 147, in open
    self._link.open()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/utility/concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 592, in open
    self._packet_count = self._protocol.dap_info(self.ID.MAX_PACKET_COUNT)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/cmsis_dap_core.py", line 138, in dap_info
    raise DAPAccessIntf.DeviceError()
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.DeviceError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/__main__.py", line 362, in run
    self._COMMANDS[self._args.cmd](self)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/__main__.py", line 556, in do_erase
    with session:
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/core/session.py", line 302, in __enter__
    self.open()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/core/session.py", line 416, in open
    self._probe.open()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/cmsis_dap_probe.py", line 167, in open
    six.raise_from(self._convert_exception(exc), exc)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.ProbeError

```
Thanks

flit, 2020-09-23:
(Terribly sorry I missed this issue until now!)

The exception that is being raised here is due to the [DAP_Info](https://arm-software.github.io/CMSIS_5/DAP/html/group__DAP__Info.html) command not returning a valid packet. Specifically, the first byte of the response packet must be the command ID, and this exception is raised because the first byte does not match the DAP_Info command ID.

Bunch of questions:

- What host system? (Linux on PC, Windows, Mac, RasPi, etc)
- What implementation of CMSIS-DAP is the probe running?
- Is it based on the official CMSIS-DAP reference code?
- Is the probe firmware open source so I can take a look?

It's going to be very hard to diagnose and fix the issue without very detailed information (i.e. code) about the firmware, and, more importantly, USB bus traces. Do you happen to have a USB bus tracer available? 😀  There are also software solutions for Linux (e.g., usbmon) and Windows (several), but software bus trace is in my experience less reliable and quite limited (since it shows the OS view, not the actual bus view).

sabas1080, 2020-09-23:
Hi @flit 
thanks for answering, answering your questions

- What host system? (Linux on PC, Windows, Mac, RasPi, etc)
Mac
- What implementation of CMSIS-DAP is the probe running?
- Is it based on the official CMSIS-DAP reference code?
https://github.com/ElectronicCats/DAP-Cat-Programmer/tree/master/firmware/CMSIS_DAP

- Is the probe firmware open source so I can take a look? Yes, and hardware https://github.com/ElectronicCats/DAP-Cat-Programmer

closed 2020-12-14 by sabas1080