pyocd/pyOCD #949: pyocd cannot flash .bin if there only flash alogrithm defined in .pdsc

Hoohaha, 2020-09-09:
The DFP packs works well with keil, but use pyocd it report no identifiable boot memory.
Looks pyocd cannot recognize the flash algo as flash memory.

Device family definition in DFP pack:
```xml
  <devices>
    <family Dfamily="LPC54018M" Dvendor="NXP:11">
      <description>The LPC540xx are ARM Cortex-M4 based microcontrollers for embedded applications. These devices include an ARM Cortex-M4 coprocessor, up to 360 KB of on-chip SRAM, full-speed and high speed USB device interface with Crystal-less operation, a DMIC subsystem with PDM microphone interface and I²S, five general-purpose timers, one SCTimer/PWM, one RTC/alarm timer, one 24-bit Multi-Rate Timer (MRT), a Windowed Watchdog Timer (WWDT), eight flexible serial communication peripherals (each of which can be a USART, SPI, or I²C interface), and one 12-bit 5.0 Msamples/sec ADC, and a temperature sensor. The ARM Cortex-M4 is a 32-bit core that offers system enhancements such as low power consumption, enhanced debug features, and a high level of support block integration. The ARM Cortex-M4 CPU incorporates a 3-stage pipeline, uses a Harvard architecture with separate local instruction and data buses as well as a third bus for peripherals, and includes an internal prefetch unit that supports speculative branching. The ARM Cortex-M4 supports single-cycle digital signal processing and SIMD instructions. A hardware floating-point unit is integrated in the core.</description>
      <device Dname="LPC54018M">
        <processor Dcore="Cortex-M4" Dfpu="SP_FPU" Dmpu="MPU" Dendian="Little-endian" Dclock="180000000"/>
        <environment name="iar">
          <file category="linkerfile" name="iar/LPC54018M_spifi_flash.icf"/>
        </environment>
        <memory name="SRAMX" start="0x00000000" size="0x030000" access="rw" default="1"/>
        <memory name="SRAM_0_1_2_3" start="0x20000000" size="0x028000" access="rw" default="1"/>
        <memory name="USB_RAM" start="0x40100000" size="0x2000" access="rw" default="1"/>
        <algorithm name="arm/LPC540xx_W25Q32JVWJX.FLM" start="0x10000000" size="0x00400000" RAMstart="0x20000000" RAMsize="0x00001000" default="1" boot="1"/>
        <debug svd="LPC54018M.xml"/>
        <variant Dvariant="LPC54018J2MET180">
          <compile header="fsl_device_registers.h" define="CPU_LPC54018J2MET180"/>
        </variant>
        <variant Dvariant="LPC54018J4MET180">
          <compile header="fsl_device_registers.h" define="CPU_LPC54018J4MET180"/>
        </variant>
      </device>
    </family>
  </devices>
```

```
$pyocd flash hello_world.bin --pack C:/NXP/LPC54018M_DFP.12.1.0.pack -t LPC54018J2MET180
0001293:WARNING:cmsis_pack:CMSIS-Pack device LPC54018J2MET180 has no identifiable boot memory
0001643:CRITICAL:__main__:No boot memory is defined for this device
Traceback (most recent call last):
  File "c:\mycode\pyocd\pyocd\__main__.py", line 398, in run
    self._COMMANDS[self._args.cmd](self)
  File "c:\mycode\pyocd\pyocd\__main__.py", line 575, in do_flash
    file_format=self._args.format)
  File "c:\mycode\pyocd\pyocd\flash\file_programmer.py", line 157, in program
    self._format_handlers[file_format](file_obj, **kwargs)
  File "c:\mycode\pyocd\pyocd\flash\file_programmer.py", line 170, in _program_bin
    raise exceptions.TargetSupportError("No boot memory is defined for this device")
TargetSupportError: No boot memory is defined for this device
```

```shell
$ pyocd flash hello_world.bin --pack C:/NXP/LPC54018M_DFP.12.1.0.pack -t LPC54018J2MET180 -a 0x10000000
0001065:WARNING:cmsis_pack:CMSIS-Pack device LPC54018J2MET180 has no identifiable boot memory
0001417:CRITICAL:__main__:no memory region defined for address 0x10000000
Traceback (most recent call last):
  File "c:\mycode\pyocd\pyocd\__main__.py", line 398, in run
    self._COMMANDS[self._args.cmd](self)
  File "c:\mycode\pyocd\pyocd\__main__.py", line 575, in do_flash
    file_format=self._args.format)
  File "c:\mycode\pyocd\pyocd\flash\file_programmer.py", line 157, in program
    self._format_handlers[file_format](file_obj, **kwargs)
  File "c:\mycode\pyocd\pyocd\flash\file_programmer.py", line 176, in _program_bin
    self._loader.add_data(address, data)
  File "c:\mycode\pyocd\pyocd\flash\loader.py", line 115, in add_data
    raise ValueError("no memory region defined for address 0x%08x" % address)
ValueError: no memory region defined for address 0x10000000
```

flit, 2020-09-24:
The DFP is missing a `<memory>` element for the flash at 0x1000_0000.

It makes sense to create a memory region when there is an `<algorithm>` defined that has no matching `<memory>`, as long as it doesn't overlap with another memory region. This is actually already on my todo list, just haven't had time to get to it. If you have time to help it would be greatly appreciated 😄 .

LONGZR007, 2020-10-15:
I wonder how this should be solved?If I were to add the <memory> element to my local pack, such a change would not make sense on pyOCD. @flit

flit, 2020-10-18:
Well, the _best_ way would be to implement the fix in pyOCD so it creates a memory region for an algorithm when there isn't one, then submit a pull request for the change. 😁 

If you modify your local pack, you can tell pyOCD to use it by using the `--pack` argument or `pack` session option.