pyocd/pyOCD #951: Unlock and lock ST series chips

LONGZR007, 2020-09-11:
Why not support ST series chip unlock and lock? When can you support it?

flit, 2020-09-11:
When can you provide a pull request to add it? 😉  

Seriously, pyOCD, like all open source projects, thrives from community contributions. So if you have a feature you need, please think about implementing it yourself. If you have questions or need help, just ask!

LONGZR007, 2020-09-12:
I did this: https://github.com/mbedmicro/pyOCD/blob/master/docs/adding_new_targets.md, according to the instructions below to increase a new target,Because I needed to use a different FlashAlgo when I wanted to read and write the option bytes of STM32 than When I programmed FLASH, I thought of adding a new target, but I found that one target only corresponds to one FlashAlgo,
So I don't know how to add FlashAlgo that reads and writes STM32 option bytes to pyOCD, should I merge them into a FlashAlgo?Or am I implementing the wrong idea?

In addition, if I have successfully added the option byte FlashAlgo for reading and writing STM32 to pyOCD, how can I associate it with locked and UNLOCK commands?

flit, 2020-09-22:
Really sorry for the slow reply! Will get back to you shortly.

LONGZR007, 2020-09-25:
I have added opt-related algorithms to the download algorithm of STM32. I can use unlock command to call mass_erase to unlock the chip, but the method of locking the chip is different for each chip. I think this requires a new framework to add locking methods for different chips.If yes can you write the frame well?I'm not very familiar with Python yet, so I'm working hard on it.

LONGZR007, 2020-09-26:
Hi，@flit， I am looking forward to your reply

flit, 2020-09-27:
It looks like the CMSIS Packs for STM32 families include flash algos for the OPT memories. For example, `Keil.STM32L4xx_DFP.2.4.0.pack` has `CMSIS/Flash/STM32L4xx_OTP.FLM` and several others. The device descriptions in that pack's .pdsc define `<algorithm>` elements for the OPT regions, but there are two issues:

1. The `<algorithm>` elements have the `default` attribute set to 0, which causes pyOCD to ignore the it.
2. There is not a corresponding `<memory>` attribute for the OPT region. pyOCD currently requires matching `<memory>` and `<algorithm>` definitions.

Imo, the best way to support OPT programming for STM32 is to fix these two issues.

For issue #1, there needs to be a way to let users enable a non-default flash algo (`default` attribute is 0). You can't just create regions for all non-default algos. Often there are multiple non-default algos defined for different types of off-chip memories, that would conflict if they were all enabled. See, for instance, the list of algos immediately after the `<family>` element in the Keil.STM32L4xx_DFP pack. The difficult question is how to do this in a command-line environment. You need some way for a user to list available non-default algos, probably either an extension to `pyocd list` or another subcommand. Then probably a session option to enable named algos.

Places in the code that need to be changed to implement fixes for the above issues:

- `pyocd/target/pack/cmsis_pack.py` in the `CmsisPackDevice` class, around the methods that create memory regions from PDSC elements.
- `pyocd/coresight/coresight_target.py` in `CoreSightTarget.create_flash()`.
- `pyocd/__main__.py` for subcommand changes.
- `pyocd/tools/lists.py` for listing non-default algos.

(See the [CMSIS-Pack docs](https://arm-software.github.io/CMSIS_5/Pack/html/index.html) for the specification.)