pyocd/pyOCD #963: Couldn't detect Atmel CMSIS-DAP for SAMC21 on pyocd 0.27.3

PiyushJadhav, 2020-09-22:
Hi, I faced some glitches to connect Atmel EDBG CMSIS-DAP probe on the latest pyocd but reverting back the version to 0.23.0 I could get the probe detected. I thought it is worth raising issue for this. 

Details - Recently I was connecting linux machine (raspberry) to perform some continuous integration on embedded development. We had used pyocd 0.23.0 version in the past but on NUCLEO-F091RC [stm32f091rc]. This time I started with the latest version 0.27.3 but for ATSAMC21J18A (which is not built in, but can be added via `pyocd packs --install ATSAMC21J18A` ). After then I set the appropriate udev rules for the Atmel probe. But I could not get the probe detected. The same results when I reverted back to 0.24.0 (which was the earliest but backward compatible). Something wierd but the NUCLEO-F091RC [stm32f091rc] was detected in both the versions.
Then I thought to revert back to 0.23.0 and change the python APIs to be compatible with this version and voila! the probes are detected.
Hope the info provided is enough to investigate. I'm happy to share further if needed.

flit, 2020-09-22:
Definitely thanks for creating this issue, it's not something we were aware of.

The issue may be related to some changes to checks for the CMSIS-DAP names. Fortunately I have an Atmel board (though not SAMC2) so I should be able to reproduce.

patrislav1, 2020-11-03:
I just made a pull request [for Atmel-ICE support](https://github.com/pyocd/pyOCD/pull/991), maybe it fixes this issue as well?

PR #991 (2020-11-03): Fix Atmel-ICE compatibility issue
The `MatchCmsisDapv1Interface` class, which is used for finding CMSIS-DAPv1 interfaces, expects the first endpoint of the probe to be an _interrupt in_, the second one (optional) to be an _interrupt out_.
The Atmel-ICE CMSIS-DAP probe (`03eb:2141`) has its endpoints laid out the other way around: first one _interrupt out_, second one _interrupt in_ - so it fails this test.

The code in `pyusb_backend.py` that uses the probe, doesn't actually care about the sequence order of endpoints - it [just uses any "in" endpoint as ep_in and the other one as ep_out](https [... cut]
closed 2020-11-03 by flit