pyocd/pyOCD #973: Target type is nrf52840 No ACK received

symeonmattes, 2020-10-05:
Hi,

I’m using the Agora board [Agora Board](https://os.mbed.com/platforms/AGORA-DEV/). I'm trying to setup the Mbed Studio: 1.2.1 on debian 10, however I receive the error "PyOCD deploy failed, deploying using mass storage device on Agora Board". Some info on the issue I have written on [Mbed Studio](https://forums.mbed.com/t/pyocd-deploy-failed-deploying-using-mass-storage-device-on-agora-board/10378)

My System has the following setup:
```
gcc-arm-none-eabi-9-2019-q4-major
debian 10 
mbed 1.10.4
pyocd 0.28.0
```

I have been checking around trying to understand the error, and it seems that the pyocd cannot connect to my device. By executing **pyocd list** I receive:

` 0   ep_agora [nrf52840]   2600360253274e450012800df7c3002fd811000097969900 `

However when I'm using 
`pyocd erase -t nrf52840 --chip`
or
`pyocd flash -t nrf52840 mbed-os-example-blinky.hex 
`

I receive:
```
0000581:CRITICAL:__main__:No ACK received
Traceback (most recent call last):
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/probe/cmsis_dap_probe.py", line 250, in read_dp
    result = self._link.read_reg(reg_id, now=now)
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 839, in read_reg
    return read_reg_cb()
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 833, in read_reg_cb
    res = transfer.get_result()
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 143, in get_result
    self.daplink.flush()
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/utility/concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 667, in flush
    self._read_packet()
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/utility/concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 918, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 449, in decode_data
    data = self._decode_transfer_block_data(data)
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 418, in _decode_transfer_block_data
    self._check_response(data[3])
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 341, in _check_response
    raise DAPAccessIntf.TransferError("No ACK received")
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferError: No ACK received

The above exception was the direct cause of the following exception:

Traceback (most recent call last):

  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/__main__.py", line 398, in run
    self._COMMANDS[self._args.cmd](self)
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/__main__.py", line 568, in do_flash
    with session:
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/core/session.py", line 343, in __enter__
    self.open()
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/core/session.py", line 461, in open
    self._board.init()
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/board/board.py", line 85, in init
    self.target.init()
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/core/soc_target.py", line 120, in init
    seq.invoke()
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/coresight/dap.py", line 303, in _connect
    connector.connect(self._protocol)
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/coresight/dap.py", line 167, in connect
    self._idr = self.read_idr()
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/coresight/dap.py", line 194, in read_idr
    dpidr = self._probe.read_dp(DP_IDR, now=True)
  File "/home/symeon/.local/lib/python3.7/site-packages/pyocd/probe/cmsis_dap_probe.py", line 252, in read_dp
    six.raise_from(self._convert_exception(error), error)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.TransferError: No ACK received
```

trowbridgec, 2020-10-08:
@flit Do you have any insight here?

symeonmattes, 2020-10-09:
Hi,

I have some more logs in case they are useful:

In https://github.com/pyocd/pyOCD/blob/master/udev/README.md. it says about some dev rules that could help although they are installed automatically by pyocd. I have run dmesg, as requested and when plug in the board I receive:

[ 4208.436839] usb 2-2: new full-speed USB device number 4 using xhci_hcd
[ 4208.591366] usb 2-2: New USB device found, idVendor=0d28, idProduct=0204, bcdDevice=10.00
[ 4208.591370] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4208.591373] usb 2-2: Product: DAPLink CMSIS-DAP
[ 4208.591375] usb 2-2: Manufacturer: ARM
[ 4208.591377] usb 2-2: SerialNumber: 2600360253274e450012800df7c3002fd811000097969900
[ 4208.593844] usb-storage 2-2:1.0: USB Mass Storage device detected
[ 4208.594181] scsi host3: usb-storage 2-2:1.0
[ 4208.594831] cdc_acm 2-2:1.1: ttyACM0: USB ACM device
[ 4208.598358] hid-generic 0003:0D28:0204.0003: hiddev0,hidraw0: USB HID v1.00 Device [ARM DAPLink CMSIS-DAP] on usb-0000:00:14.0-2/input3
[ 4209.613609] scsi 3:0:0:0: Direct-Access     MBED     VFS              0.1  PQ: 0 ANSI: 2
[ 4209.614268] sd 3:0:0:0: Attached scsi generic sg2 type 0
[ 4209.614425] sd 3:0:0:0: [sdb] 131200 512-byte logical blocks: (67.2 MB/64.1 MiB)
[ 4209.614598] sd 3:0:0:0: [sdb] Write Protect is off
[ 4209.614600] sd 3:0:0:0: [sdb] Mode Sense: 03 00 00 00
[ 4209.614766] sd 3:0:0:0: [sdb] No Caching mode page found
[ 4209.614771] sd 3:0:0:0: [sdb] Assuming drive cache: write through
[ 4209.627580]  sdb:
[ 4209.628714] sd 3:0:0:0: [sdb] Attached SCSI removable disk

I checked a little bit the code of pyocd on the line that I receive the error of "No ACK received" and I receive the error on "~/.local/lib/python3.7/site-packages/pyocd/probe/cmsis_dap_probe.py"
    def read_dp(self, addr, now=True):
        reg_id = self.REG_ADDR_TO_ID_MAP[self.DP, addr] # it gives REG.DP_0x0
        try:
            result = self._link.read_reg(reg_id, now=now) # here is the error

It seems as if there is no communication with the board.

flit, 2020-10-09:
pyOCD doesn't install the udev rules automatically, though Mbed Studio might. But it's clear that's not an issue—you wouldn't even see the board in `pyocd list`.

You are correct that there is no communication with the board. The "No ACK received" error means that there was no response at all from the SWD request. The exception log says this happened on the very first access (trying to read the DP's ID register).

This most often means that the device is asleep. Please make sure you are using a debug build of your code that disables sleep.

You might try change to the under-reset connect mode by passing `--connect=under-reset` on the command line. This only works if the MCU has a hardware nRESET signal. It forces the device to wake by reset as pyOCD connects.

symeonmattes, 2020-10-10:
Hi @flit,

You said:

> This most often means that the device is asleep. Please make sure you are using a debug build of your code that disables sleep.

To be honest I'm not sure what it means what you said. I done the following steps:
1. git clone git@github.com:ARMmbed/mbed-os-example-blinky.git
2. mbed compile -m EP_AGORA -t GCC_ARM --flash 
This compiles the software and flashes it onto the board correctly...it led is blinking for some seconds and then stops
3. I just want to do debugging, i.e. from an IDE put a breakpoint and see what I have in the env variables.
4. I have tried diffferent IDEs with no success, and in all I have receiving that PYOCD No ACK received. That's why I have tried to find why pyocd is as such.

I have tried the 

> You might try change to the under-reset connect mode by passing --connect=under-reset on the command line. This only works if the MCU has a hardware nRESET signal. It forces the device to wake by reset as pyOCD connects.

But again I receive the same error.

Thanks

flit, 2020-10-10:
The M-profile architecture defines two architectural levels of low power mode called sleep and deep sleep. The silicon vendor then maps these modes to system level power modes, usually with additional levels and controls. See the nRF RM for details. The WFI/WFE instruction is then used to enter a sleep mode. Debug will (for almost all MCUs) be disabled when the MCU is asleep in the lower power modes. The only way to connect debug to a device in this mode is for the device to somehow wake up, either via the usual firmware-controlled methods (timer, pin wake, etc) _or via external pin reset_.

The default Mbed build does not enable debug. You need to build with the debug profile both to enable debugging, _and_ to disable automatic sleep on idle. See the Mbed docs for more.

Connecting under reset does not work most likely because the nRF52840 does not by default expose the nRESET signal on a pin. See the `UICR.PSELRESETn` registers, page 45 of nRF52840_PS_v1.1.pdf.

Assuming (a) the mbed blinky or similar firmware is already loaded, (b) the device is actually asleep in a low power mode that disables debug, (c) the nRESET pin is not enabled, the only way to gain control of the device is to attempt to connect during the time when the firmware wakes up to blinky the LED, or as the MCU powers on. @trowbridgec May have other suggestions.

symeonmattes, 2020-10-17:
I have tried the same thing on Windows 10 machine with the following setup:
- python 3.6.8
- pip 20.2.3
- mbed-cli 1.10.4
- pyocd 0.28.2

but I still receive the same error

symeonmattes, 2020-10-18:
Hi @flit,

I checked a little bit the board  [flidor](https://os.mbed.com/components/Flidor/) and it seems there are some pins that I'm not sure if they could help:

- Target Reset (It's supposed to reset the device and rerun the previous flashed software)
- Besed Target Reset there are 3 pins which says (J3) ISO, RESET ISOLATION, BYP 

I connect my computer through USB to the "DHD USB"

flit, 2020-10-18:
Just from the name, Target Reset seems like the hardware nRESET pin. But as mentioned above, unless the nRF's UICR is programmed this pin is not enabled (muxed to GPIO by default).

@trowbridgec or someone else from EP will have to comment further on the pin functions.

symeonmattes, 2020-10-21:
@flit 

It was a jamber J16. It had to go to USB mode than Jacks mode.

Thanks

closed 2020-10-21 by symeonmattes

flit, 2020-10-22:
Sorry, what is a jamber J16? (I googled and didn't find anything. 😓 )

symeonmattes, 2020-10-23:
Hi @flit 

In [flidor](https://os.mbed.com/components/Flidor/) there is a J16 connector with 3 pins. The first two are supposed to be for Jacks and the last two for USB. It was in Jacks and I changed it to USB.

However I'm not sure what it does exactly...it just worked.

trowbridgec, 2020-10-23:
@symeonmattes J16 on the 'Flidor' dev board (see more info here: https://os.mbed.com/components/Flidor/) controls the input voltage source for the target board. We designed the dev board to be flexible enough to allow a developer to source power either from the USB port (J4) or from any of the external power interfaces (J001, J7, J8, J9, J10).

Sorry for the confusion on this and glad you were able to move forward!

flit, 2020-10-23:
Thanks for the explanation.

smrtos, 2022-03-20:
Is there any software method to re-enable the connection except hardware unplugging and plugging again?

flit, 2022-04-08:
@smrtos Sorry, what connection are you trying to reenable through software—the SWD connection? Just make sure you are developing with any sleep mode disabled (to prevent sleep related problems in the first place), or are connecting under reset to wake the device, and that the reset pin is physically connected to the debug probe. I don't know anything about the reel_board specifics, so I can't help there.

smwikipedia, 2025-04-22:
> pyOCD doesn't install the udev rules automatically, though Mbed Studio might. But it's clear that's not an issue—you wouldn't even see the board in `pyocd list`.
>
> You are correct that there is no [...]

Indeed. I got below error when connecting a reel board:
```
ming@ming-NUC01:~/dev$ pyocd commander --target nrf52840  -v
0000340 W Board ID OSHW is not recognized [mbed_board]
0000342 I Target type is nrf52840 [board]
0000491 E Error while initing target: No ACK received [commander]
```
Turns out I forgot to switch on the reel board. After switching it on,  it works:
```
ming@ming-NUC01:~/dev$ pyocd commander --target nrf52840  -v
0000354 W Board ID OSHW is not recognized [mbed_board]
0000356 I Target type is nrf52840 [board]
0000456 I DP IDR = 0x2ba01477 (v1 rev2) [dap]
0000495 I AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2) [discovery]
0000497 I AP#1 IDR = 0x02880000 (AP var0 rev0) [discovery]
0000499 I NRF52840 not in secure state [target_nRF52]
0000508 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=244 part=008) [rom_table]
0000513 I [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
0000516 I [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0000519 I [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=003> [rom_table]
0000522 I [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0000527 I [4]<e0040000:TPIU M4 class=9 designer=43b:Arm part=9a1 devtype=11 archid=0000 devid=ca1:0:0> [rom_table]
0000532 I [5]<e0041000:ETM M4 class=9 designer=43b:Arm part=925 devtype=13 archid=0000 devid=0:0:0> [rom_table]
0000538 I CPU core #0: Cortex-M4 r0p1, v7.0-M architecture [cortex_m]
0000538 I   Extensions: [DSP, FPU, FPU_V4, MPU] [cortex_m]
0000538 I   FPU present: FPv4-SP-D16-M [cortex_m]
0000546 I This appears to be an nRF52840 QIAA C0 [target_nRF52]
0000550 I 4 hardware watchpoints [dwt]
0000553 I 6 hardware breakpoints, 4 literal comparators [fpb]
Connected to NRF52840 [Sleeping]: OSHW000041114e45002e3006bcf000330571000097969900
```