pyocd/pyOCD #980: Gdbserver reports "overlapping regions in memory map" to Gdb for stm32l4 Devices

DrTobe, 2020-10-12:
When I use `pyocd gdbserver --target stm32l432kcux` to connect to my nucleo-l432kc board and connect `arm-none-eabi-gdb` with `target extended :3333` afterwards, I get the "warning: Overlapping regions in memory map: ignoring" message. Then, it is impossible to use `gdb` to flash the device.

The same issue is present on the nucleo-l412-rb-p board but pyocd's gdbserver works flawlessly with an stm32l071kbu mcu which I have available for comparison here. Since pyocd uses DFPs to handle different targets and the stm32 DFPs are organized according to the device families, I suspect that this issue could be present on (all) stm32l4 devices but not on stm32l0 devices, caused by the stm32l4 DFP.

I have skimmed the DFP for the L0 and the L4 families and found memory information in the two `.pdsc` files. Below, some lines that looked interesting to me:

`Keil.STM32L0xx_DFP.pdsc`
```xml
    <subFamily DsubFamily="STM32L071">
        [...]
        <!-- *************************  Device 'STM32L071KBUx'  ***************************** -->
        <device Dname="STM32L071KBUx">
          <memory id="IROM1"                                   start="0x08000000" size="0x00020000" startup="1" default="1"/>
          <memory id="IRAM1"                                   start="0x20000000" size="0x00005000" init   ="0" default="1"/>
          [...]
        </device>
```

`Keil.STM32L4xx_DFP.pdsc`
```xml
      <subFamily DsubFamily="STM32L432">
        [...]
        <memory id="IRAM1"                                 start="0x20000000" size="0x0000C000" init="0" default="1"/>
        <memory id="IRAM2"                                 start="0x10000000" size="0x00004000" init="0" default="0"/>
        [...]
        <!-- *************************  Device 'STM32L432KCUx'  ***************************** -->
        <device Dname="STM32L432KCUx">
          <memory name="Main_Flash"  access="rx"   id="IROM1"     start="0x08000000" size="0x00040000" startup="1" default="1"/>
          <memory name="SRAM"        access="rwx"  id="IRAM1"     start="0x20000000" size="0x0000C000" init="0"    default="1"/>
          [...]
        </device>
```

Apparently, the memory regions are actually defined twice for the L4 family, once in the `subFamily` tag, once in the `device` child tag. In the description for the L0 family, this information is not present in the `subFamily` tag so that the memory regions are only defined once there. So it seems to me that pyocd is merging the `memory` tags from `subFamily` and `device` and forwards this (duplicate) information to gdb when requested.

Since I do not know the specification which the `.pdsc` files have to adhere to, I do not know if this is a bug in pyocd or in the pack archive for the L4 family (I assume the latter). Anyways, it would be nice if pyocd could only report the memory regions once to gdb so that flashing works successfully afterwards. Maybe, duplicate memory regions could be identified and only kept once? This could, if desired, be accompanied by a warning to the user or it could be done silently, especially if the `id` fields of the `memory` tags match like in this case here. This (comparably small?) fix could fix a whole device family.

If there is anything I could do to help with this issue, please let me know. If desired, I would try to fix it myself but I would like to know if such a PR would be or if there are different plans to cope with this issue.

flit, 2020-10-18:
Thanks for looking into this (and the PR!). There are definitely some issues with the DFP region handling in pyocd.

Fyi, the most up to date PDSC spec is at https://arm-software.github.io/CMSIS_5/Pack/html/pdsc_family_pg.html#element_memory. (CMSIS is now managed on GitHub, so you can create issue and PRs.)

According to the Keil folks, the region names are supposed to be used to to override outer level regions. But in practice it seems the rules they use are more complicated. We definitely need to gracefully handle duplicate or overlapping regions by address. (Fwiw, I do think there are some bugs in the STM32 packs.)

DrTobe, 2020-10-21:
I was already expecting that this PR would just open up the discussion on how it has to be done properly, so let's discuss ...

- Overriding by region names: Where did you find that this is the preferred way to override outer level regions? I just saw that it did not work in my example case so I changed it to identify memory regions only by "required" attributes which seemed more reliable to me. But if using the region names actually is the preferred way of overriding regions, the bug is in the DFP and should be fixed there.
- Partly overlapping memory regions: Is this actually happening somewhere? If so, what to do then? I would definitely think that it is a bug if I stumbled upon something like that. So my best guess would be to print an error (warning) message from pyocd then and delete the outer level region. Other ideas?

flit, 2020-10-25:
For region names, it was from an internal Arm discussion (basically trying to resolve the same question you just filed for CMSIS_5) that region names were said to be the preferred way. And I thought there was a CMSIS_5 commit that improved the documentation, but I can't find it right now so I'll have to dig deeper.

However, there are definitely bugs, or at least inconsistencies, in some of the STM32 DFPs. One issue is the distinction between the `id` and `name` attributes. There are some STM32 DFP devices that have an outer memory region with an `id` but no `name`, that are apparently intended to be overridden by an inner region with a `name` but no `id`.

Overlapping regions might exist in cases where the same region is redefined with a different size due to family variants. In this case the start address will almost always be the same (I can imagine some cases where start would differ, but it would be rare). As long as the given regions have the same name (or id) then it is ok and the inner overrides as usual. Otherwise your proposal to log a warning and delete the outer region is exactly what I would suggest.

Ultimately, pyOCD needs to be smart enough to handle anything it sees and not produce invalid memory maps. So a combination of name/id, plus matching on attributes and handling overlaps.

flit, 2020-10-29:
That sounds good. And good point about multiple warnings; limited to a single warning per DFP is probably wise. (I've had to add similar limits in other parts of pyocd.)

DrTobe, 2020-10-30:
I have updated the pull request but I could not try if the identification of memory regions by name/id works properly because I do not know any DFP which would be affected by this logic. Do you know one so I could test it?