pyocd/pyOCD #981: Registering a new target

ksigurdsson, 2020-10-13:
I've defined a new (CoreSightTarget subclass) target and successfully tested it by following the instructions in adding_new_targets.md to add it to the list of builtin targets

Is there a way to import the new target source file without changing the contents of pyocd/target/builtin?

It would be nice if I could just point to this new file in the pyocd.yaml file in a similar way you can point to a CMSIS pack
e.g.
target:
  - <path_to>/target_<device>.py

flit, 2020-10-18:
Yeah, I totally agree it would be nice to easily add additional targets externally to pyocd. Both the option you suggested via config, as well as by dropping a target file in a specific directory.

Right now the best way to do this is by putting the new target module(s) in a directory with a `pyocd_user.py` script that looks like this:

```py
import sys
import os

# Add this directory to python path.
sys.path.append(os.path.dirname(os.path.abspath(__file__)))

# Add the new target.
import my_new_target
```

Each target module should include this (in addition to the target class, of course):

```py
from pyocd.target import TARGET
TARGET['my-new-target-name'] = MyNewTargetClass
```

(This could be done instead in the `pyocd_user.py` script.)

You may also wish to include a `pyocd.yaml` that sets `target_override` appropriately. Then you can set the `--project-directory` argument to the directory and everything will auto-load.

Not the simplest, but it works for now. There are plenty of variants of this technique.

ksigurdsson, 2020-11-06:
I've been able to add the new target as suggested. This works well.
Many thanks.