pyocd/pyOCD #982: LEDs are not controlled

mubes, 2020-10-14:
Hi there,

I can't see that the on probe LEDs for CONNECT and RUN (commanded via set_led) are ever actioned in the current code...is there any chance of adding those in?  Perfectly happy to help with testing if that's useful.

Regards

DAVE

flit, 2020-10-18:
Sure, that can be added. It won't be particularly high priority, but should be fairly straightforward to implement. Any chance you can help implement?

mubes, 2020-10-18:
Hi Chris,

Yeah, I'll add it to my to-dos, no problem.  Like you it's not a really 
high priority so we'll have a race to see who gets to it first :-)  I'm 
mostly concerned about state being inconsistent if the connection is 
lost (e.g. if pyOCD crashes without closing the connection cleanly) 
because AFAIK there's no indication that the PC end went away....I guess 
that's just something to be put up with.

Regards

DAVE

flit, 2020-10-18:
The probe firmware would be able to tell the host went away because there would be a USB reset. Often the probe will even be power cycled, for instance if it's connected to a laptop USB port (depends on the host controller and hardware design though).

mubes, 2020-10-18:
On 18/10/2020 22:49, Chris Reed wrote:
>
> The probe firmware would be able to tell the host went away because 
> there would be a USB reset. Often the probe will even be power cycled, 
> for instance if it's connected to a laptop USB port (depends on the 
> host controller and hardware design though).
>
OK, That must be an interface-level reset rather than a device-level 
reset then 'cos I've not noticed anything (My device has a few 
interfaces on it!), but there again I've not looked hard either.

flit, 2020-10-18:
I meant a device-level reset (the USB J/K Reset signaling), but for some reason I was thinking the whole host crashing and rebooting. You're absolutely right that there is no way for the probe to detect just pyocd (or another client) suddenly disconnecting.