pyocd/pyOCD #983: From Pack to built-in targets

LONGZR007, 2020-10-15:
Can PyOCD now generate built-in targets in a. Pack file?(Extract the algorithm from the.pack to make the target built-in)。

The advantage is that pyocd can be small enough when I want to package it as an exe; a .pack file is just too big。

flit, 2020-10-18:
No, sorry. Generating a built-in target module is still a manual process. It's an interesting idea, but I think the use cases would be limited.

It wouldn't be too hard to write a script to do this. Part of the code is already written in the [generate_blobs.py](https://github.com/pyocd/FlashAlgo/blob/master/scripts/generate_blobs.py) script from FlashAlgo. The main part that is missing is the memory region generation. If you do write a script, I'll be happy to merge a pull request.

Regardless, you can shrink a .pack file's size by removing all but these files:

- `*.pdsc`
- `*.flm`

These are the only files needed by pyocd. Make sure to keep the files in the same directories within the pack.

Fyi, Mbed Studio ships with stripped packs for some of the targets that it supports.

LONGZR007, 2020-10-20:
Is this <*.svd> file necessary?When I don't have it in my <*.pack>, I get the following error:
Exception in thread load-svd:

Traceback (most recent call last):
  File "e:\users\administrator\appdata\local\programs\python\python38-32\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "e:\users\administrator\appdata\local\programs\python\python38-32\lib\site-packages\pyocd\debug\svd\loader.py", line 74, in run
    self._svd_location.load()
  File "e:\users\administrator\appdata\local\programs\python\python38-32\lib\site-packages\pyocd\debug\svd\loader.py", line 50, in load
    self.device = SVDParser.for_xml_file(self.filename).get_device()
  File "e:\users\administrator\appdata\local\programs\python\python38-32\lib\site-packages\pyocd\debug\svd\parser.py", line 75, in for_xml_file
    return cls(ET.parse(path), remove_reserved)
  File "e:\users\administrator\appdata\local\programs\python\python38-32\lib\xml\etree\ElementTree.py", line 1202, in parse
    tree.parse(source, parser)
  File "e:\users\administrator\appdata\local\programs\python\python38-32\lib\xml\etree\ElementTree.py", line 584, in parse
    source = open(source, "rb")
TypeError: expected str, bytes or os.PathLike object, not NoneType

flit, 2020-10-21:
Oh, yes, the `*.svd` files are required. Sorry I forgot about those. (List above updated.)

LONGZR007, 2020-10-21:
Ok, thank you

flit, 2020-10-21:
Looks like a bit of a bug in handling the missing file, though.