pyocd/pyOCD #986: SemiHosting - TypeError: string argument without an encoding

joerg65, 2020-10-26:
I try to get debug messages via semihosting from a nrf52840 device and each message I get TypeError:

```3631888:ERROR:gdbserver:Unhandled exception in handle_message: string argument without an encoding
Traceback (most recent call last):
  File "/home/joerg/.local/lib/python3.8/site-packages/pyocd/gdbserver/gdbserver.py", line 418, in handle_message
    reply = handler(msg[msgStart:])
  File "/home/joerg/.local/lib/python3.8/site-packages/pyocd/gdbserver/gdbserver.py", line 666, in v_command
    return self.v_cont(cmd)
  File "/home/joerg/.local/lib/python3.8/site-packages/pyocd/gdbserver/gdbserver.py", line 717, in v_cont
    return self.resume(None)
  File "/home/joerg/.local/lib/python3.8/site-packages/pyocd/gdbserver/gdbserver.py", line 603, in resume
    was_semihost = self.semihost.check_and_handle_semihost_request()
  File "/home/joerg/.local/lib/python3.8/site-packages/pyocd/debug/semihost.py", line 441, in check_and_handle_semihost_request
    result = handler(args)
  File "/home/joerg/.local/lib/python3.8/site-packages/pyocd/debug/semihost.py", line 519, in handle_sys_write0
    msg = self._get_string(args)
  File "/home/joerg/.local/lib/python3.8/site-packages/pyocd/debug/semihost.py", line 478, in _get_string
    target_str = six.ensure_str(bytes(bytearray('')), encoding="ascii", errors="ignore")
TypeError: string argument without an encoding
```
BTW. I use a Pitaya-Link, and I don't know if this issue is related to pyOCD or Pitaya-Link firmware.

joerg65, 2020-10-30:
As no one seems to have time to look inside this, I as a noob in Python replaced this line 478 in semihost.py and the error is gone:
```
#target_str = six.ensure_str(bytes(bytearray('')), encoding="ascii", errors="ignore")
target_str = ""
```

joerg65, 2020-10-31:
> Hi @joerg65, just getting to it. Sorry about the bug! Thanks for reporting it. Sadly, this would have been caught by the semihosting unit test, but that test was somehow causing Python to crash on Travis CI so it was disabled.

Thank you, I tried your commit and it works for me.

flit, 2020-10-31:
Good to hear! Thanks for reporting back.

PR #987 (2020-11-01): Fix semihost issues with string/byte conversions
This patch fixes several issues with the semihosting code in how it handled data conversions. See #986 for one of the issues.

In addition, the semihosting unit is re-enabled and fixed. (It previously caused mysterious crashes of Python when run in Travis CI, so it had to be disabled.)

Fixes #986
closed 2020-11-01 by flit