pyocd/pyOCD #990: Eclipse and pyOCD v0.28.2, pyOCD can not start debugging.

lennvn, 2020-11-03:
I updated pyCOD to v0.28.2. When I launch the pyOCD debug configuration, pyCOD stop at "0001590:INFO:server:Semihost server started on port 4444 (core 0)" and the there's no debug window. I used pyOCD 0.22.1 before.

danielRep, 2020-11-03:
Afaik, in the case that you're using "Embedded C/C++ PyOCD Debugging" plugin as me, I experience the same issue as you. The problem is related with the use of the deprecated pyocd-gdbserver which is typically setup. I was able to get things working by starting the "pyocd gdb" server externally in a terminal. 

PS: I didn't test it, but yesterday @flit commit could be possibly solving that issue (https://github.com/pyocd/pyOCD/commit/1ba9b17f3288aaeecb168116b1ff556c678bc105). Hope that works for you.

flit, 2020-11-03:
Yes, #988 fixes the problem with starting `pyocd-gdbserver`. (The problem was reported by @joergw on the Eclipse Embedded CDT forum.)

I'm going to go ahead and close this issue since the fix is committed. A 0.28.3 version will be released probably today or tomorrow that includes the fix.

closed 2020-11-03 by flit

danielRep, 2020-11-04:
That kind of improvements will be great @flit !