pyocd/pyOCD #996: 0.28.2 pyocd-gdbserver crashing?

Microsparky, 2020-11-08:
Hello,

I am trying to set up a development environment using VS code, cortex-debug, arm-none-eabi and pyocd. 

When I start debugging with cortex-debug I get the following output:

VS Code popup:
```
Failed to launch GDB: Remote communication error. Target disconnected.: No connection could be made because the target machine actievly refused it. (from target-select extend-remote localhost:50000)
```

Debug Console:
```
Please check OUTPUT tab (Adapter Output) for output from pyocd-gdbserver
Launching server: "pyocd-gdbserver" "--persist" "--port" "50000" "--telnet-port" "50001"
Launching GDB: "arm-none-eabi-gdb.exe" "-q" "--interpreter=mi2"
undefinedC:\Program Files (x86)\GNU Arm Embedded Toolchain\9 2020-q2-update\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
Reading symbols from E:\Projects\libopencm3\microsparky-blink\src\microsparky-blink.elf...
```

Output:
```
0000775:WARNING:gdb_server:pyocd-gdbserver is deprecated; please use the new combined pyocd tool.
0001524:INFO:board:Target type is stm32f303retx
0001545:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
0001560:INFO:ap:AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
0001592:INFO:rom_table:AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=020 part=446)
0001608:INFO:rom_table:[0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0001619:INFO:rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
0001629:INFO:rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
0001640:INFO:rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
0001650:INFO:rom_table:[4]<e0040000:TPIU-M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=ca1:0:0>
0001661:INFO:rom_table:[5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
0001668:INFO:cortex_m:CPU core #0 is Cortex-M4 r0p1
0001687:INFO:cortex_m:FPU present: FPv4-SP-D16-M
0001697:INFO:dwt:4 hardware watchpoints
0001714:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0001767:INFO:server:Semihost server started on port 50001 (core 0)
```

I ran `netstat -a` and found no open port 50000 so I figured pyocd-gdbserver had not started the server, I tried to run the same command cortex-debug was using `-d debug` to get more information. I get the output below then after 4 or 5 seconds the prompt returns. 

I don't know where to go from here, did pyocd-gdbserver crash? What can I try?
```
> pyocd-gdbserver --version
0.28.2

> python --version
Python 2.7.13

> [System.Environment]::OSVersion.Version

Major  Minor  Build  Revision
-----  -----  -----  --------
10     0      18363  0

> pyocd-gdbserver -l
0000682:WARNING:gdb_server:pyocd-gdbserver is deprecated; please use the new combined pyocd tool.
  #   Probe                           Unique ID
----------------------------------------------------------------
  0   NUCLEO-F303RE [stm32f303retx]   066EFF494849887767242533

> pyocd-gdbserver --persist --port 50000 --telnet-port 50001 -d debug
0000702:WARNING:gdb_server:pyocd-gdbserver is deprecated; please use the new combined pyocd tool.
0000888:DEBUG:windows:Fetching mounted devices from disk service registry entry
0000888:DEBUG:windows:Fetching usb storage devices from USBSTOR service registry entry
0000890:DEBUG:windows:target_id_usb_id -> mount_point mapping: {u'066EFF494849887767242533': 'D:'}
0000890:DEBUG:windows:Unknown capabilities from the following ids: [u'usb\\class_ff&subclass_ff&prot_ff', u'usb\\class_ff&subclass_ff', u'usb\\class_ff']
0000891:DEBUG:windows:Composite device subdevice key SYSTEM\CurrentControlSet\Enum\USB\VID_0483&PID_374B&MI_03 was not found, skipping
0000891:DEBUG:windows:Composite device subdevice key SYSTEM\CurrentControlSet\Enum\USB\VID_0483&PID_374B&MI_04 was not found, skipping
0000894:DEBUG:session:Project directory: C:\Users\peter
0001423:INFO:board:Target type is stm32f303retx
0001430:DEBUG:stlink:STLink probe 066EFF494849887767242533 firmware version: V2J29M18
0001430:DEBUG:sequencer:Running task load_svd
0001432:DEBUG:sequencer:Running task pre_connect
0001432:DEBUG:sequencer:Running task dp_init
0001433:DEBUG:sequencer:Running task get_probe_capabilities
0001433:DEBUG:sequencer:Running task connect
0001449:DEBUG:dap:Default wire protocol selected; using SWD
0001452:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
0001453:DEBUG:sequencer:Running task clear_sticky_err
0001453:DEBUG:sequencer:Running task power_up_debug
0001457:DEBUG:sequencer:Running task check_version
0001459:DEBUG:sequencer:Running task create_discoverer
0001459:DEBUG:sequencer:Running task discovery
0001460:DEBUG:sequencer:Running task find_aps
0001466:DEBUG:sequencer:Running task create_aps
0001467:DEBUG:sequencer:Running task create_ap.0
0001470:DEBUG:ap:Using accelerated memory access interface
0001470:INFO:ap:AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
0001478:DEBUG:ap:AHB-AP#0 default HPROT=3 HNONSEC=0
0001480:DEBUG:ap:AHB-AP#0 implemented HPROT=3 HNONSEC=0
0001485:DEBUG:sequencer:Running task find_components
0001486:DEBUG:sequencer:Running task init_ap.0
0001503:INFO:rom_table:AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=020 part=446)
0001523:INFO:rom_table:[0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0001534:INFO:rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
0001546:INFO:rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
0001557:INFO:rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
0001580:INFO:rom_table:[5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
0001581:DEBUG:sequencer:Running task create_cores
0001581:DEBUG:discovery:Creating SCS-M4 component
0001585:INFO:cortex_m:CPU core #0 is Cortex-M4 r0p1
0001601:INFO:cortex_m:FPU present: FPv4-SP-D16-M
0001605:DEBUG:sequencer:Running task set_default_reset_type
0001607:DEBUG:sequencer:Running task create_components
0001607:DEBUG:discovery:Creating DWT component
0001614:INFO:dwt:4 hardware watchpoints
0001628:DEBUG:discovery:Creating FPB component
0001631:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0001634:DEBUG:fpb:fpb has been disabled
0001651:DEBUG:discovery:Creating ITM component
0001667:DEBUG:discovery:Creating TPIU-M4 component
0001670:DEBUG:sequencer:Running task check_for_cores
0001671:DEBUG:sequencer:Running task halt_on_connect
0001671:DEBUG:cortex_m:halting core 0
0001674:DEBUG:sequencer:Running task post_connect
0001674:DEBUG:sequencer:Running task post_connect_hook
0001676:DEBUG:sequencer:Running task create_flash
0001677:DEBUG:sequencer:Running task notify
0001680:DEBUG:cortex_m:Setting vector catch to 0x00000001
0001696:INFO:server:Semihost server started on port 50001 (core 0)
0001713:DEBUG:session:uninit session <pyocd.core.session.Session object at 0x0000000002FDEF98>
0001715:DEBUG:board:uninit board <pyocd.board.mbed_board.MbedBoard object at 0x000000000481E860>
0001746:DEBUG:cortex_m:resuming core 0
0001748:DEBUG:manager:added=[] removed=[]
0001749:DEBUG:manager:bps after flush={}
```

Microsparky, 2020-11-08:
pyocd gdbserver works as expected and doesn't return to the prompt
```
> pyocd gdbserver
0001430:INFO:board:Target type is stm32f303retx
0001448:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
0001464:INFO:ap:AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
0001497:INFO:rom_table:AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=020 part=446)
0001514:INFO:rom_table:[0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0001526:INFO:rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
0001536:INFO:rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
0001547:INFO:rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
0001559:INFO:rom_table:[4]<e0040000:TPIU-M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=ca1:0:0>
0001569:INFO:rom_table:[5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
0001572:INFO:cortex_m:CPU core #0 is Cortex-M4 r0p1
0001589:INFO:cortex_m:FPU present: FPv4-SP-D16-M
0001599:INFO:dwt:4 hardware watchpoints
0001618:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0001671:INFO:server:Semihost server started on port 4444 (core 0)
0001680:INFO:gdbserver:GDB server started on port 3333 (core 0)
```

flit, 2020-11-08:
Hi @Microsparky , this is a known bug that's fixed on the `master` branch. Version 0.28.3 with the fix will be released shortly.

flit, 2020-11-08:
Closing as a duplicate of #990.

closed 2020-11-08 by flit

Microsparky, 2020-11-08:
Thanks @flit I will try install directly from the source

flit, 2020-11-08:
No problem! The other issue wasn't particularly clear since it referenced Eclipse in the title.