pyocd/pyOCD #1001: Memory transfer fault on flash or erase, when using -uremote

patrislav1, 2020-11-11:
`pyocd server` reproducibly throws exception when flashing or erasing a chip. It is always the `flush` operation throwing a `TransferFaultError`.
The command continues and finishes, and the client side doesn't report an error.

This only happens when using the "remote probe access" feature. All `pyocd` subcommands (flash, erase, etc.) work OK when directly connected to the probe.

How to reproduce:
`pyocd server` in one terminal, `pyocd erase --chip -uremote:localhost` or `pyocd flash file.bin -uremote:localhost` in another one.
(Target and pack settings are in a `pyocd.yml`)
BTW., I see no exceptions when using `reset` or `gdbserver` with `-uremote`.

Server log file attached.
[probe_server_log.txt](https://github.com/pyocd/pyOCD/files/5526125/probe_server_log.txt)

flit, 2020-11-19:
Thanks for the report. Odd that the client doesn't report an error, but there may be a good reason.

What is the target's part number?
Could you please attach the client-side debug log too?

patrislav1, 2020-11-23:
I attached both client and server logs together with the yaml config (command is `pyocd erase --chip -uremote:localhost`, target is `atsam4ls8c` using the `SAM4L` pack file from Keil):
[pyocd_logs.zip](https://github.com/pyocd/pyOCD/files/5582076/pyocd_logs.zip)

BTW. for my own dev environment i moved away from pyocd client/server mode, as it was much too slow over internet (flashing a chip takes like 15 minutes). I created a little [remote wrapper](https://github.com/patrislav1/pyocd_remote) for pyOCD that uses SSH to make it look like a local pyocd instance - so all the SWD stuff is done at the remote machine and doesn't have to go back and forth through the network bottleneck.