pyocd/pyOCD #1003: HIDError

LONGZR007, 2020-11-12:
I got the following error when using API. This problem has been bothering me for a long time. Could you please help me see the reason?I packaged it as an exe and that's the same problem:
Traceback (most recent call last):
  File "G:\GitHub\others-github\DMCUProg\dap_test.py", line 64, in <module>
    test.on_btnWrite_clicked()
  File "G:\GitHub\others-github\DMCUProg\dap_test.py", line 21, in on_btnWrite_clicked
    self.cpu = self.openDAP()
  File "G:\GitHub\others-github\DMCUProg\dap_test.py", line 51, in openDAP
    self.daplink.open()
  File "G:\GitHub\others-github\DMCUProg\pyocd\probe\cmsis_dap_probe.py", line 137, in open
    self._link.open()
  File "G:\GitHub\others-github\DMCUProg\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 569, in open
    self._packet_count = self._protocol.dap_info(self.ID.MAX_PACKET_COUNT)
  File "G:\GitHub\others-github\DMCUProg\pyocd\probe\pydapaccess\cmsis_dap_core.py", line 125, in dap_info
    self.interface.write(cmd)
  File "G:\GitHub\others-github\DMCUProg\pyocd\probe\pydapaccess\interface\pywinusb_backend.py", line 149, in write
    self.report.send([0] + data)
  File "C:\Python39\lib\site-packages\pywinusb\hid\core.py", line 1474, in send
    raise HIDError("Report size has to be %d elements (bytes)" \
pywinusb.hid.helpers.HIDError: Report size has to be 1025 elements (bytes)

closed 2020-11-12 by LONGZR007

flit, 2020-11-12:
@LONGZR007 Did you figure out the problem?

LONGZR007, 2020-11-13:
@flit  Yes, I saw on [#686](https://github.com/pyocd/pyOCD/issues/686) that you answered this question for others, thank you