pyocd/pyOCD #1004: Build pyinstaller

zhepoch, 2020-11-13:


flit, 2020-11-14:
Take a look at my [feature/binary_build](https://github.com/flit/pyOCD/tree/feature/binary_build) branch. There is a `scripts/pyocd.spec` file and `scripts/build_pyinstaller.sh`.  Unfortunately, it doesn't work yet due to an issue with the prettytable dependency. It builds ok, but when you run pyocd a `pkg_resources.DistributionNotFound` exception is raised when prettytable tries to call `pkg_resources.get_distribution()`. There is an attempt at a hook under `scripts/hooks` based on pyinstaller/pyinstaller#4809, but it doesn't work.

If you can make progress, that would by great! You can either comment here with your improvements or create a PR to my feature/binary_build branch.

Note that another option I haven't yet explored but seems promising is PyOxidizer.

Littermonkey, 2020-11-16:
I found the root cause that package (Prettytable) __init__.py is an issue. and  I just added some code in that file by follow up :
`
from pkg_resources import get_distribution, DistributionNotFound
import os.path

try:
    _dist = get_distribution('prettytable')
    # Normalize case for Windows systems
    dist_loc = os.path.normcase(_dist.location)
    here = os.path.normcase(__file__)
    if not here.startswith(os.path.join(dist_loc, 'prettytable')):
        # not installed, but there is another version that *is*
        raise DistributionNotFound
except DistributionNotFound:
    __version__ = 'Please install this project with setup.py'
else:
    __version__ = _dist.version
`
it can work,  but when I execute './pyocd list --target' expose some warnning: 
`FileNotFoundError: [Errno 2] No such file or directory: '/tmp/_MEIacUmZi/pyocd/debug/svd/svd_data.zip'
0000288:WARNING:loader:unable to open builtin SVD file: [Errno 2] No such file or directory: '/tmp/_MEIacUmZi/pyocd/debug/svd/svd_data.zip'
Traceback (most recent call last):
  File "pyocd/debug/svd/loader.py", line 37, in from_builtin
  File "pkg_resources/__init__.py", line 1135, in resource_stream
  File "pkg_resources/__init__.py", line 1383, in get_resource_stream
  File "pkg_resources/__init__.py", line 1386, in get_resource_string
  File "pkg_resources/__init__.py", line 1553, in _get
  File "/home/glb/build_pyocd/env/lib/python3.6/site-packages/PyInstaller/loader/pyimod03_importers.py", line 341, in get_data
    with open(path, 'rb') as fp:
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/_MEIacUmZi/pyocd/debug/svd/svd_data.zip'
0000288:WARNING:loader:unable to open builtin SVD file: [Errno 2] No such file or directory: '/tmp/_MEIacUmZi/pyocd/debug/svd/svd_data.zip'
Traceback (most recent call last):
  File "pyocd/debug/svd/loader.py", line 37, in from_builtin
  File "pkg_resources/__init__.py", line 1135, in resource_stream
  File "pkg_resources/__init__.py", line 1383, in get_resource_stream
  File "pkg_resources/__init__.py", line 1386, in get_resource_string
  File "pkg_resources/__init__.py", line 1553, in _get
  File "/home/glb/build_pyocd/env/lib/python3.6/site-packages/PyInstaller/loader/pyimod03_importers.py", line 341, in get_data
    with open(path, 'rb') as fp:
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/_MEIacUmZi/pyocd/debug/svd/svd_data.zip'
  Name                      Vendor                  Part Number                  Families   Source   
-----------------------------------------------------------------------------------------------------
  cc3220sf                  Texas Instruments       CC3220SF                                builtin  
  cortex_m                  Generic                 CoreSightTarget                         builtin  
  cy8c64_sysap              Cypress                 cy8c64_sysap                            builtin  
  cy8c64x5_cm0              Cypress                 cy8c64x5_cm0                            builtin  
  cy8c64x5_cm0_full_flash   Cypress                 cy8c64x5_cm0_full_flash                 builtin  
  cy8c64x5_cm4              Cypress                 cy8c64x5_cm4                            builtin  
  cy8c64x5_cm4_full_flash   Cypress                 cy8c64x5_cm4_full_flash                 builtin  
  cy8c64xa_cm0              Cypress                 cy8c64xA_cm0                            builtin  
  cy8c64xa_cm0_full_flash   Cypress                 cy8c64xA_cm0_full_flash                 builtin  
  cy8c64xa_cm4              Cypress                 cy8c64xA_cm4                            builtin  
  cy8c64xa_cm4_full_flash   Cypress                 cy8c64xA_cm4_full_flash                 builtin  
  cy8c64xx_cm0              Cypress                 cy8c64xx_cm0                            builtin  
  cy8c64xx_cm0_full_flash   Cypress                 cy8c64xx_cm0_full_flash                 builtin  
  cy8c64xx_cm0_nosmif       Cypress                 cy8c64xx_cm0_nosmif                     builtin  
  cy8c64xx_cm0_s25hx512t    Cypress                 cy8c64xx_cm0_s25hx512t                  builtin  
  cy8c64xx_cm4              Cypress                 cy8c64xx_cm4                            builtin  
  cy8c64xx_cm4_full_flash   Cypress                 cy8c64xx_cm4_full_flash                 builtin  
  cy8c64xx_cm4_nosmif       Cypress                 cy8c64xx_cm4_nosmif                     builtin  
  cy8c64xx_cm4_s25hx512t    Cypress                 cy8c64xx_cm4_s25hx512t                  builtin  
  cy8c6xx5                  Cypress                 CY8C6xx5                                builtin  
  cy8c6xx7                  Cypress                 CY8C6xx7                                builtin  
  cy8c6xx7_nosmif           Cypress                 CY8C6xx7_nosmif                         builtin  
  cy8c6xx7_s25fs512s        Cypress                 CY8C6xx7_S25FS512S                      builtin  
  cy8c6xxa                  Cypress                 CY8C6xxA                                builtin  
  hc32f003                  HDSC                    HC32F003                                builtin  
  hc32f005                  HDSC                    HC32F005                                builtin  
  hc32f030                  HDSC                    HC32F030                                builtin  
  hc32f072                  HDSC                    HC32F072                                builtin  
[... 75 lines cut ...]
[... cut ...]
  stm32l031x6               STMicroelectronics      STM32L031x6                             builtin  
  stm32l475xc               STMicroelectronics      STM32L475xC                             builtin  
  stm32l475xe               STMicroelectronics      STM32L475xE                             builtin  
  stm32l475xg               STMicroelectronics      STM32L475xG                             builtin  
  w7500                     WIZnet                  W7500                                   builtin  

`

Littermonkey, 2020-11-16:
I found the root cause that package (Prettytable) __init__.py is an issue. and I just added some codes in that file like this :
`from pkg_resources import get_distribution, DistributionNotFound
import os.path
try:
_dist = get_distribution('prettytable')
dist_loc = os.path.normcase(_dist.location)
here = os.path.normcase(file)
if not here.startswith(os.path.join(dist_loc, 'prettytable')):
raise DistributionNotFound
except DistributionNotFound:
version = 'Please install this project with setup.py'
else:
version = _dist.version
`

Littermonkey, 2020-11-16:
Despair is that when I execute "./pyocd list --target"  explore some warnnings, so I'm not sure if it's still a build binary issue. 
`./dist/pyocd list --target
0000215:WARNING:loader:unable to open builtin SVD file: [Errno 2] No such file or directory: '/tmp/_MEIp1l1zs/pyocd/debug/svd/svd_data.zip'
Traceback (most recent call last):
  File "pyocd/debug/svd/loader.py", line 37, in from_builtin
    zip_stream = pkg_resources.resource_stream("pyocd", BUILTIN_SVD_DATA_PATH)
  File "pkg_resources/__init__.py", line 1135, in resource_stream
  File "pkg_resources/__init__.py", line 1383, in get_resource_stream
  File "pkg_resources/__init__.py", line 1386, in get_resource_string
  File "pkg_resources/__init__.py", line 1553, in _get
  File "/home/glb/build_pyocd/env/lib/python3.6/site-packages/PyInstaller/loader/pyimod03_importers.py", line 341, in get_data
    with open(path, 'rb') as fp:
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/_MEIp1l1zs/pyocd/debug/svd/svd_data.zip'
`