pyocd/pyOCD #1009: Enhancement Request: Publish SWO/SWV stream to TCP port

mubes, 2020-11-17:
Hi,

For a CMSIS-DAP v2 interface, would it be possible to publish the SWO output (the second USB IN stream) to a TCP port? That would allow utilities such as Orbuculum to connect directly to it (indeed, if it was on port 3443 it would be transparent and wouldn't even need any command line options). I think OpenOCD is supporting this in development builds now.

Regards
DAVE

flit, 2020-11-19:
Definitely would be nice to have this. I can't really provide an estimate as to when it will be implemented; there are many other tasks on the table. (Any help would be appreciated.)

Thanks for the info about port 3443, I wasn't aware that was Orbuculum's default port.

mubes, 2020-11-20:
Hi Chris,

I'm the author of Orbuculum and will happily do some reasonably thorough 
testing with it once you've got it in pyocd. I'd do it myself but my 
Py-foo is weak and it would just get added to the 1001 other things I'm 
not doing... If you use 3443 then pyocd will plug directly into the 
various orb clients and orbuculum itself (which is the multiplexer) can 
be bypassed; Having access to orbtop alone makes it definitely worth the 
effort of adding the functionality.

Regards

DAVE

flit, 2020-11-20:
🤣 Nice to meet you! And I totally understand you about the 1001 other things…

PR #1014 (2020-11-29): Publish raw SWV stream to TCP port
This patch implements a [raw SWO stream server](https://github.com/pyocd/pyOCD/issues/1009) for tools like [orbuculum](https://github.com/orbcode/orbuculum).

I introduced a new option `tpiu_port` that is set to 3443 by default. The TPIU raw server is started implicitly, when `enable_swv` is activated. (If this is not desirable, then the default could also be changed to zero and the user would have to set it explicitly to 3443 or whatever to enable it).

I tested it on a STM32 Nucleo board with `pyocd gdbserver --target stm32l433rctxp -O enable_swv=true -O sw [... cut]
closed 2020-11-29 by flit