pyocd/pyOCD #1026: "STLink error (16): AP wait" connecting to STM32F750

j4cbo, 2020-12-14:
I'm using an STLINK-V3MINI to try to talk to an STM32F750 chip over SWD. On connect I get a "Error attempting to probe CoreSight component referenced by ROM table entry #0: STLink error (16): AP wait", followed by "No cores were discovered!"; I haven't found any other reports of the "AP wait" error so I'm not sure where to go from here :/

The code on the micro isn't using and hasn't entered any low-power modes (`WFI` etc) at the point where I'm trying to connect.

Here's the full output:

```$ pyocd commander -t cortex_m -vv
0001332:DEBUG:session:Project directory: /Users/jacob
0001332:INFO:board:Target type is cortex_m
0001337:DEBUG:stlink:STLink probe 0039001F3438511134313939 firmware version: V3J7M2
0001338:DEBUG:sequencer:Running task load_svd
0001338:DEBUG:sequencer:Running task pre_connect
0001339:DEBUG:sequencer:Running task dp_init
0001339:DEBUG:sequencer:Running task get_probe_capabilities
0001339:DEBUG:sequencer:Running task connect
0001340:DEBUG:dap:Default wire protocol selected; using SWD
0001340:INFO:dap:DP IDR = 0x5ba02477 (v2 rev5)
0001340:DEBUG:sequencer:Running task clear_sticky_err
0001341:DEBUG:sequencer:Running task power_up_debug
0001342:DEBUG:sequencer:Running task check_version
0001342:DEBUG:sequencer:Running task create_discoverer
0001342:DEBUG:sequencer:Running task discovery
0001342:DEBUG:sequencer:Running task find_aps
0001343:DEBUG:sequencer:Running task create_aps
0001343:DEBUG:sequencer:Running task create_ap.0
0001344:DEBUG:ap:Using accelerated memory access interface
0001344:INFO:ap:AHB-AP#0 IDR = 0x74770001 (AHB-AP var0 rev7)
0001345:DEBUG:ap:AHB-AP#0 default HPROT=b HNONSEC=0
0001346:DEBUG:ap:AHB-AP#0 implemented HPROT=f HNONSEC=1
0001346:DEBUG:sequencer:Running task find_components
0001346:DEBUG:sequencer:Running task init_ap.0
0001348:INFO:rom_table:AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fd000 (designer=020 part=449)
0001349:INFO:rom_table:[0]<e00fe000:ROM class=1 designer=43b part=4c8>
0001349:INFO:rom_table:  AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00fe000 (designer=43b part=4c8)
0001350:INFO:rom_table:  [0]<e00ff000:ROM class=1 designer=43b part=4c7>
0001350:INFO:rom_table:    AHB-AP#0 Class 0x1 ROM table #2 @ 0xe00ff000 (designer=43b part=4c7)
0001352:ERROR:rom_table:Error attempting to probe CoreSight component referenced by ROM table entry #0: STLink error (16): AP wait
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pyocd/coresight/rom_table.py", line 349, in _read_table
    self._handle_table_entry(entry, entryNumber)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pyocd/coresight/rom_table.py", line 399, in _handle_table_entry
    cmpid.read_id_registers()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pyocd/coresight/rom_table.py", line 111, in read_id_registers
    regs = self.ap.read_memory_block32(self.top_address + self.IDR_READ_START, self.IDR_READ_COUNT)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pyocd/probe/stlink_probe.py", line 272, in read_memory_block32
    return conversion.byte_list_to_u32le_list(self._link.read_mem32(addr, size * 4, self._apsel))
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pyocd/probe/stlink/stlink.py", line 387, in read_mem32
    return self._read_mem(addr, size, Commands.JTAG_READMEM_32BIT, self.MAXIMUM_TRANSFER_SIZE, apsel)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pyocd/probe/stlink/stlink.py", line 347, in _read_mem
    raise self._ERROR_CLASSES[status](error_message)
pyocd.core.exceptions.TransferTimeoutError: STLink error (16): AP wait
0001353:INFO:rom_table:    [1]<e0001000:DWT class=14 designer=43b part=002>
0001354:INFO:rom_table:    [2]<e0002000:FPB class=14 designer=43b part=00e>
0001355:INFO:rom_table:    [3]<e0000000:ITM class=14 designer=43b part=001>
0001355:INFO:rom_table:  [1]<e0041000:ETM-M7 class=9 designer=43b part=975 devtype=13 archid=4a13 devid=0:0:0>
0001356:INFO:rom_table:[1]<e0040000:TPIU-M7 class=9 designer=43b part=9a9 devtype=11 archid=0000 devid=ca1:0:0>
0001356:DEBUG:sequencer:Running task create_cores
0001356:DEBUG:sequencer:Running task create_components
0001356:DEBUG:discovery:Creating DWT component
0001357:INFO:dwt:4 hardware watchpoints
0001359:DEBUG:discovery:Creating FPB component
0001360:INFO:fpb:8 hardware breakpoints, 1 literal comparators
0001360:DEBUG:fpb:fpb has been disabled
0001363:DEBUG:discovery:Creating ITM component
0001365:DEBUG:discovery:Creating TPIU-M7 component
0001365:DEBUG:sequencer:Running task check_for_cores
Exception while initing board: No cores were discovered!
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pyocd/commands/commander.py", line 206, in _post_connect
    self.session.open(init_board=not self.args.no_init)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pyocd/core/session.py", line 461, in open
    self._board.init()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pyocd/board/board.py", line 85, in init
    self.target.init()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pyocd/core/soc_target.py", line 120, in init
    seq.invoke()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pyocd/coresight/coresight_target.py", line 220, in check_for_cores
    raise exceptions.DebugError("No cores were discovered!")
pyocd.core.exceptions.DebugError: No cores were discovered!

```

After this, the chip is in an _extremely_ stuck state, to the point that even asserting the RESET pin seems to have no effect.

flit, 2020-12-14:
Hi @j4cbo , the STM32 devices based on Cortex-M7 (F7 and H7) have been problematic for some reason.

Can you try connecting under reset? Just add the `-connect=under-reset` argument to the command line.

I checked the Keil STM32F7 CMSIS-Pack and didn't find any extra debug logic, so at this point the only thing I can think of is the connect under reset option. I'll try to find time to read through the F750 reference manual to see if they mention anything that's out of the ordinary.

j4cbo, 2020-12-14:
No change with `--connect=under-reset`.

I wonder if this is an ST-Link V3 issue? I don't have a standalone ST-Link V2, but I do have an STM32F750 Discovery on hand, which has an onboard ST-Link _V2_, and connecting to it seems to work just fine. (Though it's running different firmware, and the dev board has an STM32F750N8H6 vs. my DUT's STM32F750V8T6.) The V2 on the dev board is connected to its microcontroller via solder bridges, so I'll see if I can cobble it together to my board and narrow down the problem that way.

j4cbo, 2020-12-16:
Well, I switched to my other (theoretically identical) board and it seems to work exactly like it should. I'm still at the hand-soldered prototype stage here, so I think this was just a hardware problem with one of my boards. Sorry for the false alarm!

closed 2020-12-16 by j4cbo

flit, 2020-12-16:
Ok, I'm glad to hear that, but I'm not entirely convinced that it's not my fault. 😁 It could be an issue in pyocd related to STLink V2 vs V3. If you have more trouble, please reopen this issue. Thanks for reporting back!

Btw, if you haven't already, make sure the firmware in your V3 is updated. There were some bugs in early versions.