pyocd/pyOCD #1043: Wrong r12 offset in FreeRTOS stack frame

newbrain, 2021-01-09:
Offset of r12 in FreeRTOS stack frame dictionary at line 102 in pyocd/rtos/freertos.py is not correct:
https://github.com/pyocd/pyOCD/blob/f146817ad2233d247e70fbba6adfe03b2c3fd8cc/pyocd/rtos/freertos.py#L101-L103

The correct value should be 52, (42 is, oddly, not the correct answer here).

flit, 2021-01-12:
Oops, thanks for the report. 

> (42 is, oddly, not the correct answer here)

🤣 

Fyi, I have an update to the FreeRTOS plugin that supports FreeRTOS 10.x. It needs more testing, and I haven't yet checked whether it breaks compatibility with FreeRTOS 8.x/9.x. The branch is https://github.com/flit/pyOCD/tree/feature/freertos_10.

newbrain, 2021-01-12:
> Fyi, I have an update to the FreeRTOS plugin that supports FreeRTOS 10.x.
Interesting, thanks.

I'm working on a ThreadX (AzureRTOS) plugin, that's why I've noticed the typo.
It seems to work but I need to set up a compliant development environment and to test it according to the Contributing.md document, till now I've just added the new code on my installation(s).
If all goes well, I'll make a pull request in 1-2 weeks.

flit, 2021-01-13:
> I'm working on a ThreadX (AzureRTOS) plugin

Sweet! If you need any help, just ask. You can either email me or use the new GitHub [Discussions](/pyocd/pyOCD/discussions).

PR #1045 (2021-01-13): FreeRTOS: fix r12 offset for FPU-enabled normal frame
Fixes #1043 

Thanks to @newbrain for finding and reporting the issue!
closed 2021-01-13 by flit