pyocd/pyOCD #1050: Test test_no_timeout will always fail on Windows

newbrain, 2021-01-16:
On Windows, the monotonic clock used for sleep() by Python platform has about 15 ms resolution, as described in [PEP564](https://www.python.org/dev/peps/pep-0564/#id23).

The test below is bound to fail as each iteration will consume at least 15 ms in sleep(0.01), for a total of at least 60 ms.
The test can pass if a reasonably higher than 50 ms timeout is used.

https://github.com/pyocd/pyOCD/blob/b0ab779c3e347a29adf1f5e9f8b2d676c3c9cf05/test/unit/test_timeout.py#L23-L33

As demonstrated by the following simple script, a possible value should probably not be much lower than 100 ms.
```
(venv3) ~\Git\pyOCD [master ≡ +0 ~1 -0 !]> python.exe
Python 3.7.9 (tags/v3.7.9:13c94747c7, Aug 17 2020, 16:30:00) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from time import time
>>> from time import sleep
>>> Max = 0
>>> for i in range(10000):
...     a=time()
...     sleep(0.01); sleep(0.01); sleep(0.01); sleep(0.01)
...     a= time()-a
...     if a > Max:
...             Max = a
...
>>> Max 
0.09961342811584473
```

flit, 2021-01-17:
Thanks! I'll try to get a fix in soon. Actually, I've been wanted to rewrite the test to mock `sleep()` so it doesn't actually sleep.

newbrain, 2021-03-09:
Success!
```
test\unit\test_timeout.py ....                                                                                         [100%]

===================================================== warnings summary ====================================================== 
venv\lib\site-packages\pyreadline\py3k_compat.py:8
  C:\Users\feder\Git\pyOCD\venv\lib\site-packages\pyreadline\py3k_compat.py:8: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated since Python 3.3, and in 3.10 it will stop working        
    return isinstance(x, collections.Callable)

-- Docs: https://docs.pytest.org/en/stable/warnings.html

----------- coverage: platform win32, python 3.9.1-final-0 -----------
Coverage HTML written to dir htmlcov

============================================== 859 passed, 1 warning in 20.73s ============================================== 
(venv) ~\Git\pyOCD [(v0.30.0)]> 
```
OTOH, I found a problem with _my code_ when giving an explicit uid on the command line and a picoprobe is _not connected_
I'll write an issue, and a PR to correct it.
I apologize, I was quite sure I had tested that specific case. 😔

flit, 2021-03-09:
Cool, thanks for testing.

picoprobe uid issue: Yeah, I got a report of that today from the Mbed Studio team, too, as they are testing to update to v0.30. I might get to it before you. And no worries, I've done equivalent and worse! 😄

closed 2021-03-10 by flit